espressif/esp-idf #10430: ULP I2C only available on GPIO0-3

readmodifywrite, 2022-12-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I was surprised to get this error on the ULP RISCV I2C init: 

E (533) ulp_riscv_i2c: SDA pin can only be configured as GPIO#1 or GPIO#3
E (541) ulp_riscv_i2c: i2c_set_pin(91): RTC I2C GPIO config invalid
E (548) ulp_riscv_i2c: ulp_riscv_i2c_master_init(418): Failed to configure RTC I2C GPIOs

This is apparently some kind of hardware limitation, though you have to dig pretty deep into the TRM to find it.

Why is this?  The I2C can normally map to any pin, along with every other peripheral on the chip.  This seems like a very odd exception.

It's especially troublesome because GPIO0 is strapped as the boot pin, and GPIO3 is used for USB JTAG select.  So there's only 2 completely safe pins to use for the ULP I2C.

I guess I'll be writing a bit bang driver, but I'd at least like to understand why this extremely specific limitation exists for this extremely commonly used peripheral.  Are there any other weird pin connection limitations like this?

igrr, 2022-12-26:
That's indeed a hardware limitation. For example, for ESP32-S3 it is mentioned both in the RTC I2C controller section and in the table 6­4 "RTC Functions of RTC IO MUX Pins". I'm not sure we can answer the "why" question in the context of the ESP-IDF project. Probably the chip designers have decided that it's not reasonable to spend more hardware resources for having a larger multiplexer or crossbar for the RTC IO controller.  I do agree with you that it ends up being pretty limiting this way.

readmodifywrite, 2022-12-28:
@igrr Thanks for the response.

This, plus the sub-register limitations, unfortunately make it pretty useless in the general case.

I'm going to switch to a bit-bang driver for the ULP, as it is particularly well suited to bit-banging anyway.

closed 2022-12-28 by readmodifywrite