espressif/esp-idf #10431: esptool flash load hanging on master

readmodifywrite, 2022-12-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Master branch, commit cfef24863f1acd4418227a727dbb6e356438a385 (5.x development)

ESP32-S3

I've noticed esptool hangs very frequently on flash loads:

esptool esp32s3 -p /dev/ttyACM1 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 bootloader/bootloader.bin 0x10000 twine.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.dev0
Serial port /dev/ttyACM1

It just sits there.  Power consumption indicates it has made contact with the chip, but then it stops indefinitely.  Sometimes it works, sometimes it doesn't.  This never happened on the 4.4.x branch, it used to work 100% of the time.

radimkarnis, 2022-12-23:
Hello @readmodifywrite,
thanks for reporting this. I suspect a stub flasher regression was introduced in esptool v4.4. 

Could you please try installing esptool v4.3 (`pip install esptool==4.3`) in your environment and running the command again? Also, you could just try running the command with the `--no-stub` argument. Thank you!

Fishwaldo, 2022-12-24:
Hi - Not the reporter here, but I was having similar issues. 
* hangs
* errors after starting to flash
Downgrading to 4.3 or using no-stub fixed the issue for me.

radimkarnis, 2022-12-28:
Hello @Fishwaldo,
thanks for your feedback! Could I please ask you to try [this possible fix](https://github.com/radimkarnis/esptool/tree/test/lower_cpu_freq)? It would help a lot if you could test this.

To try:
1) `git clone https://github.com/radimkarnis/esptool`
2) `cd esptool`
3) `git checkout test/lower_cpu_freq`
4) `pip install -e .`
5) run your esptool commands

You can do this when your IDF environment is active and try with `idf.py`, or by running `esptool.py` separately. Thanks again!

Fishwaldo, 2022-12-28:
I can test but can I revert/uninstall if it doesn't work? (I need my dev machine to be functional!)

radimkarnis, 2022-12-28:
@Fishwaldo that would be great! 
of course, you can run `pip install esptool==4.3` afterwards. That will remove the development version and replace it with a release again.

readmodifywrite, 2022-12-28:
@radimkarnis Thanks for the response, I'll try that as soon as I get a chance (after the holidays, most likely).

liux-pro, 2023-01-04:
Same problem on windows 10, esp32s3, IDF v5.0-release with default esptool v4.4. 
Especially when the firmware is large.

esptool 4.3 and 4.5-dev (@radimkarnis fork) works fine.

Actually, esptool v4.4 may somehow break the chip flash temporary,  all version of esptool not work before I run `esptool erase_flash`.

radimkarnis, 2023-01-10:
The fix has been released in estpool `v4.5.dev1`. This version containing the fix will be automatically installed if you re-run the ESP-IDF install script.

closed 2023-01-10 by radimkarnis