espressif/esp-idf #10432: ulp_riscv_i2c_master_read_from_device erroneously writes to sub register before reading

readmodifywrite, 2022-12-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2509-gcfef24863f

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I2C works properly in ULP

### What is the actual behavior?

I2C does not work properly in ULP.

It looks like the ULP I2C driver is designed for a very specific use case that doesn't universally apply to I2C chips.

The  master read and write functions both begin by writing to an 8 bit sub register, and then performing the requested operation.  This is completely incorrect behavior for an I2C driver:

1. Not all I2C devices use sub registers in this way.
2. When they do, they are often not limited to 8 bits in length.
3. Some devices require a write to a subregister, then a delay, and then a read without another preceding write.

A generic I2C driver doesn't need to do anything more than write bytes and read bytes - exactly the bytes the engineer programs it to use.  Adding an automatic sub-register set up functionality just gets in the way and breaks a ton of I2C device protocols.

### Steps to reproduce.

Use the ULP I2C driver with something other than the BMP180 demo.

dobraMorda, 2022-12-24:
@readmodifywrite  I fully agree with that. I believe that the entire RISCV ULP I2C driver should be rebuilt. I want to use it with sensiron sensors but it has 2bytes addressing and dont need sub register to read data.

sudeep-mohanty, 2022-12-28:
Hi @readmodifywrite, @dobraMorda,
Thanks for trying out the ULP RISC-V I2C driver. We really appreciate the feedback.
Unfortunately, this driver is limited by the hardware peripheral design and as it stands, the RTC I2C peripheral *always* expects a sub register address to be programmed. If a sub register is not programmed then the peripheral writes `0x00` as the sub register address.

I completely agree that this limits the usage of the RTC I2C peripheral and is not universally compatible with all I2C sensors. Not all sensors need a sub register to be programmed. There are also other limitations in the driver such as resorting to a polling mechanism to read/write multiple bytes which at times does not work reliably with some sensors. This is again due to the hardware peripheral not having buffer to read/write data and can only perform single byte read/writes. We are working on improving the driver to make it more universally compatible with I2C sensors. There is a task for the same under #9990.

readmodifywrite, 2022-12-28:
@sudeep-mohanty  Thanks for the response.  It sounds like this is a hardware limitation, so I won't hold my breath for a fix - I'm committed to the S3 on this design.

closed 2022-12-28 by readmodifywrite

NoNullptr, 2023-06-30:
Just wow. I am currently in the process of moving a firmware over to using the ULP, which is the sole reason why we decided to use the ESP32-S3 chip. My entire experience with espressif has been a disappointment, their employees not understanding bug reports, closing bug reports even if it's not resolved, and now I learn that they mess up on the simplest protocol in existence, namely I2C. The fact that the chip is advertised to support I2C in RTC/deep sleep is simply false advertising. The entire specification that needed to be implemented (https://www.nxp.com/docs/en/user-guide/UM10204.pdf) starts on page 3 and ends on page 14. This whole "subaddresses" thing is completely made up by espressif and simply doesn't work with or apply to the vast majority of sensors. Such a letdown, and now I've got to inform management that the espressif chips have to scrapped and we have to look into alternatives

sudeep-mohanty, 2023-06-30:
@NoNullptr Apologies for the trouble you have had with the ULP I2C. Unfortunately, the hardware limitation with the peripheral can't be worked around with. We have taken this as a feedback and have improved the low-power I2C peripheral on our newer chip, the ESP32-C6 where the I2C peripheral follows the I2C specs more closely. We have [documented](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/ulp-risc-v.html#_CPPv439ulp_riscv_i2c_master_set_slave_reg_addr7uint8_t) this behavior in the API docs but I suppose we can improve on the documentation to make this limitation more prominent.
Also, the ULP-RISCV should be able to handle bit-bangled I2C implementation so that could be an alternative to using the built in I2C peripheral in your design.

NoNullptr, 2023-06-30:
@sudeep-mohanty Thank you for your rapid response. I am looking at the hardware specification right now to see if it could be salvaged for our case, but it doesn't seem like it. I will certainly not write a bit-banged I2C implementation that we will then have to maintain long-term – the whole point of using integrated chips is that basic functions like I2C are integrated. Tbh I will also recommend against using any espressif chips in the future – this project sadly has been the worst developer experience I've had in any project so far. And while the code you're contributing might be of better quality, the overall code quality of espressif seems to be shockingly low – one constantly encounters bugs, there are multiple and often bad implementations doing the same thing in esp-idf, the documentation is dubious at best, etc. While espressif's chips are very cheap, we also have to factor in developer time on our end, the support and reaction times we have seen, etc. Overall espressif simply doesn't come close to the industry standard. If the ESP32-C6 was pin-compatible to the ESP32-S3 I could ask to get it evaluated with what we have developed for the S3, but that's not the case either, so it's simply not going to happen.

NoNullptr, 2023-06-30:
@sudeep-mohanty Case in point about the documentation. The implementation `ulp_riscv_i2c.c` has to use some secret hardware documentation, since the technical reference manual does not contain even a fraction of the information needed to actually talk to the hardware and come up with the implementation or know about the limitations. E.g. I2C_RD is illustrated as always performing the same operations, yet there are command0 to command6 that do not have such a restriction. The ulp_riscv_i2c sets it up in a fixed way, but the documentation doesn't even mention that the commands have to be set up like this (and why would that be necessary from a hardware standpoint of view anyways?). Then the code sets SET_PERI_REG_BITS with undocumented bits. The quality of the documentation is simply infuriatingly bad

sudeep-mohanty, 2023-06-30:
@NoNullptr We are cognisant of the fact that the TRM documentation for the RTC I2C peripheral is lacking in details and we have internal tickets open to address this difference between the documentation and the driver implementation. Hopefully, we shall have the docs sorted soon!

NoNullptr, 2023-06-30:
@sudeep-mohanty It's not just the RTC I2C. During the last few months, whenever I was personally involved in this project, I referred to the TRM quite regularly, and literally every single chapter I looked at is missing key information, rendering the document quite useless. Espressif is simply not a company that can be taken seriously. For your own sake, I hope that you can find a better company to work for – everyone is always looking for competent engineers, and the pay will certainly be better than whatever Espressif is coming up with, otherwise the overall quality wouldn't be what it is.

readmodifywrite, 2023-07-04:
FWIW, IO is not the ESP32's strength.  It is designed to be a pretty good WiFi MCU at a *very* competitive price point, and I think Espressif really nailed that.  If you want amazing IO, you get an ST or Atmel or Silabs, etc.

I would love to see better docs (especially on the newer parts like the S3/C3), and frankly the I2C issue on the S3 ULP is a real showstopper for my current use case (I'm designing the S3 out and going to a C3 with a separate MCU for IO).  And I really recommend Espressif do *something* for the I2C here, because this really is a pretty major problem for a very heavily used (and not very complicated) interface.

But I don't think it's fair to say Espressif can't be taken seriously.  They have been very responsive on issues I've filed, and the WiFi itself works great.  In general the IDF works pretty well too, and the firmware tooling is far better than many competitors.  Most importantly: *no one* seems to have any interest in offering a WiFi MCU at a price point competitive to Espressif.  I have had multiple designs that are able to exist for the sole reason that Espressif made a good and inexpensive WiFi module.

Like every design, compromises had to be made.  Do I wish Espressif was better?  Sure.  I can say the same about ST, Atmel, Silabs, NXP, Freescale, Microchip, TI, etc.  Every single vendor I've ever worked with has had some kind of thing that I wish worked better.  You wouldn't even believe some of the screwups I've had to deal with from some of those other companies (things like, "we do not employ anyone who understands how this broken component works").  That's just the reality of the job.

No design is perfect, no engineer is perfect, and no company is perfect.  It makes the day a bit less stressful when we can all try to keep that in mind.

fixajteknik, 2024-04-05:
> @readmodifywrite I fully agree with that. I believe that the entire RISCV ULP I2C driver should be rebuilt. I want to use it with sensiron sensors but it has 2bytes addressing and dont need sub register to read data.

Hdc1080 also