espressif/esp-idf #10433: idf.py fullclean shows ERROR in remove_managed_components

AxelLin, 2022-12-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2509-gcfef24863f

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3DevKit-M1

### Power Supply used.

USB

### What is the expected behavior?

"idf.py fullclean" should work without any error.

### What is the actual behavior?

After upgrade to v5.1-dev-2509-gcfef24863f, now everytime executing "idf.py fullclean" shows below error:
I didn't change anythin in managed_components folder, and this error does not happen before upgrading esp-idf.

$ idf.py fullclean
Executing action: fullclean
Executing action: remove_managed_components
ERROR:
        Some components (joltwallet__littlefs) in the "managed_components" directory were modified         on the disk since the last run of the CMake. Content of this directory is managed automatically.

        If you want to keep the changes, you can move the directory with the component to the "components"         directory of your project.

        I.E. for "joltwallet__littlefs" run:
        mv /home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs /home/axel/esp/esp-idf-dev/apps/myproj/components/joltwallet__littlefs

        Or, if you want to discard the changes remove the ".component_hash" file from the component's directory.

        I.E. for "joltwallet__littlefs" run:
        rm /home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs/.component_hash

### Steps to reproduce.

idf.py build ; idf.py fullclean

kumekay, 2022-12-23:
Hi @AxelLin could you please provide more details about your environment to help us reproduce this problem:

- Linux Distro with version
- File system type
- Content of dependencies.lock

Do you experience the problem only with this component or with other too?

AxelLin, 2022-12-23:
```
$ lsb_release -a
LSB Version:	core-11.1.0ubuntu4-noarch:printing-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

filesystem: ext4

$ cat dependencies.lock 
dependencies:
  idf:
    component_hash: null
    source:
      type: idf
    version: 5.1.0
  joltwallet/littlefs:
    component_hash: 2fa0ef8823451b00cec4eec1a18f810e9267b1f588513a64526be456c8a9d59a
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 1.5.1
manifest_hash: c14ee59777e03272dc787e6a72f0e5ccf67cf0c3820f182730fbe2e28d652138
target: esp32c3
version: 1.0.0
```

AxelLin, 2022-12-23:
> Do you experience the problem only with this component or with other too?

I only use littlefs, so I'm not sure if other components have issue or not.
```
$ cat main/idf_component.yml
## IDF Component Manager Manifest File
dependencies:
  joltwallet/littlefs: "^1.4.0"
  ## Required IDF version
  idf:
    version: ">=4.1.0"
  # # Put list of dependencies here
  # # For components maintained by Espressif:
  # component: "~1.0.0"
  # # For 3rd party components:
  # username/component: ">=1.0.0,<2.0.0"
  # username2/component2:
  #   version: "~1.0.0"
  #   # For transient dependencies `public` flag can be set.
  #   # `public` flag doesn't have an effect dependencies of the `main` component.
  #   # All dependencies of `main` are public by default.
  #   public: true
```

kumekay, 2022-12-23:
@AxelLin thank you so much, 

Could you please also share the content of `managed_components/joltwallet__littlefs/.component_hash` file?

AxelLin, 2022-12-23:
$ cat managed_components/joltwallet__littlefs/.component_hash
2fa0ef8823451b00cec4eec1a18f810e9267b1f588513a64526be456c8a9d59a

BTW, everytime I execute "idf.py menuconfig" it will show message to ask
deleting managed_components/joltwallet__littlefs/.component_hash.

I delete it, then execute "idf.py menuconfig". It will generate exactly the same content again:
$ cat managed_components/joltwallet__littlefs/.component_hash
2fa0ef8823451b00cec4eec1a18f810e9267b1f588513a64526be456c8a9d59a

AxelLin, 2022-12-23:
It was working with v5.1-dev-2356-gedd815af2e, and fails after upgrade to v5.1-dev-2495-ge5926d1b1b.
BTW, ./install.sh updates below packages, I'm not sure if it's related or not. just FYI.

  Downloading https://dl.espressif.com/pypi/idf-component-manager/idf_component_manager-1.2.1-py3-none-any.whl (116 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 116.1/116.1 kB 2.7 MB/s eta 0:00:00
  Downloading https://dl.espressif.com/pypi/esptool/esptool-4.5.dev0-py3-none-any.whl (310 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 310.2/310.2 kB 4.8 MB/s eta 0:00:00  
  Downloading https://dl.espressif.com/pypi/packaging/packaging-22.0-py3-none-any.whl (42 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 42.6/42.6 kB 4.8 MB/s eta 0:00:00

kumekay, 2022-12-24:
@AxelLin Thank you so much. We were able to reproduce the problem.

While a proper solution is on its way, could you please stick to an older version of the component manager by running 

```
pip install idf-component-manager==1.1.4
```

in your ESP-IDF environment (after `. $IDF_PATH/export.sh`)?

We will move this issue to http://github.com/espressif/idf-component-manager project.