espressif/esp-idf #10434: Dynamic light sleep strange behaviour

daniSi90, 2022-12-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3 WROOM

### Power Supply used.

External 3.3V

### What is the expected behavior?

The goal was to wake up the MCU from light sleep dynamically. In a small example the sleep time increases each sleep time for 5s. The sleep time we can monitor with a pin which should be low during sleep and high when awake (each time 2s). The code is as followed:
```
gpio_set_level(DEVICE_DEBUG_PIN, 1);
vTaskDelay(pdMS_TO_TICKS(5000));
uint64_t testTim = 5 * 1000000;
printf("START SLEEP TIME: %lld\n", testTim);
gpio_set_level(DEVICE_DEBUG_PIN, 0);
vTaskDelay(pdMS_TO_TICKS(5)); /* Small delay just to make sure pin goes low before sleep */
esp_timer_start_periodic
while(1){
    esp_sleep_enable_timer_wakeup(testTim);
    esp_light_sleep_start();
 
    gpio_set_direction(DEVICE_DEBUG_PIN, GPIO_MODE_OUTPUT);
    gpio_set_level(DEVICE_DEBUG_PIN, 1);
    vTaskDelay(pdMS_TO_TICKS(2000));
    testTim += 5000000;
    gpio_set_level(DEVICE_DEBUG_PIN, 0);
    vTaskDelay(pdMS_TO_TICKS(5)); /* Small delay just to make sure pin goes low before sleep */
}
```

### What is the actual behavior?

The actual behaviour was monitored with a logic analyser where the output is as followed:

1. Sleep (t_s1) is correct 
2. Sleep (t_s2) is skipped
3. Sleep (t_s3) is correct
4. Sleep (t_s4) is wrong (t_s4 = 2.5s)
5. Sleep (t_s5) is equal to t_s1 + t_s4
6. Sleep (t_s5) is equal to t_s2 + t_s4
[image]

### Steps to reproduce.

copy/paste the previous code.

esp-lis, 2025-03-04:
Locally, based on the **release/v4.4** ([a9d0f221](https://github.com/espressif/esp-idf/commit/a9d0f22193acdf47a5a4db36832ae7068818962b)) version, I ran tests using the **$IDF_PATH/examples/system/light_sleep/** example, and it seems to work success. Below are my test code and waveform diagram:
```
/* Light sleep example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <time.h>
#include <sys/time.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_sleep.h"
#include "driver/gpio.h"

#define DEVICE_DEBUG_PIN    (GPIO_NUM_1)

void app_main(void)
{
    /* Configure the button GPIO as output */
    gpio_config_t config = {
            .pin_bit_mask = BIT64(GPIO_NUM_1),
            .mode = GPIO_MODE_OUTPUT
    };
    ESP_ERROR_CHECK(gpio_config(&config));

    gpio_set_level(DEVICE_DEBUG_PIN, 1);

    vTaskDelay(pdMS_TO_TICKS(5000));
    uint64_t testTim = 5 * 1000000;
    printf("START SLEEP TIME: %lld\n", testTim);

    gpio_set_level(DEVICE_DEBUG_PIN, 0);
    vTaskDelay(pdMS_TO_TICKS(5)); /* Small delay just to make sure pin goes low before sleep */

    while(1){
        esp_sleep_enable_timer_wakeup(testTim);
        esp_light_sleep_start();
     
        gpio_set_direction(DEVICE_DEBUG_PIN, GPIO_MODE_OUTPUT);
        gpio_set_level(DEVICE_DEBUG_PIN, 1);

        vTaskDelay(pdMS_TO_TICKS(2000));
        testTim += 5000000;

        gpio_set_level(DEVICE_DEBUG_PIN, 0);
        vTaskDelay(pdMS_TO_TICKS(5)); /* Small delay just to make sure pin goes low before sleep */
    }
}
```
[image]

Indastri, 2025-04-23:
Since there was no response from the user - closing this issue. If you will encounter any further problems feel free to reopen this or create a new issue.

Cheers!

closed 2025-04-23 by Indastri