espressif/esp-idf #10437: esp_https_ota() crashes in spi_flash_disable_interrupts_caches_and_other_cpu when stack is in external memory

bdpdx, 2022-12-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-D0WD custom board

### Power Supply used.

External 5V

### What is the expected behavior?

esp_https_ota() should not crash

### What is the actual behavior?

when my app is built with:

```
CONFIG_SPIRAM=y
CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y

```

and I allocate my task in external memory (4MB mapped into malloc()) like so:

```
        uint32_t stackSize = 16 * 1024;
        if ((stack = (StackType_t *) heap_caps_malloc(stackSize, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT | MALLOC_CAP_32BIT)) == NULL) err = ENOMEM;
        if (!err && (taskBuffer = (StaticTask_t *) heap_caps_malloc(sizeof(StaticTask_t), MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT | MALLOC_CAP_32BIT)) == NULL) err = ENOMEM;
        if (!err && (task = xTaskCreateStaticPinnedToCore((void (*)(void *)) taskEntry, name, stackSize, this, priority, stack, taskBuffer, coreID)) == NULL) err = ESP_FAIL;

```

If I call `esp_https_ota()` I get a crash in `spi_flash_disable_interrupts_caches_and_other_cpu()`:

```
Backtrace: 0x400824fd:0x3f825170 0x4008fd01:0x3f825190 0x40096fc9:0x3f8251b0 0x40086c62:0x3f8252d0 0x40086fa7:0x3f825300 0x400fe18a:0x3f825320 0x400fd972:0x3f825340 0x400fdd6a:0x3f825360 0x40128d4f:0x3f825380 0x4012917f:0x3f8253c0 0x400e0eb4:0x3f8253f0 0x400dfd45:0x3f8254d0 0x400e002d:0x3f8254f0 0x400e0069:0x3f825510 0x400e7bc7:0x3f825530 0x400e7c15:0x3f825570 0x4009321d:0x3f825590
0x400824fd: panic_abort at /Users/brian/.espressif/esp-idf/components/esp_system/panic.c:412

0x4008fd01: esp_system_abort at /Users/brian/.espressif/esp-idf/components/esp_system/esp_system.c:135

0x40096fc9: __assert_func at /Users/brian/.espressif/esp-idf/components/newlib/assert.c:78

0x40086c62: spi_flash_disable_interrupts_caches_and_other_cpu at /Users/brian/.espressif/esp-idf/components/spi_flash/cache_utils.c:152 (discriminator 1)

0x40086fa7: spi_flash_protected_read_mmu_entry at /Users/brian/.espressif/esp-idf/components/spi_flash/flash_mmap.c:323

0x400fe18a: spi_flash_cache2phys at /Users/brian/.espressif/esp-idf/components/spi_flash/flash_mmap.c:395

0x400fd972: esp_ota_get_running_partition at /Users/brian/.espressif/esp-idf/components/app_update/esp_ota_ops.c:558

0x400fdd6a: esp_ota_get_next_update_partition at /Users/brian/.espressif/esp-idf/components/app_update/esp_ota_ops.c:586

0x40128d4f: esp_https_ota_begin at /Users/brian/.espressif/esp-idf/components/esp_https_ota/src/esp_https_ota.c:309

0x4012917f: esp_https_ota at /Users/brian/.espressif/esp-idf/components/esp_https_ota/src/esp_https_ota.c:676
```

If I allocate the task stack in internal memory then esp_https_ota() does not crash and completes successfully.

### Steps to reproduce.

```
        #include "esp_crt_bundle.h"
        #include "esp_https_ota.h"

        esp_http_client_config_t httpConfig = {};
        esp_https_ota_config_t otaConfig = {};

        httpConfig.crt_bundle_attach = &esp_crt_bundle_attach;
        httpConfig.url = firmwareURL;
        httpConfig.user_data = this;

        otaConfig.http_config = &httpConfig;

        esp_https_ota(&otaConfig);
```

kayzen9, 2022-12-26:
Did you check https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#restrictions? For low internal memory try: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#

i use `CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC` with spiram

igrr, 2022-12-26:
Unfortunately it's not currently possible to perform SPI Flash operations in tasks with stack in external RAM, unless `CONFIG_SPIRAM_FETCH_INSTRUCTIONS` and `CONFIG_SPIRAM_RODATA` are also enabled. These options are not supported on ESP32, though.

mahavirj, 2023-01-05:
Closing per the comments and pointers shared above. Please feel free to re-open if you face any further issues.

closed 2023-01-05 by mahavirj