espressif/esp-idf #10438: Float division in ULP risc V

schafon, 2022-12-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2066-g7869f4e151-dirty

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board esp32 s2

### Power Supply used.

USB

### What is the expected behavior?

Print the correct value from ulp memory

### What is the actual behavior?

printing incorrect value

### Steps to reproduce.

1. Create a program that uses ULP risc V
2. divide number by another number for example test = -3076856.0/7864320.0;
3. wake the main core
4. print the variable "test" (printf("ulp_test var = %e\n",ulp_test);)

### More Information.

Doing floats division in ULP doesn't seems to work, when I try to print the value in ULP using ulp_riscv_print_hex(test);
I get 0, and when trying to print it after waking the main core I get 1.749542e+00 instead of -0.3912424723

schafon, 2022-12-24:
@sudeep-mohanty - Your field of expertise

sudeep-mohanty, 2022-12-27:
@schafon The float division does goes through as expected on the ULP core. I needed to do some convoluted printing to get the result correctly displayed -

`printf("ulp_test var = %e\n", *(float *)(&ulp_test));` gives the result `-3.912425e-01` as expected.

Please try out and let us know if this works. Thanks!

closed 2022-12-30 by schafon