espressif/esp-idf #10439: Existing app, recompile for esp32s3, doesn't pick up a component dependency with 4.4.3

dannybackx, 2022-12-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Full build to complete.

### What is the actual behavior?

Doesn't pick up on esp_https_server dependency, independent of where it is in the REQUIRES list (see below).

### Steps to reproduce.

Copy complete application tree
idf.py set-target esp32s3
idf.py clean
idf.py build

### Build or installation Logs.

```plain
In file included from /home/danny/src/esp-idf/s3/espalarm/main/App.h:75,
                 from /home/danny/src/esp-idf/s3/espalarm/main/Keypad.cpp:23:/home/danny/src/esp-idf/s3/espalarm/main/WebServer.h:30:10: fatal error: esp_https_server.h: No such file or directory
 #include <esp_https_server.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### More Information.

hp: {121} cat main/CMakeLists.txt
file(GLOB SOURCES *.cpp *.c
        ../libraries/rfid/*.cpp
        ../libraries/qr/*.c
        ../libraries/RFM69/*.cpp
        ../libraries/Adafruit_BME280/*.cpp
        ../libraries/Adafruit_MCP9808/*.cpp
        ../libraries/Adafruit_AHTX0/*.cpp
        ../libraries/Adafruit_BusIO/*.cpp
        ../libraries/Adafruit_Sensor/*.cpp
        )

idf_component_register(SRCS "${SOURCES}"
  INCLUDE_DIRS "." ../libraries ../components ../libraries/rfid ../libraries/qr
    ../libraries/RFM69 ../libraries/Adafruit_BME280 ../libraries/Adafruit_MCP9808
    ../libraries/Adafruit_AHTX0 ../libraries/Adafruit_BusIO ../libraries/Adafruit_Sensor
    ../components/arduino ${BUILD_DIR}
  REQUIRES arduino esp_littlefs acmeclient TFT_eSPI arduinojson app_update mqtt
    wpa_supplicant firebase rcswitch ftpclient esp_https_server)

add_custom_command(OUTPUT ${BUILD_DIR}/build.h
        COMMAND ${COMPONENT_DIR}/generate_build_date ${BUILD_DIR}/build.h
        DEPENDS ${SOURCES}
        VERBATIM)

add_custom_target(buildh DEPENDS ${BUILD_DIR}/build.h)
add_dependencies(${COMPONENT_LIB} buildh)

igrr, 2022-12-25:
Do you have CONFIG_ESP_HTTPS_SERVER_ENABLE=y in your sdkconfig.defaults file?
`sdkconfig` file is generated from scratch when you run `idf.py set-target`, so it's recommended to keep your config changes in sdkconfig.defaults.

dannybackx, 2022-12-26:
Thanks, that does explain it.

closed 2022-12-26 by dannybackx