espressif/esp-idf #10440: idf.py build fails - cannot find 'In archive' in libxtensa.a dump

Coobik, 2022-12-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2509-gcfef24863f

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect "idf.py build" to run without errors

### What is the actual behavior?

cloned esp-idf and followed the manual
created a dummy project
tried to build it with "idf.py build"
- build fails on esp-idf/tools/ldgen/ldgen/entity.py", in add_sections_info with error: Expected 'In archive', found ''

### Steps to reproduce.

run "idf.py build" in a project directory

### Build or installation Logs.

```plain
Traceback (most recent call last):
  File "/Users/USER/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/pyparsing/core.py", line 817, in _parseNoCache
    loc, tokens = self.parseImpl(instring, pre_loc, doActions)
  File "/Users/USER/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/pyparsing/core.py", line 2337, in parseImpl
    if instring[loc] == self.firstMatchChar and instring.startswith(
IndexError: string index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/USER/projects/esp/esp-idf/tools/ldgen/ldgen/entity.py", line 157, in add_sections_info
    results = parser.parseString(first_line, parseAll=True)
  File "/Users/USER/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/pyparsing/core.py", line 1141, in parse_string
    raise exc.with_traceback(None)
pyparsing.exceptions.ParseException: Expected 'In archive', found end of text  (at char 1), (line:2, col:1)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/USER/projects/esp/esp-idf/tools/ldgen/ldgen.py", line 175, in <module>
    main()
  File "/Users/USER/projects/esp/esp-idf/tools/ldgen/ldgen.py", line 133, in main
    sections_infos.add_sections_info(dump)
  File "/Users/USER/projects/esp/esp-idf/tools/ldgen/ldgen/entity.py", line 159, in add_sections_info
    raise ParseException('Parsing sections info for library ' + sections_info_dump.name + ' failed. ' + p.msg)
pyparsing.exceptions.ParseException: Parsing sections info for library /my-project/build/esp-idf/xtensa/libxtensa.a failed. Expected 'In archive'  (at char 0), (line:1, col:1)
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

[... 160 lines cut ...]

Sections:
Idx Name                Size     VMA      Type
  0                     00000000 00000000
  1 .text               00000033 00000000 TEXT
  2 .data               00000400 00000000 DATA
  3 .rela.data          00000900 00000000
  4 .bss                00000000 00000000 BSS
  5 .xtensa.info        00000038 00000000
  6 .debug_line         000000e6 00000000
  7 .rela.debug_line    000000f0 00000000
  8 .debug_info         00000023 00000000
  9 .rela.debug_info    00000048 00000000
 10 .debug_abbrev       00000014 00000000
 11 .debug_aranges      00000020 00000000
 12 .rela.debug_aranges 00000024 00000000
 13 .debug_str          00000083 00000000
 14 .xt.prop            00000054 00000000
 15 .rela.xt.prop       00000054 00000000
 16 .symtab             00000550 00000000
 17 .strtab             0000034a 00000000
 18 .shstrtab           0000009b 00000000

--
there is no trace of the words "In archive" in the dump, neither in the first line (it is blank) nor in the whole dump

Coobik, 2022-12-25:
run command manually to see there are no words "In archive" in output
/usr/bin/objdump -h /my-project-path/build/esp-idf/xtensa/libxtensa.a

Coobik, 2022-12-25:
it looks like the problem is the difference in behaviour between macos objdump (/usr/bin/objdump) and the one in linux. they have very different output.

on macos, we can install binutils with homebrew: brew install binutils
then linux version of objdump is available under /usr/local/opt/binutils/bin
so we can run it
/usr/local/opt/binutils/bin/objdump -h /my-prohect-path/build/esp-idf/xtensa/libxtensa.a
and observe the output - the first line is indeed:
In archive /my-project-path/build/esp-idf/xtensa/libxtensa.a:

so the tool chain for macos should include linux version of objdump to get the "In archive" line

PS
i don't understand why do we need objdump to read this line because it's just filepath which is already known

Coobik, 2022-12-25:
for some (yet unknown) reason
/Users/USER/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump
is not picked up by build scripts (even after . ./export.sh is run), so macos default objdump is used

Coobik, 2022-12-25:
tried to put --objdump "xtensa-esp32-elf-objdump" into /Users/USER/projects/esp/esp-idf/tools/cmake/ldgen.cmake explicitly (instead of using CMAKE_OBJDUMP which is always "/usr/bin/objdump" for unknown reason), so that ldgen step is passed,

then got the following error:
/Users/USER/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: warning: cannot find entry symbol call_start_cpu0; not setting start address

/Users/USER/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/USER/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libnosys.a(sbrk.o):(.literal+0x4): undefined reference to `end'

igrr, 2022-12-26:
Hi @Coobik, sorry that the project build didn't work as expected. To help us troubleshoot this, could you please attach the `CMakeCache.txt` file found inside `build` directory? Then remove the `build` directory, run `idf.py build` again, and attach the entire build log you get.

Coobik, 2022-12-28:
hi!
i don't know what is happening, but today i have tried with reverted ldgen.cmake,
i've put back --objdump "${CMAKE_OBJDUMP}" instead of explicit xtensa-esp32-elf-objdump
and it looks like now it picks up the correct objdump (xtensa-esp32-elf-objdump)
maybe that was a magic of reboot

but aside from that, it's not clear why objdump is required, because i've logged the results
and the lib paths and the first lines from the output of xtensa-esp32-elf-objdump are the same

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/xtensa/libxtensa.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/xtensa/libxtensa.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_ringbuf/libesp_ringbuf.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_ringbuf/libesp_ringbuf.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/efuse/libefuse.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/efuse/libefuse.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/driver/libdriver.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/driver/libdriver.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_pm/libesp_pm.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_pm/libesp_pm.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/mbedtls/libmbedtls.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/mbedtls/libmbedtls.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_app_format/libesp_app_format.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_app_format/libesp_app_format.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/bootloader_support/libbootloader_support.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/bootloader_support/libbootloader_support.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_partition/libesp_partition.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_partition/libesp_partition.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/app_update/libapp_update.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/app_update/libapp_update.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/spi_flash/libspi_flash.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/spi_flash/libspi_flash.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/pthread/libpthread.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/pthread/libpthread.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_system/libesp_system.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_system/libesp_system.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_rom/libesp_rom.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_rom/libesp_rom.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/hal/libhal.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/hal/libhal.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/log/liblog.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/log/liblog.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/heap/libheap.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/heap/libheap.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/soc/libsoc.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/soc/libsoc.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_hw_support/libesp_hw_support.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/esp_hw_support/libesp_hw_support.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/freertos/libfreertos.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/freertos/libfreertos.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/newlib/libnewlib.a
[... 96 lines cut ...]
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/sdmmc/libsdmmc.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/fatfs/libfatfs.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/fatfs/libfatfs.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/json/libjson.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/json/libjson.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/mqtt/libmqtt.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/mqtt/libmqtt.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/perfmon/libperfmon.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/perfmon/libperfmon.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/spiffs/libspiffs.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/spiffs/libspiffs.a:

[... cut ...] ulp.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/ulp/libulp.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/wifi_provisioning/libwifi_provisioning.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/wifi_provisioning/libwifi_provisioning.a:

lib:                   /Users/USER/projects/esp/friwave32n/build/esp-idf/main/libmain.a
first_line: In archive /Users/USER/projects/esp/friwave32n/build/esp-idf/main/libmain.a:

Coobik, 2022-12-28:
the second problem "cannot find entry symbol call_start_cpu0" and "undefined reference to `end'" - it is now also gone, but there came another one - undefined reference to `app_main'

I've looked around and it seems that now as defined in esp-idf/tools/cmake/project.cmake, the source directory of a project must be named "main".
but i have used platformio (vscode plugin, latest update) to create a project, and it has created source directory as "src".
after i renamed the directory to "main" this dummy project is now buildable by idf.py build

but not platformio itself, struggle continues...

Coobik, 2022-12-28:
so for platformio to be compatible with idf.py build and recognize main directory instead of src, we need to to put

[platformio]
src_dir = ./main

into platformio.ini

igrr, 2022-12-30:
Given that you have solved the issue on your side and that we were unable to reproduce it, I'll close the issue. If it happens again, please re-open it and attach the files I have mentioned in https://github.com/espressif/esp-idf/issues/10440#issuecomment-1364939905.

closed 2022-12-30 by igrr

Ou7law007, 2023-01-02:
Same error on Mac. Not using platformio

```
pyparsing.exceptions.ParseException: Parsing sections info for library /path/build/esp-idf/xtensa/libxtensa.a failed. Expected 'In archive'  (at char 0), (line:1, col:1)
```
The file is not empty

Coobik, 2023-01-02:
> Same error on Mac. Not using platformio
> 
> ```
> pyparsing.exceptions.ParseException: Parsing sections info for library /path/build/esp-idf/xtensa/libxtensa.a failed. Expected 'In archive'  (at char 0), (line:1, col:1)
> ```
> 
> The file is not empty

you need to make sure that xtensa-esp32-elf-objdump is used and not the system objdump.
also put all the paths that are exported by export.sh to your PATH