espressif/esp-idf #10441: GappServer Example build with undefined reference to `esp_ble_gap_ext_adv_set_params'

kensherman, 2022-12-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using the Gappserver project as a starting point and have added a simple feature that requires the function esp_ble_gap_ext_adv_set_params which is a member of [components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h]

I have an include in this in the file that uses this function. The project build fails with undefined reference to `esp_ble_gap_ext_adv_set_params'

This function is part of the api and it's not being found. How do I resolve

xiewenxiang, 2022-12-29:
Please enable `BT_BLE_50_FEATURES_SUPPORTED`

kensherman, 2023-01-10:
This was a configuration issue

closed 2023-01-10 by kensherman