espressif/esp-idf #10442: ESP32S3 RMT rmt_sync_reset introduces 100µs delay

higaski, 2022-12-26:
### Is your feature request related to a problem?

I'm not really sure if that's a feature request, a bug report or just merely a question. I've recently started to use the RMT peripheral on an ESP32S3 on two synchronized outputs. From my observations I suspect that `rmt_sync_reset` must be called between every transmission to keep both channels in sync otherwise they quickly diverge.

I've tried to queue multiple requests and call `rmt_sync_reset` inside the transmission complete callback but that doesn't seem to work.

Instead I now only start a single transmission and notify a high priority task to restart transmission as soon as possible. This works fine, but it introduces a 100µs delay before the transmission is actually restart.

Here's a signal I've captured which shows this behavior:
yellow = channel 0
cyan = channel 1
magenta = transmission complete callback
blue = notified high priority task
[image]

### Describe the solution you'd like.

Is there a technical reason why the transmission is delayed those 100µs?
Can I currently somehow lower that time? Apparently the task and data would be ready way sooner than that.

Rephrasing that question one could also ask:
"Is there a way to do continuous transmissions with multiple synchronized channels?"

suda-morris, 2022-12-27:
About the `rmt_sync_reset` function:

1. It is not allowed to run in the ISR context, the critical section used inside that function is not `ISR` version.
2. calling this function in any channel's tx done callback sounds not reasonable to me. A sync manager usually manages multiple channels, you can't reset it in channelA's callback while channelB might still not finished.

About the 100us delay in between each transaction:

The driver fetches the transaction descriptor from a queue, reconfigures the RMT channel, and start the RMT encoder.
All of this will consume negligible time (plus interrupt latency ~2us). If the RMT encoder is time consuming, this will cost more time I think.

> Is there a way to do continuous transmissions with multiple synchronized channels?

From my previous test, the sync can be lost after one transaction is finished. We have to resync them manually, that's why the `rmt_sync_reset` is provided.

higaski, 2022-12-27:
Thank you for your answer. Do you have any suggestions how to improve those times?
Any chance we might get an asynchronous way to pre-encode symbols in the future? E.g. by splitting `rmt_transmit` into `rmt_encode` and `rmt_start`.

suda-morris, 2022-12-28:
yes, you can pre-encode the symbols asynchronously, but can't write the symbols to the RMTMEM (a dedicated hardware memory) until the last transaction is done.  To copy the pre-encoded symbols into the RMTMEM, you need to use the RMT copy encoder. This encoder is simple and fast, just a wrapper of "memcpy".

higaski, 2022-12-30:
I can't figure out how to reliably re-trigger a transmission at a certain point. The delay itself isn't really an issue, I have up to ~480µs to prepare the next command in advance, but when it's time to transmit it it must be sent immediately and without any delay. The jitter introduced through the driver layer is a real problem for my use-case.

Any suggestions how I can work around the issue without hardware changes? (My current understanding is that a single output wouldn't suffer from the issue as I wouldn't have to call `rmt_sync_reset` in between.)

closed 2023-11-07 by suda-morris