espressif/esp-idf #10443: ESP32S3 USB host composite MSC

barbiani, 2022-12-26:
### Is your feature request related to a problem?

I am willing to bootload a rp2040 that is a usb msc device.

Loolks like its rom bootloader enumerates as 2 devices. One to reset the chip and other is a 128mb drive.

`I (171) cpu_start: App cpu up.
I (220) cpu_start: Pro cpu start user code
I (221) cpu_start: cpu freq: 160000000
I (221) cpu_start: Application information:
I (223) cpu_start: Project name:     usb-host
I (228) cpu_start: App version:      0fc0610-dirty
I (234) cpu_start: Compile time:     Dec 26 2022 15:06:05
I (240) cpu_start: ELF file SHA256:  4dba95f12a20d4b1...
I (246) cpu_start: ESP-IDF:          v4.4.2-388-g755ce1077d
I (252) heap_init: Initializing. RAM available for dynamic allocation:
I (259) heap_init: At 3FC94710 len 00055000 (340 KiB): D/IRAM
I (266) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (272) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (279) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (286) spi_flash: detected chip: generic
I (290) spi_flash: flash io: dio
I (294) sleep: Configure to isolate all GPIO pins in sleep state
I (300) sleep: Enable automatic switching of GPIO sleep configuration
I (308) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (328) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (328) gpio: GPIO[42]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (2468) main: Waiting for USB stick to be connected
I (2838) main: MSC device connected
I (2838) main: Device address: 1

E (2938) USB_MSC: COMPOSITE DEVICES UNSUPPORTED
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 1.10
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x2e8a
idProduct 0x3
bcdDevice 1.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 55
bNumInterfaces 2
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 500mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0x8
        bInterfaceSubClass 0x6
        bInterfaceProtocol 0x50
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x2    EP 2 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 1
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0xff
        bInterfaceSubClass 0x0
        bInterfaceProtocol 0x0
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x3    EP 3 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x84   EP 4 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
Device info:
         Capacity: 127 MB
         Sector size: 512
         Sector count: 262143
         PID: 0x   3
         VID: 0x2E8A
         iProduct: RP2 Boot
         iManufacturer: Raspberry Pi
         iSerialNumber: E0C912952D54
I (3078) main: Mounting
ESP_ERROR_CHECK failed: esp_err_t 0x1701 (ERROR) at 0x40379264
0x40379264: _esp_error_check_failed at C:/Users/esp/esp-idf/components/esp_system/esp_err.c:42

file: "./main/main.c" line 156
func: app_main
expression: msc_host_vfs_register(msc_device, "/usb", &mount_config, &vfs_handle)

abort() was called at PC 0x40379267 on core 0
0x40379267: _esp_error_check_failed at C:/Users/esp/esp-idf/components/esp_system/esp_err.c:43

Backtrace: 0x40375a4a:0x3fcf3d00 0x40379271:0x3fcf3d20 �ESP-ROM:esp32s3-20210327
0x40375a4a: panic_abort at C:/Users/esp/esp-idf/components/esp_system/panic.c:402

0x40379271: esp_system_abort at C:/Users/esp/esp-idf/components/esp_system/esp_system.c:128
`

### Describe the solution you'd like.

Any way to mount it. Maybe ignore the first endpoint.

### Describe alternatives you've considered.

None yet.

Alvin1Zhang, 2023-02-15:
@barbiani Thanks for reporting, would you please help share if any further updates for the ticket? Thanks.

barbiani, 2023-02-15:
@Alvin1Zhang 

I was not able to bootload the pico using its built-in msc bootloader.

Using the same hardware I was able to copy the UF2 file to a USB pendrive though (from a sd card connected to the esp32s3).

tore-espressif, 2023-04-14:
@barbiani we did more testing with composite USB devices, including the RPi Pico. Everything seems fine on v5.0 and later (we will backport to v4.4 too).

Could you please provide some logs or error behavior?

Please note that RPi Pico is not 'standard MSC' device. It won't retain the data written and the default files are read-only

tore-espressif, 2023-04-28:
Hello @barbiani ,
I modified the USB MSC host example in my fork: https://github.com/tore-espressif/esp-idf/commit/632e7d42d8af5432a0a05cd7b37c89cbf32834ff

The example flashes [blink example](https://github.com/raspberrypi/pico-examples/tree/master#first--examples) to RPi Pico, as soon as it is connected to ESP32-S3 usb port

I didn't encounter any issue related to our MSC implementation, so I'll close this issue. Feel free to reopen if you encounter any problem

closed 2023-04-28 by tore-espressif