espressif/esp-idf #10445: Characteristics and Bytes of Notification

meakashrao, 2022-12-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using the GATT_Server_Table example and have created 26 Characteristics in one service with each having notify, read, write properties.

MTU is modified to respective limit+3
The strange thing that I observe is when i have 12 characteristics, I am able to notify 200/300 bytes of data in each characteristics
but when I have 16 or 26, I am only able to go upto 30 bytes of data in each notification.

Please point to me how can or where can I find something about this the whole ESP IDF Example or libraries files

xiewenxiang, 2022-12-29:
Please check the maximum length of the attribute value. In particular, the characteristic that you mentioned where the exception occurred.

meakashrao, 2022-12-29:
I found the solution after reading you message, 
changed #define ESP_GATT_ATTR_HANDLE_MAX 100 

Thank you for your help

closed 2022-12-29 by meakashrao