espressif/esp-idf #10446: Tiny usb MSC not working with sdmmc

nikhil-robinson, 2022-12-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello there i am trying to integrate tinyUSB MSC with sdcard part as mount point but when i include " #include "esp_vfs_fat.h" " it is throwing all kind of bellow errors.

code:

```
#include <unistd.h>
#include <errno.h>
#include <dirent.h>
#include "esp_log.h"
#include "esp_console.h"
#include "tinyusb.h"
#include "class/msc/msc_device.h"
#include "tusb_msc_storage.h"
#include "driver/gpio.h"

#include "esp_vfs_fat.h"
#include "sdmmc_cmd.h"
#include "driver/sdmmc_host.h"
#include "esp_err.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_vfs.h"
#include "esp_spiffs.h"

static const char *TAG = "example_msc_main";
#define PROMPT_STR CONFIG_IDF_TARGET

/********* TinyUSB MSC callbacks ***************/

/** SCSI ASC/ASCQ codes. **/
/** User can add and use more codes as per the need of the application **/
#define SCSI_CODE_ASC_MEDIUM_NOT_PRESENT 0x3A /** SCSI ASC code for 'MEDIUM NOT PRESENT' **/
#define SCSI_CODE_ASC_INVALID_COMMAND_OPERATION_CODE 0x20 /** SCSI ASC code for 'INVALID COMMAND OPERATION CODE' **/
#define SCSI_CODE_ASCQ 0x00

static void _mount(void);
static void _unmount(void);
static bool is_eject = false;

// Invoked when received SCSI_CMD_INQUIRY
// Application fill vendor id, product id and revision with string up to 8, 16, 4 characters respectively
void tud_msc_inquiry_cb(uint8_t lun, uint8_t vendor_id[8], uint8_t product_id[16], uint8_t product_rev[4])
{
    (void) lun;
    ESP_LOGD(TAG, "tud_msc_inquiry_cb() invoked");

    const char vid[] = "TinyUSB";
    const char pid[] = "Flash Storage";
    const char rev[] = "0.1";

    memcpy(vendor_id, vid, strlen(vid));
    memcpy(product_id, pid, strlen(pid));
    memcpy(product_rev, rev, strlen(rev));
}

// Invoked when received Test Unit Ready command.
// return true allowing host to read/write this LUN e.g SD card inserted
bool tud_msc_test_unit_ready_cb(uint8_t lun)
{
    (void) lun;
[... 504 lines cut ...]
      |              ^~~~~~~~~~~~~~
/home/mr-venom/esp-5.0/esp-idf/components/log/include/esp_log.h:410:81: note: in definition of macro 'ESP_LOG_LEVEL'
  410 |         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
      |                                                                                 ^~~
/home/mr-venom/esp-5.0/esp-idf/components/log/include/esp_log.h:342:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  342 | #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
      |                                      ^~~~~~~~~~~~~~~~~~~
/home/mr-venom/sample_project/main/main.c:378:5: note: in expansion of macro 'ESP_LOGI'
  378 |     ESP_LOGI(TAG_PC_STORAGE, "Initializing SD card");
      |     ^~~~~~~~
/home/mr-venom/sample_project/main/main.c: In function 'app_main':
/home/mr-venom/sample_project/main/main.c:481:18: error: 'f_size' undeclared (first use in this function); did you mean 'wl_size'?
  481 |         .func = &f_size,
      |                  ^~~~~~
      |                  wl_size
/home/mr-venom/sample_project/main/main.c:489:18: error: 'f_unmount' undeclared (first use in this function); did you mean '_unmount'?
  489 |         .func = &f_unmount,
      |                  ^~~~~~~~~
      |                  _unmount
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/mr-venom/sample_project/build/log/idf_py_stderr_output_54370 and /home/mr-venom/sample_project/build/log/idf_py_stdout_output_54370

```

Idf version : ESP-IDF v5.1-dev-2509-gcfef24863f

esp-saurabhbansal, 2022-12-28:
"esp_vfs_fat.h" internally includes "ff.h".
ff.h has the APIs for FatFs module application.
For Ex - 
FRESULT f_read (FIL* fp, void* buff, UINT btr, UINT* br);			/* Read data from the file */
FRESULT f_write (FIL* fp, const void* buff, UINT btw, UINT* bw);	/* Write data to the file */
FRESULT f_mount (FATFS* fs, const TCHAR* path, BYTE opt);			/* Mount/Unmount a logical drive */
#define f_size(fp) ((fp)->obj.objsize)
#define f_unmount(path) f_mount(0, path, 0)

These names are also used in the above example code.
So, the compile errors are because of redefinition of these functions.

nikhil-robinson, 2022-12-28:
@esp-saurabhbansal i tried changing te names of the function and it compiled fine but now my problem is the usb is not at all detecting on my system like its completly dead or something can open serial terminal / not eve showing the usb port.

closed 2023-01-17 by nikhil-robinson

esp-saurabhbansal, 2023-01-21:
Integrated SD MMC Card as a storage media that is accessed by USB-host as a USB Mass Storage Device.

Solved by commit [490f9ebd50d9a70b3576a55ce0c7f47df6d2dbee](https://github.com/espressif/esp-idf/commit/490f9ebd50d9a70b3576a55ce0c7f47df6d2dbee)

esp-saurabhbansal, 2023-04-14:
Hello,

Just for your information, We have recently released USB Device MSC driver. Please check if it is useful to you. We highly appreciate the feedback.

The driver makes ESP based device recognizable by USB-hosts as a USB Mass Storage Device.
It either allows the embedded application to access the partition or Host PC accesses the partition over USB MSC. They can't be allowed to access the partition at the same time.

The driver supports storage media of two types:
1. SPI Flash
2. SD MMC Card

Driver can be found [here](https://github.com/espressif/idf-extra-components/blob/master/usb/esp_tinyusb/tusb_msc_storage.c).
Sample application can be found [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc).

Thanks.

texruska, 2023-04-21:
@esp-saurabhbansal is it possible to make that work with external SPI flash on different SPI interfaces (e.g. on SPI2 and SPI3)?

esp-saurabhbansal, 2023-05-10:
Hello,

Sorry for the delay in answering.
Yes, external SPI flash can also be used here.
External Flash has to be registered first using API `esp_partition_register_external`.
The example that shows the usage of this API is at `examples/storage/ext_flash_fatfs/main/ext_flash_fatfs_example_main.c`

Thanks

youngguyl, 2024-08-07:
@esp-saurabhbansal hello, sir.