espressif/esp-idf #10448: Random error wdt timeout

cosmoiler, 2022-12-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Code execution without failures

### What is the actual behavior?

The program periodically crashes with the wdt timeout error in a random place in the code. The code contains several FreeRTOS tasks. Very rarely there are no failures.

### Steps to reproduce.

The error manifests itself during the execution of FreeRTOS tasks, and in a random place, but after starting a certain task.

Code for debug log #2:

`bool DRVLed::run() {
    Task::wait();
    if (m_ptrState) m_ptrState->handle();
    return true;
}`

frt::Task<DRVLed, 2048u>::wait() :

`void wait()
{
	ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
}`

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4008dfe8  PS      : 0x00060034  A0      : 0x8008c482  A1      : 0x3ffafb80
0x4008dfe8: vListInsert at C:/Esp/esp-idf/components/freertos/list.c:166 (discriminator 1)

A2      : 0x3ffaf674  A3      : 0x3ffafcd0  A4      : 0x00000013  A5      : 0x00060023
A6      : 0x3ffb1408  A7      : 0x00000011  A8      : 0x3ffafcd0  A9      : 0x3ffafcd0
A10     : 0x00000001  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x00060023
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4008dfe5:0x3ffafb80 0x4008c47f:0x3ffafba0 0x4008b4de:0x3ffafbc0 0x4008345f:0x3ffafc00 0x4008e061:0x3ffafc20
0x4008dfe5: vListInsert at C:/Esp/esp-idf/components/freertos/list.c:166 (discriminator 1)

0x4008c47f: vTaskPlaceOnEventList at C:/Esp/esp-idf/components/freertos/tasks.c:3566

0x4008b4de: xQueueSemaphoreTake at C:/Esp/esp-idf/components/freertos/queue.c:1688

0x4008345f: ipc_task at C:/Esp/esp-idf/components/esp_ipc/src/esp_ipc.c:54

0x4008e061: vPortTaskWrapper at C:/Esp/esp-idf/components/freertos/port/xtensa/port.c:131

Core  1 register dump:
PC      : 0x4008e1c4  PS      : 0x00060034  A0      : 0x8008c2d8  A1      : 0x3ffc6770
0x4008e1c4: compare_and_set_native at C:/Esp/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at C:/Esp/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at C:/Esp/esp-idf/components/freertos/port/xtensa/port.c:288

A2      : 0x3ffb11c8  A3      : 0xffffffff  A4      : 0x8008a875  A5      : 0x3ffb1130  
A6      : 0x00000000  A7      : 0x3ffc6840  A8      : 0x0000cdcd  A9      : 0x0000abab
A10     : 0x00060023  A11     : 0x0000cdcd  A12     : 0xb33fffff  A13     : 0x00060023
A14     : 0x3ffb11b8  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4008e1c1:0x3ffc6770 0x4008c2d5:0x3ffc6790 0x4008a5e0:0x3ffc67b0 0x4008a592:0x00000003 |<-CORRUPTED
0x4008e1c1: cpu_ll_get_cycle_count at C:/Esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:45
 (inlined by) spinlock_acquire at C:/Esp/esp-idf/components/esp_hw_support/include/soc/spinlock.h:113
 (inlined by) xPortEnterCriticalTimeout at C:/Esp/esp-idf/components/freertos/port/xtensa/port.c:288

0x4008c2d5: vPortEnterCritical at C:/Esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
 (inlined by) vTaskSwitchContext at C:/Esp/esp-idf/components/freertos/tasks.c:3458

0x4008a5e0: _frxt_dispatch at C:/Esp/esp-idf/components/freertos/port/xtensa/portasm.S:436

0x4008a592: _frxt_int_exit at C:/Esp/esp-idf/components/freertos/port/xtensa/portasm.S:231

----------------------------------------------------------------------------------------------

#2:
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4008dfe2  PS      : 0x00060a34  A0      : 0x8008c482  A1      : 0x3ffafb80
0x4008dfe2: vListInsert at C:/Esp/esp-idf/components/freertos/list.c:166

A2      : 0x3ffaf674  A3      : 0x3ffafcd0  A4      : 0x00000011  A5      : 0x00060a23
A6      : 0x3ffc36e0  A7      : 0x00000000  A8      : 0x3ffafcd0  A9      : 0x3ffafcd0
A10     : 0x00000001  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x00060a23
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4008dfdf:0x3ffafb80 0x4008c47f:0x3ffafba0 0x4008b4de:0x3ffafbc0 0x4008345f:0x3ffafc00 0x4008e061:0x3ffafc20
0x4008dfdf: vListInsert at C:/Esp/esp-idf/components/freertos/list.c:166

0x4008c47f: vTaskPlaceOnEventList at C:/Esp/esp-idf/components/freertos/tasks.c:3566

0x4008b4de: xQueueSemaphoreTake at C:/Esp/esp-idf/components/freertos/queue.c:1688

0x4008345f: ipc_task at C:/Esp/esp-idf/components/esp_ipc/src/esp_ipc.c:54

0x4008e061: vPortTaskWrapper at C:/Esp/esp-idf/components/freertos/port/xtensa/port.c:131

Core  1 register dump:
PC      : 0x4008e1c9  PS      : 0x00060534  A0      : 0x8008d2ee  A1      : 0x3ffb9960
0x4008e1c9: compare_and_set_native at C:/Esp/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at C:/Esp/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at C:/Esp/esp-idf/components/freertos/port/xtensa/port.c:288

A2      : 0x3ffb11c8  A3      : 0xffffffff  A4      : 0x0022dd97  A5      : 0x00000001
A6      : 0x3ffb4180  A7      : 0x00000000  A8      : 0x0000abab  A9      : 0x0000abab  
A10     : 0x00060520  A11     : 0x0000cdcd  A12     : 0xb33fffff  A13     : 0x00060523
A14     : 0x3ffb4178  A15     : 0x3ffb3ca0  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4008e1c6:0x3ffb9960 0x4008d2eb:0x3ffb9980 0x400e72a6:0x3ffb99a0 0x400d807b:0x3ffb99c0 0x4008e061:0x3ffb99e0
0x4008e1c6: compare_and_set_native at C:/Esp/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
 (inlined by) spinlock_acquire at C:/Esp/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
 (inlined by) xPortEnterCriticalTimeout at C:/Esp/esp-idf/components/freertos/port/xtensa/port.c:288

0x4008d2eb: vPortEnterCritical at C:/Esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
 (inlined by) ulTaskGenericNotifyTake at C:/Esp/esp-idf/components/freertos/tasks.c:5372

0x400e72a6: frt::Task<DRVLed, 2048u>::wait() at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/main/Lib/Frt/src/frt.hpp:186
 (inlined by) DRVLed::run() at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/main/Drivers/Led/src/tskFlasher.cpp:192

0x400d807b: frt::Task<DRVLed, 2048u>::entryPoint(void*) at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/main/Lib/Frt/src/frt.hpp:255 (discriminator 1)

0x4008e061: vPortTaskWrapper at C:/Esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

0xjakob, 2022-12-28:
@cosmoiler Thanks for the issue report. However, I'm afraid the information is not enough to sufficiently analyze the problem. Hence, we have a few more questions:

Do you have a minimal reproducing code example? What is the configuration you are using (sdkconfig)? Could you enable stack overflow protection in menuconfig, in particular the watchpoint (`CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK`) and then try again?

cosmoiler, 2022-12-28:
Changed CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK to "y" - the crashes did not stop. Periodic error in a random place of the code.

Previously, the code worked (on esp-idf version 4.3). There were no global architecture changes except for the addition of an intermediary (class Mediator). There were no changes in the code regarding working with FreeRTOS. Now sometimes the code also works fine, but more often it crashes.

### sdkconfig

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
# CONFIG_BOOTLOADER_APP_ANTI_ROLLBACK is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
[... 2530 lines cut ...]
		bool popFromInterrupt(const T& item)
		{
			return xQueueReceiveFromISR(handle, &item, &higher_priority_task_woken_from_pop);
		}

		void finalizePopFromInterrupt() __attribute__((always_inline))
		{
			if (higher_priority_task_woken_from_pop) {
				detail::yieldFromIsr();
			}
		}

	private:
		QueueHandle_t handle;
		BaseType_t higher_priority_task_woken_from_push;
		BaseType_t higher_priority_task_woken_from_pop;
#if configSUPPORT_STATIC_ALLOCATION > 0
		uint8_t buffer[ITEMS * sizeof(T)];
		StaticQueue_t state;
#endif
	};

}

```

kriegste, 2022-12-28:
How is the signal strength? We had similar and rare WDT timeouts on very very poor WiFi connections in random parts of the IDF. We were not yet able to create a minimum code example.

0xjakob, 2022-12-29:
@cosmoiler

Thanks for the reproducing code. We'll take a closer look. Meanwhile, you can also follow @AxelLin's suggestion.

AxelLin, 2023-04-23:
> @cosmoiler
> 
> Thanks for the reproducing code. We'll take a closer look. 

@0xjakob 
So can you reproduce the issue? Do you have further finding?

0xjakob, 2024-08-13:
@cosmoiler Did you have a change to try your code on v5.0 or above and did the issue persist there?

0xjakob, 2024-09-05:
@HDLA-BG Thanks for letting us know. Would you be able to provide us with the complete code for a reproducer? The aforementioned code is missing several header files, so we can't even build it. Ideally, a zip file of the minimal reproducing example application, including sdkconfig and excluding the build folder, would be great.

HDLA-BG, 2024-09-11:
@0xjakob Sorry I commented a wrong issue. 
I deleted my comment above

AdityaHPatwardhan, 2024-09-17:
@cosmoiler Any update on the issue?
Are you able to reproduce this issue on the v5.x branch of esp-idf?

AdityaHPatwardhan, 2024-09-20:
Hi @cosmoiler I will go ahead and close this issue since there hasnt been any updates for much time.
If you still face the same issue then please re-open the issue and help provide a short reproducing code.
Thanks,
Aditya

closed 2024-09-20 by AdityaHPatwardhan