espressif/esp-idf #10450: [Feature request] Support for ETH switches with RMII (RTL8306MB for example)

DjordjeMandic, 2022-12-28:
### Is your feature request related to a problem?

"Daisy chaining" multiple projects on ethernet would be a good idea, its pretty easy to implement ethernet today with most modules and those chips from title are pretty cheap. For my project RTL8306MB would be perfect. Its outside, i might add ip cameras and etc... Ethernet Switch comes in handy everywhere... I also believe if this gets supported oficially, esphome will adopt it also and there will be modules from china pretty quick because there is a lot of these chips from old routers waiting to be recycled (yeah they do that with stuff from china, its made sometimes with recycled components).

### Describe the solution you'd like.

Implement basic support for RTL8306MB for start to make it possible to connect ESP32 directly to RMII port of switch ic like its implemented for emac-phy chips. Other ports can be autoconfigured for now. Later it would be nice to add option to control other ports or maybe do some basic vlan configuration if eth switch ic supports it.

### Describe alternatives you've considered.

I considered to use EMAC-PHY chip (RTL8201) and connect it directly to ETH switch chip (RTL8306MB) with just capacitors without magnetics. It would work in my opinion but there is increased number of components, more used space on pcb and etc... 

### Additional context.

[image]

https://datasheet.lcsc.com/lcsc/2204011630_Realtek-Semicon-RTL8306MB-CG_C2901899.pdf

https://www.lcsc.com/product-detail/Ethernet-ICs_Realtek-Semicon-RTL8306MB-CG_C2901899.html

https://www.realtek.com/en/products/communications-network-ics/item/rtl8306mb-cg

suda-morris, 2022-12-28:
FYI, there's already an Ethernet switch solution in https://github.com/espressif/esp-eth-drivers (KSZ8863)

DjordjeMandic, 2022-12-29:
Thanks, didn't knew that. I will probably do that. KSZ8863 is pretty advanced switch for some basic use (and a bit expensive). I left an idea, maybe someone will think about this in future, currently i do not have access to either KSZ8863 or RTL830xMB

closed 2022-12-29 by DjordjeMandic

kostaond, 2023-01-02:
@DjordjeMandic if you are looking for a cheap daisy-chaining solution, you can also consider realizing it by software bridge and SPI Ethernet modules, see https://github.com/espressif/esp-idf/tree/master/examples/network/bridge. Note that the bridge is currently tested with W5500 only and please also expect the performance will be lower than using IC's (like KSZ8863) specifically designed for switching.