espressif/esp-idf #10452: Incorrect rx_buffer when using 2 SPI buses on separate devices

LMBooth, 2022-12-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF version: 5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32-s3

### Power Supply used.

USB

### What is the expected behavior?

I have two spi devices each on completely separate pins, one device uses SPI2_HOST on the following pins:
#define PIN_NUM_MISO 5
#define PIN_NUM_MOSI 6
#define PIN_NUM_CLK  7
#define PIN_NUM_CS   4
The other device uses SPI3_HOST on the following pins:
#define PIN_CS                 11
#define PIN_CLK                8
#define PIN_SDI                9
#define PIN_SDO                12
I expect to be able to receive correct data from transactions calling the individual device handles created from the above pins.

Individually both devices work correctly, only when both spi devices are initialised.

### What is the actual behavior?

If i setup the device on SPI2_HOST and then make a call on SPI3_HOST i can see the correct data being sent on the MOSI and response on the MISO, but the data in my rx_buffer from spi_device_transmit is 0x00.

Tried swapping the SPI hosts around too, seems to be only the device attached to the following pins whcih isnt reading back the same data i see on the scope:
#define PIN_CS                 11
#define PIN_CLK                8
#define PIN_SDI                9
#define PIN_SDO                12

### Steps to reproduce.

Setup dev1:
spi_bus_config_t buscfg={
		.miso_io_num=PIN_SDO,
		.mosi_io_num=PIN_SDI,
		.sclk_io_num=PIN_CLK,
		.quadwp_io_num=-1,
		.quadhd_io_num=-1
	};
	ret=spi_bus_initialize(SPI3_HOST, &buscfg, SPI_DMA_CH_AUTO);
	spi_device_interface_config_t devcfg={
		.clock_speed_hz = 2000000, //SPI_MASTER_FREQ_8M,           //Clock out at 8 MHz
		.mode=3,                                //SPI mode 0
		.spics_io_num= ICM42688_PIN_CS,               //CS pin, -1 for none PIN_NUM_CS
		.command_bits = 0, 						// no command or address bytes/bits, all in tx_data and rx_data
		.address_bits =0,
		.dummy_bits = 0,
		.queue_size=2                         //We want to be able to queue 2 transactions at a time
		//.flags = SPI_DEVICE_NO_DUMMY	//SPI_DEVICE_HALFDUPLEX
	};
	ret=spi_bus_add_device(SPI3_HOST, &devcfg, &icm_spi1);

Setup dev2:
spi_bus_config_t buscfg={
		.miso_io_num=PIN_NUM_MISO,
		.mosi_io_num=PIN_NUM_MOSI,
		.sclk_io_num=PIN_NUM_CLK,
		.quadwp_io_num=-1,
		.quadhd_io_num=-1
	};
	ret=spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO);
	ESP_ERROR_CHECK(ret);
	spi_device_interface_config_t devcfg={
		.clock_speed_hz = 20000000, //SPI_MASTER_FREQ_8M,           //Clock out at 8 MHz
		.mode=0,                                //SPI mode 0
		.spics_io_num= PIN_NUM_CS,               //CS pin, -1 for none PIN_NUM_CS
		.command_bits = 0, 						// no command or address bytes/bits, all in tx_data and rx_data
		.address_bits =0,
		.dummy_bits = 0,
		.queue_size=2                         //We want to be able to queue 2 transactions at a time
		//.flags = SPI_DEVICE_NO_DUMMY	//SPI_DEVICE_HALFDUPLEX
	};
	ret=spi_bus_add_device(SPI2_HOST, &devcfg, &spi1);

### Debug Logs.

```plain
No crash occurs, just get incorrect data in rx_buffer.
```

### More Information.

Individually both devices work correctly, only when both spi devices are initialised.

wanckl, 2023-01-17:
Hi @LMBooth thank you for your detailed feedback.
It seems like a pin issue as you say only attached specific pin can meet the issue, or could you try to register your two devices to one same SPI_HOST(SPI2 or SPI3 only) to see if it works.
And as nearing the spring festival, I'm sorry we may deal with your issue later, thanks again for your continuously attention, and same wishes to you!

LMBooth, 2023-01-30:
Hi @wanckl , i hope you had a good spring festival.

I've tried assigning both devices to the same SPI_HOST (2 and 3) and am getting the following error after trying to initialize the second spi device:

E (3387) spi: spi_bus_initialize(758): SPI bus already initialized.
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x420089b2

after trying to call ret=spi_bus_initialize(SPI3_HOST, &buscfg, SPI_DMA_CH_AUTO);

This is due to the devices being on separate pins and the same SPI_HOST being called.

I'm thinking i can hardwire the MOSI, MISO and CLK together from both devices and only register one bus and add two devices to that bus, since both the spi devices reads are actually wanted on the same core whilst i use the other core for some data processing and wifi handling. Time for some fine pitch QFN and 0402 rework.

wanckl, 2023-01-31:
Hi @LMBooth 
Due to your description, we may have batter plan. If you want use only one core to deal with two SPI device, than this two device actually can NOT contacted simultaneity, You can simply add two device to one SPI_HOST and use same data pin, just separate by two CS_PIN like bellow: which can simplify your pin usage and hardware design.
[image]    

You can add 6 devices to SPI2_HOST and 3 devices to SPI3_HOST by esp-idf, each device can config different baudrate, mode, etc. then your code may like:   
```c
spi_bus_config(SPI2_HOST, &bus_config);
add_device(SPI2_HOST, dev1_config, dev1_handle);
add_device(SPI2_HOST, dev2_config, dev2_handle);
```
I'll continue to look at your previous issue, but you can change your route like this if you have no need to contact two device in real simultaneity.   
Waitting for your reply.

LMBooth, 2023-02-01:
> ```c
> ALFDUPLEX,
> ```

Hi again @wanckl ,

Thanks for your suggestions, i can get correct transactions for the first device transaction on SPI_HOST2, then again correct for the second devices first transaction on SPI_HOST3, then for the second transaction to the device on SPI_HOST2 has null data in the transaction but i can see correct data on the scope, yet subsequent calls to the device on SPI_HOST3 is returning correct data.

Very strange, i've got a spare board which i'm trying to rework, will post agian when i've made progress.

I've added my current spi initialise with read and write functions for reference too:
```
spi_device_handle_t spi3;
spi_device_handle_t spi2;
void InitialiseSPI(){
	esp_err_t ret;
	gpio_set_direction(DEV1_NUM_CLK, GPIO_MODE_OUTPUT);
	gpio_set_pull_mode(DEV1_NUM_CLK, GPIO_PULLUP_ONLY);
	spi_bus_config_t bus3cfg={
		.miso_io_num=DEV1_NUM_MISO,
		.mosi_io_num=DEV1_NUM_MOSI,
		.sclk_io_num=DEV1_NUM_CLK,
		.quadwp_io_num=-1,
		.quadhd_io_num=-1
	};
	ret=spi_bus_initialize(SPI3_HOST, &bus3cfg, SPI_DMA_CH_AUTO);
	ESP_ERROR_CHECK(ret);
	// Do AD7779 spi setup
	spi_device_interface_config_t dev3cfg={
		.clock_speed_hz = 20000000, //SPI_MASTER_FREQ_8M,           //Clock out at 8 MHz
		.mode=0,                                //SPI mode 0
		.spics_io_num= DEV1_NUM_CS,               //CS pin, -1 for none PIN_NUM_CS
		.command_bits = 0, 						// no command or address bytes/bits, all in tx_data and rx_data
		.address_bits =0,
		.dummy_bits = 0,
		.queue_size=2                         //We want to be able to queue 2 transactions at a time
		//.flags = SPI_DEVICE_NO_DUMMY	//SPI_DEVICE_HALFDUPLEX
	};
	ret=spi_bus_add_device(SPI3_HOST, &dev3cfg, &spi3);

	spi_bus_config_t bus2cfg={
        .miso_io_num=DEV2_NUM_MISO,
        .mosi_io_num=DEV2_NUM_MOSI,
        .sclk_io_num=DEV2_NUM_CLK,
        .quadwp_io_num=-1,
        .quadhd_io_num=-1
    };
    spi_bus_initialize(SPI2_HOST, &bus2cfg, SPI_DMA_CH_AUTO);
	spi_device_interface_config_t dev2cfg={
		.clock_speed_hz = 2000000, 
		.mode=0,                               
		.spics_io_num= DEV2_NUM_CS,              
		.command_bits = 0, 						
		.address_bits = 8,
		.dummy_bits = 0,
		.queue_size=2                        
	};
	ret=spi_bus_add_device(SPI2_HOST, &dev2cfg, &spi2);
	assert(ret == ESP_OK);
}

void spi1_writeReg(uint8_t reg, const uint8_t *pBuf, size_t size)
{
  spi_transaction_t trans = {
    .flags = 0,
    .addr = reg,
    .length = size*8, // may need to add 1*8 for command byte?
		.tx_buffer = pBuf,
	  };
	esp_err_t err = spi_device_transmit(spi2, &trans);
  if (err != ESP_OK){
    printf("spi error: 0x%x\r\n", err);
  }
}

uint8_t spi2_readReg(uint8_t reg, uint8_t* pBuf, size_t size)
{
  spi_transaction_t trans = {
		.flags = 0,
    .addr  = reg | 0x80,
		.length =  8 * (size),  // add 1*8 for command byte
		.rxlength = 8 * (size),  //length is in bits.
		.tx_buffer = NULL,
		.rx_buffer = pBuf //pBufFI
	  };
	esp_err_t err = spi_device_transmit(spi2, &trans);
  if (err != ESP_OK){
    printf("spi error: 0x%x\r\n", err);
  }
	return err; //*(uint8_t*)trans.rx_data;
}

int writereadfromspi3(uint32_t readlength, spi_device_handle_t spi){
	spi_transaction_t trans = {
		.flags = 0,
		.length =  8 * readlength,  // may need to add 1*8 for command byte?
		//.txlength = 8 * 1,  //length is in bits.
		.rxlength = 8 * (readlength-1),  //length is in bits.
		.tx_buffer = spiTxBuf,
		.rx_buffer = spiRxBuf
	  };
	esp_err_t err = spi_device_transmit(spi, &trans);
	return err;
}
```

wanckl, 2023-02-02:
Could you past a scope graph on the time spi2 receive null data, and if any error reported during code running?

LMBooth, 2023-02-16:
Sorry @wanckl I only have a rubbish desktop oscilloscope that i can probe one channel at a time and can't get any good photos of whats happening.

I am printing data returned through a terminal though.

I'll try to explain my terminal output in more detail via my order of operation:
1. I can read id reg from one device on spi2 with its own set of pins - i can read multiple times in a row from this device
2.  When i use the devices on spi3 i get the correct register values back when i read them all, 
3. Then when i go back to the device on spi2 i can invalid reads for the device id (only value 0) where as register reads from SPI3 are still valid adn correct.