espressif/esp-idf #10453: Issue when connecting at 20Mhz - ESP32-S3

grbd, 2022-12-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
I'm raising this here in case what I'm seeing is a software issue
It's a bit difficult to tell so I figured I'd check here based on what I'm seeing

I have a ESP32-S3 ProS3

  * https://esp32s3.com/pros3.html

Recently I've been experimenting with different versions circuitpython, this includes the latest one
Which is using a fairly recent (within the last couple of weeks) version of the esp-idf

  * https://github.com/adafruit/circuitpython
  * https://github.com/adafruit/circuitpython/tree/main/ports/espressif
  * https://github.com/adafruit/esp-idf/tree/8d0f1abce769ad4a212c1bb0337999cacd7ee83d

One of the issues I've been seeing is 205 connection error's when connecting to a wifi station, or repeater
on the lower channel numbers

  * https://github.com/adafruit/circuitpython/issues/7383

What I discovered is that if the device tries to use 20Mhz of bandwidth it will error out with 205 on the lower channel numbers
But when it's using 40Mhz it seems to work fine.
Changing the "Channel bandwidth" setting on the wifi router to

  * 20/40Mhz - 205 connection errors
  * 20Mhz - 205 connection errors
  * 40Mhz - works fine

Since forcing te device to use 40Mhz makes it work with lower channel numbers
I'm not sure if this is just wifi interference, or maybe a software issue when using 20Mhz?
Perhaps it's just a hardware issue with the board with respect to the antenna
The 205 CONNECTION_FAIL error isn't much to go at unfortunatley

  * https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-ht20-40

mhdong, 2024-03-22:
Hi @Hecatron 
The ESP32-S3 has a silicon bug when USB is being used with the radio on - there is a 2-3dB loss on lower channels.

Hecatron, 2024-03-22:
Thats interesting

Sherry616, 2024-05-15:
Thanks for reporting, we will close this issue for now. 
If there are further optimizations or you have other questions, feel free to reopen.
Thanks for using our Espressif product!

closed 2024-05-15 by Sherry616