espressif/esp-idf #10455: ble-mesh connection issue when more then 150+ node provisioned

hipal31, 2022-12-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32C3

### Power Supply used.

External 3.3V

### What is the expected behavior?

I have developed a light node using BLE-MESH. When I provision more then 150+ devices in a single network, then I am facing connection issue from mobile app. In smaller network, it's working fine but problem occurs with large number of devices in a network.  I have used nRF Mesh android sdk for provisioning, connecting and operation of devices.

Also please let me know of parameters which are important for ble-mesh connection.

### What is the actual behavior?

Whenever I try to connect to BLE-MESH network from mobile app then it must be connected immediately.

### Steps to reproduce.

1. Provision 150 or more devices in a single network.
2. Try to connect in network
3. Operate in network
4. Disconnect network and try re-connecting

hipal31, 2023-01-04:
@JiaLWang please look into the issue.

JiaLWang, 2023-01-04:
Hi @hipal31 ,

Are the devices on the network sending messages frequently?

hipal31, 2023-01-04:
@JiaLWang  No.
Even after provisioning the devices in network, this issue occurs.

JiaLWang, 2023-01-04:
Will our own example(eg: ble_mesh_node example) have this problem? If it exists, we can try to reproduce it locally.

hipal31, 2023-01-04:
@JiaLWang Yes, please try on that. If you want then I can provide firmware binaries and App for you to reproduce the issue.

JiaLWang, 2023-01-04:
Firmware binaries are inconvenient to add log and debug problems.

hipal31, 2023-01-04:
You can try on example codes first. Please let me know if you are facing issue in re-generating the issue. I can give you my sdkconfig and ble-mesh configuration done in code.

hipal31, 2023-01-09:
@JiaLWang Any progress?

hipal31, 2024-06-13:
@JiaLWang  the issue is still exist. Can you help to solve?

JiaLWang, 2024-06-13:
Hi @hipal31 ,
Sorry for the long delay in getting back to you and many thanks for being patient. 

I will arrange colleagues to support the issue ASAP.