espressif/esp-idf #10456: [linux] Allow custom component names

higaski, 2022-12-28:
### Is your feature request related to a problem?

For some reason the build system currently does not allow the `COMPONENTS` variable for a linux build to be anything other than `main`.

E.g. here's the CMakeLists.txt of the [hello_world_linux_compatible](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/linux_compatible/hello_world_linux_compatible) 

```cmake
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(COMPONENTS main)
project(hello_world)
```

Trying to set COMPONENTS to anything else (e.g. "tests") results in a CMake error
```sh
CMake Error at /home/vinci/esp/esp-idf/tools/cmake/build.cmake:542 (message):
  ERROR: Cannot parse components list file.
```

### Describe the solution you'd like.

I'd like to give a component on linux the same arbitrary name I can when using any other esp* target.

igrr, 2022-12-28:
> kconfiglib.KconfigError: /home/vinci/esp/esp-idf/components/bt/Kconfig:68: '/home/vinci/esp/esp-idf/components/bt/controller/linux/Kconfig.in' not found (in 'source "$IDF_PATH/components/bt/controller/$IDF_TARGET/Kconfig.in"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.

In this case it seems that one of your components has a dependency on `bt` component, which is not yet supported on `linux` target. (Sorry, the error is not very clear, but it boils down to "no one has tried building `bt` component for `linux` yet".)

> CMake Error at /home/vinci/esp/esp-idf/tools/cmake/build.cmake:542 (message):
>  ERROR: Cannot parse components list file.

Could you please share more details about this one? Did you change the `hello_world_linux_compatible` project's CMakeLists.txt to have `set(COMPONENTS tests)`  and then rename `main` directory to `tests`, or anything else?

higaski, 2022-12-28:
Sorry for the confusion, I'm trying various things since 2h and I'm losing track myself.

So, let's try that again, clean plate.

I've copied the [hello_world_linux_compatible](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/linux_compatible/hello_world_linux_compatible) as starting point. Building the unmodified example works. 

The first thing I then tried is to rename the main folder and the COMPONENTS variable to anything else. This produces the error I mentioned at the beginning.
```sh
CMake Error at /home/vinci/esp/esp-idf/tools/cmake/build.cmake:542 (message):
  ERROR: Cannot parse components list file.
```  

Using the `EXTRA_COMPONENT_DIRS` variable instead of the `COMPONENTS` variable works, but only on esp* targets.
```cmake
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(EXTRA_COMPONENT_DIRS tests)
project(hello_world)
```

So far so good. Now I've tried to switch to the linux target. So I've run the command mentioned in the README.md
```cmake
idf.py --preview set-target linux
```
And this doesn't work for some reason (see Python traceback above).

0xjakob, 2023-01-04:
@higaski Just for my understanding: why do you need to rename the `main` component?

Regarding
> With a (defaulted) main component the build produces a config.env file which by default includes the bluetooth component 

Did you set the `COMPONENTS` to main in the `CMakeLists.txt` file of your project? if not, then the build errors are (currently) expected as the Linux target is still under development. By far not all components can be built on Linux.

higaski, 2023-01-04:
> @higaski Just for my understanding: why do you need to rename the `main` component?

Strictly speaking I wouldn't have to, but it's company policy to put tests under ./tests
My employer currently has 2 ESP32 products in the pipeline, it would be nice if we could keep consistency with existing projects.

> Regarding
>
>
> Did you set the `COMPONENTS` to main in the `CMakeLists.txt` file of your project? if not, then the build errors are (currently) expected as the Linux target is still under development. [...]

I did not no, I would like 'linux' targets to build under in different folder than main.

0xjakob, 2023-01-05:
Hi @higaski
Seems like my answer was misleading, at best. 

So yes, it is possible to rename the main component, as stated in the documentation. And it actually works with the [linux hello world example](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/linux_compatible/hello_world_linux_compatible), I just tried to run it with the `linux` target. What worked for me is setting the `CMakeLists.txt` in the project directory to this:
```
cmake_minimum_required(VERSION 3.16)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(COMPONENTS custom)
set(EXTRA_COMPONENT_DIRS "custom")
project(hello_world)
```
As can be seen, the `main` component has been renamed to `custom`. Then the `idf_component_register` call in the `CMakeLists.txt` inside `custom` has to be changed to require all necessary components. A good starting point is probably the "common requirements", [as added by the build system](https://github.com/espressif/esp-idf/blob/master/tools/cmake/build.cmake#L228-L237).

I have not tried to use the component manager to pull in any component, though.

Having a `main` component is more like a convention in IDF projects, not a necessity. However, as with all conventions, not following it likely comes at a higher cost, in this case maybe additional maintenance overhead.

higaski, 2023-01-05:
Funnily enough, I discovered this yesterday as well. It works to compile the [hello_world_linux_compatible](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/linux_compatible/hello_world_linux_compatible) example with a different component name. What I didn't expect though is that this even works without explicitly stating component dependencies... Why?

I'd have expected to need to use something along the lines of:
```cmake
idf_component_register(
  SRCS
  ${SRC}
  INCLUDE_DIRS
  ""
  REQUIRES
  esp_common
  esp_hw_support
  esp_rom
  esp_system
  freertos
  hal
  heap
  linux
  log
  soc)
```

Yet I didn't have to...?

The [Running Applications on Host](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/host-apps.html#component-linux-mock-support-overview) documentation mentions that there are mocks for the driver component (although not complete at the moment). How do I include those?

And what's the current state of FreeRTOS simulation on Linux? I may be wrong but I think I've red that only queues were supported until recently? Is this still the case?

higaski, 2023-01-10:
Thank you, I got it all working! I think we can close the issue.
The only thing I'm still couldn't figure out is the CMake target name of the executable which gets produced. This is useful when e.g. defining POST_BUILD custom commands to pack firmware or so.

igrr, 2023-01-10:
> The only thing I'm still couldn't figure out is the CMake target name of the executable which gets produced.

You can call `idf_build_get_property(var EXECUTABLE)` to get the executable (ELF file) target, here's an example with POST_BUILD:
https://github.com/espressif/esp-idf/blob/4c98bee8a4417d92a387ef2416be14ff1fdfa92a/tools/test_apps/system/no_embedded_paths/CMakeLists.txt#L8-L22

For the .bin file, you can attach POST_BUILD step to the `app` target. (`add_custom_command(TARGET app POST_BUILD`)

closed 2023-01-10 by igrr