espressif/esp-idf #10458: hope SPIFFS support directories

Dozingfiretruck, 2022-12-29:
### Is your feature request related to a problem?

SPIFFS dose not support directories

### Describe the solution you'd like.

hope SPIFFS support directories

### Describe alternatives you've considered.

no

### Additional context.

no

CharlesScoville, 2022-12-30:
1: SPIFFS is not actually a part of ESP-IDF, only used by it.

2: The SPIFFS project, including portal for feature requests, is located [here](https://github.com/pellepl/spiffs).

3: SPIFFS has been around for a WHILE with the limitation that it doesn't support directories. Good luck getting that to change.

4: There are perfectly valid reasons for SPIFFS -NOT- supporting directories. Chiefly, that SPIFFS' value position is that it is optimized for resource constrained devices. The overhead of supporting directories (specifically, some sort of file allocation table) would likely consume so much room on the very device you are trying to use to store other things that it would arguably defeat the purpose of using SPIFFS. Not a good look. 

5: A semi-awkward, but highly effective, workaround is to observe that SPIFFS allows for a forward slash [ / ] in file names. So you could have files named `www/index.html` and `www/script.js` and then also have `web/index.html` and `web/script.js` thus providing the illusion of a "www" and "web" folder, for instance. How you do this using a computer system that disallows this character in filenames itself, is the most tricky part. For the ESP8266, I have a customized version of the flashing software ([ESPlorer](https://github.com/CharlesScoville/ESPlorer)) with scripted functionality to decode a directory structure, and automatically rename the files in it accordingly, on the fly, as it is flashing them to the device.

6: ESP-IDF supports FAT, which in turn supports directories.

igrr, 2022-12-30:
To add to what @CharlesScoville said:

1. ESP-IDF integration of SPIFFS does support several directory-related functions, such as `opendir`, `closedir`, `readdir`, `telldir`, `seekdir`. (`mkdir` and `rmdir` are not supported.)
   - Note that `readdir` tries to emulate "reasonable" behavior of iterating over files in a single directory. Since SPIFFS itself doesn't provide such a function, in practice this means iterating over all the files — which is relatively slow.
   - ESP-IDF uses `/` as a directory separator in SPIFFS `readdir` implementation.
2. There was an attempt at supporting directories in SPIFFS port of ESP-IDF using file metadata: https://github.com/espressif/esp-idf/pull/1438. However that PR has not been finished. I am not sure the approach is a sound one anymore, as it is easy for the "directory structure" defined by file names and "directory files" to become out of sync as a result of a sudden power-off.

I think this feature request will likely not be implemented. SPIFFS isn't actively maintained, it is incompatible with ESP32 flash encryption, and we will be moving away from recommending it for new projects. Given that, we are unlikely to put effort  into implementing new SPIFFS features.

Dozingfiretruck, 2022-12-30:
> To add to what @CharlesScoville said:
> 
> 1. ESP-IDF integration of SPIFFS does support several directory-related functions, such as `opendir`, `closedir`, `readdir`, `telldir`, `seekdir`. (`mkdir` and `rmdir` are not supported.)
>    
>    * Note that `readdir` tries to emulate "reasonable" behavior of iterating over files in a single directory. Since SPIFFS itself doesn't provide such a function, in practice this means iterating over all the files — which is relatively slow.
>    * ESP-IDF uses `/` as a directory separator in SPIFFS `readdir` implementation.
> 2. There was an attempt at supporting directories in SPIFFS port of ESP-IDF using file metadata: [Add directory handling to spiffs driver (IDFGH-2499) #1438](https://github.com/espressif/esp-idf/pull/1438). However that PR has not been finished. I am not sure the approach is a sound one anymore, as it is easy for the "directory structure" defined by file names and "directory files" to become out of sync as a result of a sudden power-off.
> 
> I think this feature request will likely not be implemented. SPIFFS isn't actively maintained, it is incompatible with ESP32 flash encryption, and we will be moving away from recommending it for new projects. Given that, we are unlikely to put effort into implementing new SPIFFS features.

OK,thanks

closed 2022-12-30 by Dozingfiretruck