espressif/esp-idf #10460: In ESP32 while using RMT prereferral for IR blaster, Module is not able to Receive more than 93 Characters from AC remote

ShrijithMJain, 2022-12-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Wroom 

### Power Supply used.

USB

### What is the expected behavior?

ESP32 Should receive more than 180 different IR hex codes(items) from IR AC remote.

### What is the actual behavior?

It is Receiving only 93 no matter what changes we made to Memory block size.

### Steps to reproduce.

#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "freertos/semphr.h"
#include "esp_err.h"
#include "esp_log.h"
#include "driver/rmt.h"
#include "driver/periph_ctrl.h"
#include "soc/rmt_reg.h"
//#include "app_config.h"
// FreeRTOS function
#define INCLUDE_vTaskDelay 1

#define rrmt_item32_tIMEOUT_US  9500   /*!< RMT receiver timeout value */

// LED on Huzzah32 board
const int LED_BUILTIN = 32 ;

// structure used to initialize RMT inputs
// NOTE: tag is used on the monitor outputs to distinguish between channels
struct rmt_rx_inits {
	rmt_channel_t  channel;
	gpio_num_t     gpio_num;
	uint8_t        clk_div;
	uint16_t        mem_block_num;
	bool           config_filter_en;
	uint8_t        config_filter_tics_thresh;
	uint16_t       config_idle_threshold;
	char           tag[10];
}rx_inputs[1] = {
	{ 2, 19, 80, 4, true, 0, 50000, "Ch0" },
	
};

// initialize RMT receive channels
void rx_channels_init() {
    rmt_config_t rmt_rx;
    size_t i;
    size_t len = sizeof(rx_inputs) / sizeof( rx_inputs[0] );
    
    for ( i=0; i<len; i++ ) {
		rmt_rx.channel       = rx_inputs[i].channel;
		rmt_rx.gpio_num      = rx_inputs[i].gpio_num;
		rmt_rx.clk_div       = rx_inputs[i].clk_div;
		rmt_rx.mem_block_num = rx_inputs[i].mem_block_num;
		rmt_rx.rmt_mode      = RMT_MODE_RX;
		
		rmt_rx.rx_config.filter_en           = rx_inputs[i].config_filter_en;
		rmt_rx.rx_config.filter_ticks_thresh = rx_inputs[i].config_filter_tics_thresh;
		rmt_rx.rx_config.idle_threshold      = rx_inputs[i].config_idle_threshold;
		
		rmt_config(&rmt_rx);
		rmt_driver_install(rx_inputs[i].channel, 1000, 0);
	}
}

// initialize visible LED on ESP32 board
static void visible_led_init() {
    gpio_pad_select_gpio(LED_BUILTIN);
	gpio_set_direction(LED_BUILTIN, GPIO_MODE_OUTPUT);
}

/* Converts the RMT level, duration into a positive or negative integer
 * Compatible with the ESP32-RMT-server application
 * Note: most IR receivers have active-low outputs, where the
 *   ESP32-RMT-server application has active-high oututs
 * This function inverts the RMT receive level so the text output is
 *   compatible with ESP32-RMT-server application
 */
int dur( uint32_t level, uint32_t duration ) {
	if ( level == 0 ) { return duration; }
	else { return -1.0 * duration; }
}

// RMT receiver task
static void rmt_example_nec_rx_task() {
	size_t num_channels = sizeof(rx_inputs) / sizeof( rx_inputs[0] );
	size_t c, i;
    size_t rx_size = 0;
    rmt_item32_t* items = NULL;
	
    // define ringbuffer handle
    RingbufHandle_t rb;
	
    // start receiving IR data
    for ( c=0; c<num_channels; c++ ) {
		rmt_rx_start(rx_inputs[c].channel, 1);
	}
    
    // loop forever
    while (1) {
		// check each receive channel
		for ( c=0; c<num_channels; c++ ) {
			// get the ring buffer handle
			rmt_get_ringbuf_handle(rx_inputs[c].channel, &rb);
			
			// get items, if there are any
			items = (rmt_item32_t*) xRingbufferReceive(rb, &rx_size, 10);
			if(items) {
				// turn on visible led
				gpio_set_level(LED_BUILTIN, 1);
				
				// print the RMT received durations to the monitor
				printf( "  %s received %i items\n", rx_inputs[c].tag, rx_size/4 );
				for ( i=0; i<rx_size/4; i++ ) {
					if ( i>0 ) { printf(","); }
					printf( "%i", dur( items[i].level0, items[i].duration0 ) );
					printf(",%i", dur( items[i].level1, items[i].duration1 ) );
				}
				printf("\n");
                    
				// turn off visible led
				gpio_set_level(LED_BUILTIN, 0);
				
				// free up data space
				vRingbufferReturnItem(rb, (void*) items);
			}
		}
		// delay 100 milliseconds. No need to overheat the processor
		vTaskDelay( 0.1 / portTICK_PERIOD_MS );
	}
}

void app_main() {
	// initialize hardware
    rx_channels_init();
    visible_led_init();
    
    // start receive processing task
    xTaskCreate(rmt_example_nec_rx_task, "rmt_nec_rx_task", 2048, NULL, 10, NULL);
}

### Debug Logs.

```plain
Ch0 received 93 items
1021,-560,1029,-553,1026,-2587,1022,-2590,1030,-552,1027,-555,1023,-2589,1052,-2560,1050,-531,1027,-555,1055,-2557,1031,-551,1028,-2585,993,-589,1000,-582,9
96,-586,993,-589,1031,-551,1028,-555,1024,-558,1031,-551,996,-586,1003,-579,1000,-582,996,-586,1003,-579,1000,-582,996,-2616,993,-2619,1001,-581,998,-584,99
5,-588,1001,-2610,999,-583,996,-2616,1004,-578,1000,-582,997,-2615,1004,-2608,1002,-2610,999,-583,996,-586,1003,-2609,1000,-582,997,-585,1004,-2608,1001,-26
11,998,-2613,996,-586,1003,-579,1000,-582,996,-2616,1004,-578,1001,-2611,998,-584,994,-2618,1002,-2610,1000,-583,995,-587,1002,-580,999,-557,1021,-588,1001,
-2611,999,-583,995,-587,1002,-581,998,-584,994,-588,1001,-582,997,-585,993,-589,1000,-582,997,-585,993,-589,1000,-2612,997,-585,994,-588,1001,-2612,998,-261
4,995,-2617,971,-8967,735,-6489,739,-3637,741,-6483,736,-6487,742,-6482,768,-3609,737,-3639,738,-6486,732,-6492,737,-3639,739,-6486,732,0
```

### More Information.

Please let me know what to be done in order to rectify this.

suda-morris, 2022-12-29:
any chance to capture the signal that send from the IR blaster to make sure that the sender does give out the correct number of signals? AFAIK, the memory block number is set correctly. **4** memory block can hold up to **256** RMT symbols. But the ringbuffer you set is only **1000**, which can't hold the full **256** symbols (1024 bytes in size). And I see you have enabled the filter feature, but the filter width is set to 0, not sure if the filter will work in that case.

ShrijithMJain, 2022-12-30:
Hi, Suda Thanks for the quick response, I cracked it finally. It was parsing properly, thing that i changed was "rx_size/4" to "rx_size/2" that increased the length of the characters to be displayed from 93 to 186, thats what i needed, and it worked.

closed 2023-01-04 by suda-morris