espressif/esp-idf #10461: Not able to store the characters "!", "@", "#" in flash

sriha99, 2022-12-29:
Hi, 

I am using esp-idf version 4.4.0 . I am submitting a html form through a webpage hosted by the esp32 and storing the form's content in esp32's flash. I am not able to store special characters like "!", "@", "#" in esp32 flash. Instead their ascii equivalent numbers are stored in flash. For example, if I submit password as "!@#", it is getting stored as "%21%40%23" in flash and am not able to connect to mqtt broker because of it. Can someone help me on how to resolve this?

chegewara, 2022-12-29:
The problem is not with storing special characters, which you can test storing it from code, without using website.
It is problem with your code, because you have encode/decode special characters received from web page.

https://github.com/zenmanenergy/ESP8266-Arduino-Examples/blob/master/helloWorld_urlencoded/urlencode.ino#L17

sriha99, 2022-12-29:
Thank you for pointing out the mistake @chegewara. Will try to implement this in my code and get back to you.

laukik-hase, 2023-01-29:
Closed with https://github.com/espressif/esp-idf/commit/167618d6a4e2cfcaab967ba2ba920df64fb4795d

closed 2023-01-29 by laukik-hase