espressif/esp-idf #10462: RMT repeats first item on higher frequencies

LeroyOP0, 2022-12-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Stack:
- ESP-IDF v4.4.3
- ESP32
- Arduino

I'm sending a simple RMT byte (8 items) transmission but for some reason the first bit repeats itself.
In the following code I'll sign the first MSB item/bit with some delay (extended duration) to demonstrate it (sending 0x80).

Here's a simplified code:
```
#include <Arduino.h>
#include <RMT_SPI.h>

#define FREQ 4000000
#define RMT_TICKS 10            // The ticks per clock for RMT duration units
#define APB_CLK 80000000        // in [Hz] - this is the defaultive RMT clock
#define RMT_CLK (FREQ*RMT_TICKS)

#define RMT_GPIO 23

#define MSB_DELAY_TICKS 5 // Set the first MSB item/bit with a delay to sign it

rmt_item32_t items[8];

void setup() {

    Serial.begin(500000);

    rmt_init();
}

void loop() {
    
    uint8_t tx=0x80;

    rmt_fill_byte(tx);

    rmt_write_items(RMT_CHANNEL_0, items, 8, false); 

    rmt_wait_tx_done(RMT_CHANNEL_0,portMAX_DELAY);

    // Delay for a moment before transmitting again
    delay(2);
}

void rmt_init()
{
    rmt_config_t config = {};
    config.rmt_mode = RMT_MODE_TX; // Set the RMT driver to transmit mode
    config.channel = RMT_CHANNEL_0; // Set the RMT channel
    config.gpio_num = gpio_num_t(RMT_GPIO); // Set the GPIO pin
    config.mem_block_num = 1; // Set the number of memory blocks

    config.clk_div = APB_CLK/RMT_CLK;
    config.tx_config.loop_en = false; // Disable looping
    config.tx_config.carrier_en = 0; // Disable the carrier
    config.tx_config.idle_level = RMT_IDLE_LEVEL_LOW; // Set the idle level
    config.tx_config.idle_output_en = true; // Enable idle output
    rmt_config(&config); // Apply the configuration
    rmt_driver_install(config.channel, 0, 0); // Install the RMT driver

}

// Break the byte to bits/items
// 1 = 90% duty cycle
// 0 = 10% duty cycle
void rmt_fill_byte(uint8_t tx) {

  for(int bit=0; bit<8; bit++)
  {
    if (tx & 1<<(7-bit))
    {
        items[bit].level0 = 1;
        items[bit].level1 = 0;
        items[bit].duration0 = 0.9*RMT_TICKS;
        items[bit].duration1 = 0.1*RMT_TICKS;
    }
    else
    {
        items[bit].level0 = 1;
        items[bit].level1 = 0;
        items[bit].duration0 = 0.1*RMT_TICKS; 
        items[bit].duration1 = 0.9*RMT_TICKS;
    }
  }
  items[0].duration0 += MSB_DELAY_TICKS;
  
}
```
This is my scoped signal (notice there are 9 bits instead of 8).
[image]

The interesting thing is that reducing FREQ below 2000000 (Hz) solves the problem and I can see proper transmission (8 bits representing 0x80 as intended):
[image]

LeroyOP0, 2022-12-29:
Ignore the delay feature - it's useless to demonstrate the effect of repeat...
But problem still remains.

suda-morris, 2022-12-30:
Hi @LeroyOP0 Thanks for providing the reproducing code, and I can confirm the issue you met.

I think this weird behavior is related to the hardware's limitation. Lowering down the frequency can help. And also, increase the duration of each RMT item can also help. For example:

```c
void rmt_fill_byte(uint8_t tx)
{
    for (int bit = 0; bit < 8; bit++) {
        if (tx & 1 << (7 - bit)) {
            items[bit].level0 = 1;
            items[bit].level1 = 0;
            items[bit].duration0 = 0.8 * RMT_TICKS;
            items[bit].duration1 = 0.2 * RMT_TICKS;
        } else {
            items[bit].level0 = 1;
            items[bit].level1 = 0;
            items[bit].duration0 = 0.2 * RMT_TICKS;
            items[bit].duration1 = 0.8 * RMT_TICKS;
        }
    }
}
```

The above code can also workaround the issue.

I found an interesting formula on ESP32S3's TRM (sorry I know you're using ESP32)

[image]

I think there should also be a similar one for ESP32.

LeroyOP0, 2022-12-30:
Thanks champ,

By changing the duty cycle as suggested I could even ramp up the frequency to 5Mhz.

You made my smile!!

[image]

```
#define FREQ 5000000
#define RMT_TICKS 8            // The ticks per clock for RMT duration units
#define APB_CLK 80000000        // in [Hz] - this is the defaultive RMT clock
#define RMT_CLK (FREQ*RMT_TICKS)
```

```
void rmt_fill_byte(uint8_t tx) {

  for(int bit=0; bit<8; bit++)
  {
    if (tx & 1<<(7-bit))
    {
        items[bit].level0 = 1;
        items[bit].level1 = 0;
        items[bit].duration0 = 0.75*RMT_TICKS;
        items[bit].duration1 = 0.25*RMT_TICKS;
    }
    else
    {
        items[bit].level0 = 1;
        items[bit].level1 = 0;
        items[bit].duration0 = 0.25*RMT_TICKS; 
        items[bit].duration1 = 0.75*RMT_TICKS;
    }
  }
  
}
```

closed 2022-12-30 by LeroyOP0