espressif/esp-idf #10465: git_submodules.cmake - git_submodule_check - Just Check Used/Related Submodules?

vindicatorr, 2022-12-30:
### Is your feature request related to a problem?

I'm probably just being really anal, but I hate it when projects have submodules that I already have (I encounter this sort of thing often), or those that I won't ever use, but they insist on downloading all of them.

An example would be Matter/ConnectHomeIp. It ended up downloading 2.2GB before I gave up and noped out of there.

Now, while the submodules don't have to all be downloaded and up-to-date in order to compile, there are still the warnings (from the latest master fetched/pulled today):
```
CMake Warning at /<pathTo>/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
[cmake]   Git submodule components/bt/controller/lib_esp32h2/esp32h2-bt-lib is out of
[cmake]   date.  Run 'git submodule update --init --recursive' to fix.
[cmake] Call Stack (most recent call first):
[cmake]   /<pathTo>/esp-idf/tools/cmake/build.cmake:82 (git_submodule_check)
[cmake]   /<pathTo>/esp-idf/tools/cmake/build.cmake:241 (__build_get_idf_git_revision)
[cmake]   /<pathTo>/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
[cmake]   /<pathTo>/esp-idf/tools/cmake/project.cmake:12 (include)
[cmake]   CMakeLists.txt:7 (include)
```

In this case, the submodule is for `lib_esp32h2`, but I only have an `esp32c3`, so this one wouldn't be needed.

### Describe the solution you'd like.

Only check for submodules that are applicable to the current project.

### Describe alternatives you've considered.

Obviously the work-around would just to download all of the submodules, but that's its own problem (unneeded submodules).

### Additional context.

Again, this isn't something that is absolutely needed, and it has no actual effect on getting tasks done.
It's just an annoyance. Anything I make, I aim to get rid of all errors and warnings.

When I do encounter projects with these submodules that are already repeated by other projects, I just manually clone those needed submodules somewhere else, and then soft-link them into the projects.
I may have to finagle them, especially if they are forked submodules with little changes, but I deal with those as needed.

dobairoland, 2023-01-16:
Thank you for the suggestion. We will consider the implementation as soon as possible.

vindicatorr, 2023-01-20:
Heh, I have no clue how I even managed to get as far as I did.
I had just now fetched/pulled from the latest master (I also`diff`ed my changes before pulling), and for some inexplicable reason, I can't even cmake `configure` without it erroring:
```
include could not find requested file:
<pathTo>/esp-idf/components/mqtt/esp-mqtt/CMakeLists.txt
```
I know it's because I didn't clone the submodules, but I don't believe I cloned them previously (hence messages like the one in my first post).

Ohhhh, looks like y'all made a change from https://github.com/espressif/esp-idf/commit/9eccd7c0826d6cc2e9de59304d1e5f76c0063ccf#diff-56a9a462bcfcbb830782051f3c2729631df766bf6bfecc4f6ff35e8e96421ff5 to https://github.com/espressif/esp-idf/commit/5afa3d347c189217c087a098f1bcfc2d3f52ee1f
Huh, looks like that was the only one too. I appended `OPTIONAL` to the include statement like you have in https://github.com/espressif/esp-idf/blob/master/tools/cmake/scripts/component_get_requirements.cmake#L106, and it "configured" just fine.
And just for due-diligence, I went ahead and built it and it also went fine after I updated the mbedtls submodule.

I think I'll actually experiment with doing a symbolic link of the upstream origin of mbedtls, that I have elsewhere, into that dir instead.
This is the sort of thing I speak of with regard to duplicate sources from third-partys that may be used by other projects.
Granted, I do understand that some projects may be using older releases/branches/tags, so I'll have to take that into consideration for any issues I may encounter, but it'd still be good to have a central third-party path to work from rather than duplicating what already exists in your project's own path.
I could just go to that mbedtls dir and `checkout` whichever branch/tag that is needed for whatever project I'm working with (whether esp/nordic/st/...). I hope that makes sense.