espressif/esp-idf #10466: Build fails when enabling LWIP_TCPIP_CORE_LOCKING

wuyuanyi135, 2022-12-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2606-ga74ae7fba7

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

LWIP_TCPIP_CORE_LOCKING option should compile 

### What is the actual behavior?

Compilation failed due to 
```
C:/Espressif/frameworks/esp-idf-master/components/lwip/lwip/src/core/init.c:241:2: error: #error "For LWIP_NETCONN_FULLDUPLEX to work, LWIP_NETCONN_SEM_PER_THREAD is required"
  241 | #error "For LWIP_NETCONN_FULLDUPLEX to work, LWIP_NETCONN_SEM_PER_THREAD is required"
```
https://github.com/espressif/esp-lwip/blob/705dd71d46779bf29653f1f1d7b1af5a09fb2aa7/src/core/init.c#L241

Note that `LWIP_NETCONN_SEM_PER_THREAD` is set to zero by [this line](https://github.com/espressif/esp-idf/blob/8464186e67e34b417621df6b6f1f289a6c60b859/components/lwip/port/esp32/include/lwipopts.h#L849) while `LWIP_NETCONN_FULLDUPLEX` is hard coded to 1 by [the option](https://github.com/espressif/esp-idf/blob/8464186e67e34b417621df6b6f1f289a6c60b859/components/lwip/port/esp32/include/lwipopts.h#L862).

### Steps to reproduce.

1. menuconfig -> component -> lwip ->check `Enable tcpip core locking`

david-cermak, 2023-01-09:
@wuyuanyi135 Sorry for the inconvenience. This option is currently broken on master and `v5.0`. We have a fix in the pipeline, that should be merged in the following weeks. Linking one related issue: https://github.com/espressif/esp-idf/issues/9908

wuyuanyi135, 2023-01-09:
@david-cermak thanks for the heads up!

closed 2023-01-19 by espressif-bot

AxelLin, 2023-02-07:
@david-cermak 

Would you backport the fix to stable branches (v5.0 and v4.4)?
I mainly care about the thread safety issues in non-core locking (which is the default setting) case.

david-cermak, 2023-02-07:
Yes, will backport to both `v5.0` (fully) and `v4.4` (here we'll only patch the issues, rather than fixing core-locking config)

AxelLin, 2023-02-27:
> Yes, will backport to both `v5.0` (fully) and `v4.4` (here we'll only patch the issues, rather than fixing core-locking config)

Just remind I still cannot find the fix in v5.0 and v4.4.

dine909, 2023-02-27:
> Just remind I still cannot find the fix in v5.0 and v4.4.

or in the latest v5.0.1

AxelLin, 2023-03-16:
> Yes, will backport to both `v5.0` (fully) and `v4.4` (here we'll only patch the issues, rather than fixing core-locking config)

Any update? BTW, please also add the missing esp_sntp_enabled() to v5.0.