espressif/esp-idf #10467: Pre encrypted OTA fails with Mismatch chip id, expected 2, found 25300

schafon, 2022-12-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1-dev-2066-g7869f4e151

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S2-DEVKITM-1, ESP32-S2 MINI, ESP32-S2-WROOM-I (CUSTOM)

### Power Supply used.

USB

### What is the expected behavior?

Complete OTA update using HTTPS and pre encrypted firmware.

### What is the actual behavior?

Fail to complete OTA update with error Mismatch chip id, expected 2, found 25300

### Steps to reproduce.

1. get any esp32-s2 with 4mb of flash (thats what I have around)
2. Flash the pre_encrypted_ota  from https://github.com/espressif/esp-idf/tree/master/examples/system/ota/pre_encrypted_ota
3. upload the encrypted firmware to an https server
4. run the code on the device

### Debug Logs.

```plain
[0;32mI (6463) pre_encrypted_ota_example: Starting Pre Encrypted OTA example[0m
[0;32mI (6463) esp_encrypted_img: Starting Decryption Process[0m
I (6483) wifi:<ba-add>idx:1 (ifx:0, 80:03:84:07:91:f8), tid:0, ssn:0, winSize:64
[0;32mI (7723) HTTP_CLIENT: Body received in fetch header state, 0x3ffddb5f, 245[0m
[0;32mI (7723) esp_https_ota: Starting OTA...[0m
[0;32mI (7723) esp_https_ota: Writing to partition subtype 16 at offset 0x110000[0m
[0;32mI (7733) esp_encrypted_img: Magic Verified[0m
[0;32mI (7733) esp_encrypted_img: Reading RSA private key[0m
[0;32mI (8983) pre_encrypted_ota_example: Running firmware version: v5.1-dev-2066-g7869f4e151-dirty[0m
[0;31mE (8983) esp_https_ota: Mismatch chip id, expected 2, found 65529[0m
[0;31mE (8983) pre_encrypted_ota_example: Complete data was not received.[0m
[0;31mE (8993) pre_encrypted_ota_example: ESP_HTTPS_OTA upgrade failed[0m
```

### More Information.

The same code exactly will work as expected in esp32-s3 - according to my testing.
The esp32S3 has  8 mb  of flash (again, this is what I have around).

Found similar issue on esp for Arduino:
https://github.com/espressif/arduino-esp32/issues/7640

No flash encryption or secure boot is enabled.

schafon, 2022-12-31:
@mahavirj 
@hmalpani 
You guys did some work on this example lately, maybe you can check this on your end?

hmalpani, 2023-01-02:
Hello @schafon 

Can you please try the same example on `master`? The fix for this issue has already been merged. Or you can apply 8437c0d12c6db3e412bc35ffc7614a8c03dfa886 commit and then try the example. This should solve the problem you are facing. We are working on backporting the fix. Sorry for the inconvenience.

schafon, 2023-01-02:
Hi, sorry my bad! The master indeed works. I've updated to it.
Thanks and sorry for the FP

closed 2023-01-02 by schafon