espressif/esp-idf #10468: ULP RISC-V RTC I2C doesn't work after esp_restart()

schafon, 2022-12-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2066-g7869f4e151-dirty

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-s2

### Power Supply used.

USB

### What is the expected behavior?

After software esp_restart() continue normal ULP I2C operation.

### What is the actual behavior?

After a restart ulp will hang on:    ulp_riscv_i2c_master_set_slave_reg_addr(reg_addr); **OR** ulp_riscv_i2c_master_read_from_device(data_out, 1);

and will not run again.
    

### Steps to reproduce.

1. Get   ULP RISC-V I2C example from https://github.com/espressif/esp-idf/tree/master/examples/system/ulp_riscv/i2c
2. Patch ULP library and example code according to [this patch](https://github.com/espressif/esp-idf/files/10256487/ulp_riscv_i2c_crash.patch ) from https://github.com/espressif/esp-idf/issues/10301
3. (Optional) add ulp_riscv_print to debug the ulp core - I do that
4. When ulp wake up is detected restart the esp using esp_restart()
5. Watch the UART output of the ulp, after it will reach  ulp_riscv_i2c_master_set_slave_reg_addr(reg_addr); **OR** ulp_riscv_i2c_master_read_from_device(data_out, 1); the code will hang and the esp will not wake up again

### Debug Logs.

```plain
Not a ULP-RISC V wakeup (cause = 0)
Initializing RTC I2C ...
```

### More Information.

I've tested this on esp32 s2 and esp32 s3, different types of dev boards and 1 custom board.

schafon, 2022-12-30:
@sudeep-mohanty  - Can you take a look?

sudeep-mohanty, 2023-01-02:
Hi @schafon,
Thank you for reporting the issue. Really appreciate it!
I was able to reproduce the problem at my end as well. Turns out, when you perform a soft reset using `esp_restart()`, not all peripherals [are reset](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/misc_system_api.html#_CPPv411esp_restartv). Hence, some stale configuration information about the RTC I2C peripheral is also not reset. Therefore, this leads to an undefined behaviour when we try to initialise it a second time round. The solution here is to manually reset the RTC I2C config which the current driver is not doing. I have a fix in progress for this and will be deployed once it is tested and reviewed. However, to unblock your project, you may try this at your end and let me know your feedback regarding the issue. Thanks! - 

[rtc_i2c_esp_restart.patch](https://github.com/espressif/esp-idf/files/10332249/rtc_i2c_esp_restart.patch)

schafon, 2023-01-02:
The patch works perfectly.
Thank again Sudeep.

closed 2023-01-02 by schafon