espressif/esp-idf #10470: wifi station example fail to connect WPA3 PSK with esp-idf v4.3 branch

AxelLin, 2023-01-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.4-309-g389a848a6a

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E

### Power Supply used.

USB

### What is the expected behavior?

Connect to WPA3 PSK should work.

### What is the actual behavior?

Fail to connect WPA3 PSK if build with  v4.3.4-309-g389a848a6a.
It works if build with v4.4.3-347-g9ee3c8337d.

### Steps to reproduce.

Test using examples/wifi/getting_started/station example code.
The example code works with v4.4 branch but failed with current v4.3 branch.

### Debug Logs.

```plain
v4.4.3-347-g9ee3c8337d log: (success log for comparison)

I (375) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (441) wifi station: ESP_WIFI_MODE_STA
I (451) wifi:wifi driver task: 3ffbfccc, prio:23, stack:6656, core=0
I (451) system_api: Base MAC address is not set
I (451) system_api: read default base MAC address from EFUSE
I (471) wifi:wifi firmware version: b5b20ea
I (471) wifi:wifi certification version: v7.0
I (471) wifi:config NVS flash: enabled
I (471) wifi:config nano formating: disabled
I (471) wifi:Init data frame dynamic rx buffer num: 32
I (481) wifi:Init management frame dynamic rx buffer num: 32
I (481) wifi:Init management short buffer num: 32
I (491) wifi:Init dynamic tx buffer num: 32
I (491) wifi:Init static rx buffer size: 1600
I (491) wifi:Init static rx buffer num: 10
I (501) wifi:Init dynamic rx buffer num: 32
I (501) wifi_init: rx ba win: 6
I (511) wifi_init: tcpip mbox: 32
I (511) wifi_init: udp mbox: 6
I (511) wifi_init: tcp mbox: 6
[... 931 lines cut ...]
0x400d4f0c: event_handler at /home/axel/esp/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:65

I (14648) wifi station: retry to connect to the AP
I (14648) wifi station: connect to the AP fail
D (17058) event: running post WIFI_EVENT:5 with handler 0x400dbc94 and context 0x3ffbf570 on loop 0x3ffbe33c
0x400dbc94: wifi_default_action_sta_disconnected at /home/axel/esp/esp-idf/components/esp_wifi/src/wifi_default.c:107

D (17058) esp_netif_handlers: esp_netif action disconnected with netif0x3ffbf268 from event_id=5
D (17068) esp_netif_lwip: check: remote, if=0x3ffbf268 fn=0x400d9a0c
0x400d9a0c: esp_netif_down_api at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1278

D (17068) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbf268
D (17078) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbf268
D (17088) esp_netif_lwip: if0x3ffbf268 start ip lost tmr: no need start because netif=0x3ffbf2e8 interval=120 ip=0
D (17098) esp_netif_lwip: check: local, if=0x3ffbf268 fn=0x400da2dc
0x400da2dc: esp_netif_update_default_netif_lwip at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:189

D (17098) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbf268
D (17108) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (17118) event: running post WIFI_EVENT:5 with handler 0x400d4f0c and context 0x3ffb9764 on loop 0x3ffbe33c
0x400d4f0c: event_handler at /home/axel/esp/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:65

I (17128) wifi station: connect to the AP fail
I (17128) wifi station: Failed to connect to SSID:CHT Wi-Fi Home-J9K, password:0226868632
```

AxelLin, 2023-01-03:
I notice in v4.4 branch, it has a few retry before successfully join AP.
This seems not normal. 
I add debug print to show the disconnect reason and rssi:
I (3195) wifi station: Disconnect reason: 202, rssi=-27
I (5755) wifi station: Disconnect reason: 205, rssi=-128

BTW, my device is very close to the AP.
I tested several times, it always show above disconnect reason.

Below is the debug log:

I (0) cpu_start: App cpu up.
D (305) clk: RTC_SLOW_CLK calibration value: 3159603
I (317) cpu_start: Pro cpu start user code
I (317) cpu_start: cpu freq: 160000000
I (317) cpu_start: Application information:
I (322) cpu_start: Project name:     wifi_station
I (327) cpu_start: App version:      v4.4.3-347-g9ee3c8337d
I (333) cpu_start: Compile time:     Jan  3 2023 08:43:38
I (339) cpu_start: ELF file SHA256:  b319cf6da6b9cb17...
I (345) cpu_start: ESP-IDF:          v4.4.3-347-g9ee3c8337d
D (352) memory_layout: Checking 7 reserved memory ranges:
D (357) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (364) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb70c0
D (370) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (376) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (383) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (389) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at /home/axel/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1739

D (396) memory_layout: Reserved memory range 0x40080000 - 0x40093768
0x40080000: _WindowOverflow4 at /home/axel/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1739

D (402) memory_layout: Building list of available memory regions:
D (408) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (415) memory_layout: Available memory region 0x3ffb70c0 - 0x3ffb8000
D (421) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
D (428) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
D (435) memory_layout: Available memory region 0x3ffc2000 - 0x3ffc4000
D (441) memory_layout: Available memory region 0x3ffc4000 - 0x3ffc6000
D (448) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
D (454) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
D (461) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
D (468) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
D (474) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
D (481) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
D (487) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
D (494) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
D (501) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000
D (507) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000
D (514) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
D (520) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
D (527) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
D (534) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe3f20
D (540) memory_layout: Available memory region 0x3ffe4350 - 0x3ffe8000
D (547) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
D (553) memory_layout: Available memory region 0x3fff0000 - 0x3fff8000
D (560) memory_layout: Available memory region 0x3fff8000 - 0x3fffc000
D (567) memory_layout: Available memory region 0x3fffc000 - 0x40000000
D (573) memory_layout: Available memory region 0x40093768 - 0x40094000
D (580) memory_layout: Available memory region 0x40094000 - 0x40096000
D (586) memory_layout: Available memory region 0x40096000 - 0x40098000
D (593) memory_layout: Available memory region 0x40098000 - 0x4009a000
D (600) memory_layout: Available memory region 0x4009a000 - 0x4009c000
D (606) memory_layout: Available memory region 0x4009c000 - 0x4009e000
D (613) memory_layout: Available memory region 0x4009e000 - 0x400a0000
I (619) heap_init: Initializing. RAM available for dynamic allocation:
D (627) heap_init: New heap initialised at 0x3ffae6e0
I (632) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (638) heap_init: New heap initialised at 0x3ffb70c0
[... 864 lines cut ...]
D (8885) wifi:update trc
D (8895) esp_netif_lwip: check: remote, if=0x3ffbd690 fn=0x400da6d4
0x400da6d4: esp_netif_dhcpc_start_api at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1049

D (8895) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3ffbd690
D (8905) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbd690
D (8915) esp_netif_lwip: if0x3ffbd690 start ip lost tmr: no need start because netif=0x3ffbd710 interval=120 ip=0
D (8925) esp_netif_lwip: starting dhcp client
D (8925) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (8935) event: running post WIFI_EVENT:4 with handler 0x400d542c and context 0x3ffc4a38 on loop 0x3ffbc764
0x400d542c: event_handler at /home/axel/esp/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:65

D (9925) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffbd710
D (9925) esp_netif_lwip: if0x3ffbd690 ip changed=1
D (9925) event: running post IP_EVENT:0 with handler 0x400dcc24 and context 0x3ffbda3c on loop 0x3ffbc764
0x400dcc24: wifi_default_action_sta_got_ip at /home/axel/esp/esp-idf/components/esp_wifi/src/wifi_default.c:123

D (9935) wifi_init_default: Got IP wifi default handler entered
D (9935) esp_netif_handlers: esp_netif action got_ip with netif0x3ffbd690 from event_id=0
I (9945) esp_netif_handlers: sta ip: 192.168.0.106, mask: 255.255.255.0, gw: 192.168.0.1
D (9955) event: running post IP_EVENT:0 with handler 0x400d542c and context 0x3ffc4a58 on loop 0x3ffbc764
0x400d542c: event_handler at /home/axel/esp/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:65

I (9965) wifi station: got ip:192.168.0.106
I (9975) wifi station: connected to ap SSID:CHT Wi-Fi Home-J9K password:0226868632

nishanth-radja, 2023-01-03:
@AxelLin can we have sniffer captures for 4.4 and 4.3 during the Association.
From the 4.4 logs,Looks the AUTH has timed out the first time ,since it did not get the response from AP and was successful the second time.
Have you rebooted the client after successful association and trying to re associate again?

AxelLin, 2023-01-03:
Attached the wireshark log.
[wireshark-v4.4.pcapng.txt](https://github.com/espressif/esp-idf/files/10335910/wireshark-v4.4.pcapng.txt)
[wireshark-v4.3.pcapng.txt](https://github.com/espressif/esp-idf/files/10335911/wireshark-v4.3.pcapng.txt)

AxelLin, 2023-01-03:
> Have you rebooted the client after successful association and trying to re associate again?

Yes.
I run idf.py monitor to check the log.
After check the log, I press ctrl+], then run "idf.py monitor" again. (This will reset the device).

AxelLin, 2023-01-05:
> @AxelLin what is the model of the AP you are using?

Arcadyan WG630223-TC

Shreyas0-7, 2023-01-05:
Hey @AxelLin 
Can you please try to reproduce the issue using this build for v4.3
[build.zip](https://github.com/espressif/esp-idf/files/10350604/build.zip)

AxelLin, 2023-01-05:
> Hey @AxelLin Can you please try to reproduce the issue using this build for v4.3 [build.zip](https://github.com/espressif/esp-idf/files/10350604/build.zip)

I cannot do that, my testing device has flash encryption enabled.
(I don't have any device with flash disabled handy)
Can you just send me the updated wifi-lib so I can build it locally for testing.

Shreyas0-7, 2023-01-05:
Yes can you please try this patch
[wpa_parse_group_cipher.patch.txt](https://github.com/espressif/esp-idf/files/10350835/wpa_parse_group_cipher.patch.txt)

AxelLin, 2023-01-05:
> Yes can you please try this patch [wpa_parse_group_cipher.patch.txt](https://github.com/espressif/esp-idf/files/10350835/wpa_parse_group_cipher.patch.txt)

Hi @Shreyas0-7 
It does not work.
In the first time test, it shows below error for 5 times, then failed.

I (16569) wifi station: Disconnect reason: 201, rssi=-128
I (16569) wifi station: connect to the AP fail

The second time test (after reboot), it shows below
I (1649) wifi station: wifi_init_sta finished.
D (1649) esp_netif_lwip: esp_netif_start_api 0x3ffbf498
D (1659) esp_netif_lwip: esp_netif_get_hostname esp_netif:0x3ffbf498
D (1669) esp_netif_lwip: check: local, if=0x3ffbf498 fn=0x400da304
0x400da304: esp_netif_update_default_netif_lwip at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:189

D (1669) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbf498
D (1679) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (1679) event: running post WIFI_EVENT:2 with handler 0x400d4f10 and context 0x3ffb9764 on loop 0x3ffbe56c
0x400d4f10: event_handler at /home/axel/esp/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:65

D (2909) wpa: WPA: set AP RSNXE - hexdump(len=3):
D (2919) wpa: f4 01 20 
I (2919) wifi:new:<11,2>, old:<1,0>, ap:<255,255>, sta:<11,2>, prof:1
D (2919) wpa: RSN: PMKSA cache search - network_ctx=0x0 try_opportunistic=0
D (2919) wpa: RSN: Search for BSSID ac:b6:87:d1:a8:68
D (2929) wpa: RSN: No PMKSA cache entry found
D (2929) wpa: RSN: set own RSNXE - hexdump(len=3):
D (2939) wpa: f4 01 20 
I (2939) wifi:state: init -> auth (b0)
D (2949) wpa: SAE: Derive PT - group 19
D (2949) wpa: SAE: pwd-seed - hexdump(len=32):
D (2949) wpa: ce f0 8b 91 96 20 d1 08 e5 86 62 be f7 8a 3c 62 
D (2959) wpa: 56 79 32 39 54 eb 86 9a 3d 1e fa c1 b2 bf cf 4f 
D (2969) wpa: SAE: pwd-value (u1 P1) - hexdump(len=48):
D (2969) wpa: 42 fb 83 3f e3 e4 13 40 97 b0 b6 1c 41 e4 82 11 
D (2979) wpa: 79 95 15 3b 72 3b 3a 45 2a 56 7e ef b9 99 0a d6 
D (2979) wpa: 71 bc 98 e4 16 b5 22 86 7d 70 ac 1c 30 ed 10 56 
D (2989) wpa: SAE: u1 - hexdump(len=32):
D (2989) wpa: 94 9a 00 cb 98 a6 02 18 03 76 e8 6f 3e 04 41 7a 
D (2999) wpa: 1e 1e 67 37 3d 94 b9 06 f9 05 75 78 c7 86 c5 43 
D (3009) wpa: SSWU: m - hexdump(len=32):
D (3009) wpa: a0 06 1b 06 57 0f 8b a7 a4 6b 24 96 fc 4e 2a 23 
D (3009) wpa: b2 d1 5c 94 11 f4 f5 8b 1e ec 24 79 46 10 13 2f 
D (3029) wpa: SSWU: t - hexdump(len=32):
D (3029) wpa: 03 bf ac de 52 36 3d 52 47 ad e6 88 d3 6b 79 d2 
D (3029) wpa: 62 a4 50 27 34 0f 69 9c 89 ed 22 7b bb 32 d0 9f 
D (3039) wpa: SSWU: x1a = b / (z * a) - hexdump(len=32):
D (3039) wpa: a5 28 bd 86 96 bd af 99 6c 65 b9 82 d9 49 59 d3 
D (3049) wpa: 14 6f e6 a0 20 69 30 90 bd ba 13 13 23 75 f2 24 
D (3059) wpa: SSWU: x1b = (-b/a) * (1 + t) - hexdump(len=32):
D (3059) wpa: 7d b4 26 43 62 c7 be 21 63 e9 71 fd a6 06 30 df 
D (3059) wpa: 99 3e 6f 80 76 93 f8 43 fa f6 13 f4 f5 a3 d1 aa 
D (3069) wpa: SSWU: x1 = CSEL(l, x1a, x1b) - hexdump(len=32):
D (3079) wpa: 7d b4 26 43 62 c7 be 21 63 e9 71 fd a6 06 30 df 
D (3079) wpa: 99 3e 6f 80 76 93 f8 43 fa f6 13 f4 f5 a3 d1 aa 
D (3089) wpa: SSWU: gx1 = x1^3 + a * x1 + b - hexdump(len=32):
D (3089) wpa: c1 8d 77 0d 38 5f eb 58 e7 31 0f 66 af 15 9d 46 
D (3099) wpa: 81 46 b7 14 36 71 56 24 7d 9c aa 9b 53 bd 41 13 
D (3109) wpa: SSWU: x2 = z * u^2 * x1 - hexdump(len=32):
D (3109) wpa: d1 a8 84 c3 d5 71 03 2b 85 e9 a2 2a 14 d7 3a a6 
D (3119) wpa: 34 7b 21 36 5a 7d 38 88 1c 9c 50 c4 3c aa 79 a2 
D (3129) wpa: SSWU: gx2 = x2^3 + a * x2 + b - hexdump(len=32):
D (3129) wpa: 3b d7 02 e4 1a 87 8f 30 11 41 42 96 7f bd 81 cf 
D (3139) wpa: ba 49 b7 b5 48 7f 54 32 b6 53 8b 1d 6c 7c 69 48 
D (3149) wpa: SSWU: gx1^((p-1)/2) modulo p - hexdump(len=32):
D (3149) wpa: 7f ff ff ff 80 00 00 00 80 00 00 00 00 00 00 00 
D (3149) wpa: 00 00 00 00 7f ff ff ff ff ff ff ff ff ff ff ff 
D (3159) wpa: SSWU: v = CSEL(l, gx1, gx2) - hexdump(len=32):
D (3159) wpa: c1 8d 77 0d 38 5f eb 58 e7 31 0f 66 af 15 9d 46 
[... 742 lines cut ...]

I (15769) wifi station: Disconnect reason: 1, rssi=-22
I (15779) wifi station: retry to connect to the AP
I (15779) wifi station: connect to the AP fail
D (18189) event: running post WIFI_EVENT:5 with handler 0x400dbcbc and context 0x3ffbf7a0 on loop 0x3ffbe56c
0x400dbcbc: wifi_default_action_sta_disconnected at /home/axel/esp/esp-idf/components/esp_wifi/src/wifi_default.c:107

D (18189) esp_netif_handlers: esp_netif action disconnected with netif0x3ffbf498 from event_id=5
D (18199) esp_netif_lwip: check: remote, if=0x3ffbf498 fn=0x400d9a34
0x400d9a34: esp_netif_down_api at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1278

D (18199) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbf498
D (18209) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbf498
D (18219) esp_netif_lwip: if0x3ffbf498 start ip lost tmr: no need start because netif=0x3ffbf518 interval=120 ip=0
D (18229) esp_netif_lwip: check: local, if=0x3ffbf498 fn=0x400da304
0x400da304: esp_netif_update_default_netif_lwip at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:189

D (18229) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbf498
D (18239) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (18249) event: running post WIFI_EVENT:5 with handler 0x400d4f10 and context 0x3ffb9764 on loop 0x3ffbe56c
0x400d4f10: event_handler at /home/axel/esp/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:65

I (18259) wifi station: Disconnect reason: 205, rssi=-128
I (18259) wifi station: connect to the AP fail
I (18269) wifi station: Failed to connect to SSID:CHT Wi-Fi Home-J9K, password:0226868632

AxelLin, 2023-01-06:
> Hey @AxelLin thanks for your co-ordination can you please test it again with [wpa_parse_group_cipher.patch.txt](https://github.com/espressif/esp-idf/files/10357889/wpa_parse_group_cipher.patch.txt)

Hi @Shreyas0-7 

This patch works, thanks a lot.

Now the symptom is the same for v4.3 and v4.4:
When repeat testing connect AP, I always got 202 and 205 disconnect reasons before successfully connect to AP.
(But I think that is a different topic)

I (4519) wifi station: Disconnect reason: 202, rssi=-24
I (4529) wifi station: retry to connect to the AP
I (4529) wifi station: connect to the AP fail
...
I (6999) wifi station: Disconnect reason: 205, rssi=-128
I (7009) wifi station: retry to connect to the AP
I (7009) wifi station: connect to the AP fail
...
I (11269) wifi station: got ip:192.168.0.105
I (11279) wifi station: connected to ap

nishanth-radja, 2023-01-06:
@AxelLin This is because the AP is rejecting the AUTH after you have reset your STA.
Since the STA list in the AP is not cleared ,PMF kicks in once you try to associate the next time.
From the 4.4 capture, check pkt 187 from AP.
[image]

AxelLin, 2023-01-06:
> @AxelLin This is because the AP is rejecting the AUTH after you have reset your STA. 

I'm wondering if this symptom can cause this issue: https://github.com/espressif/esp-idf/issues/8269 .
So the device may join an AP with weaker signal.

nishanth-radja, 2023-01-06:
@AxelLin AP in #8269 is a wpa2 AP. The issue seen here will be caused by wpa3 AP (PMF is causing the AUTH to fail)

AxelLin, 2023-01-06:
> @AxelLin AP in #8269 is a wpa2 AP. The issue seen here will be caused by wpa3 AP (PMF is causing the AUTH to fail)

Because the solution of #8269 is adding failure_retry_cnt to wifi_sta_config_t, so I thought any connect failure will cause this issue for whatever reason. (I don't know the implementation detail, my understanding might be wrong).

AxelLin, 2023-02-04:
Fixed by c78324e407fd9ecdbd6ef6c68b20f6007537042d

closed 2023-02-04 by AxelLin