espressif/esp-idf #10471: ULP ADC reading not working after deep sleep + esp_wifi_stop()

NikLeberg, 2023-01-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2658-g0025915dc4

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

The ULP is configured to put ADC measurement into shared global variable. I'd expect that after correct configuration the ULP will continue to put ADC data into variable. This should also work after deep sleep and wifi start/stop.

### What is the actual behavior?

The ULP outputs the ADC values happily before the wifi stack is stopped. But after a deep sleep and a wifi stop the values are frozen in time and won't update anymore. It does not happen if there is no deep sleep in-between.

### Steps to reproduce.

1. Build & flash attached minimal example project: [ulp-wifi-stop-ex.zip](https://github.com/espressif/esp-idf/files/10329557/ulp-wifi-stop-ex.zip)
2. Attach some analog circuitry (like a LDR) to the ADC1 channel 6 pad (GPIO34 on ESP32) or leave it floating so that the ADC reads some random values.
3. Observe output (see below)
4. It will:
- setup ULP & ADC,
- enter deep sleep,
- be woken up by ULP,
- set up WLAN,
- stop WLAN

At every step ADC values are written to output. After call to `esp_wifi_stop()` and before second call to `ulp_adc_init()` the ADC value is frozen and will not update.

### Debug Logs.

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6972
ho 0 tail 12 room 4
load:0x40078000,len:15492
load:0x40080400,len:3844
0x40080400: _init at ??:?

entry 0x4008064c
I (29) boot: ESP-IDF v5.1-dev-2658-g0025915dc4 2nd stage bootloader
[... 407 lines cut ...]
I (39270) example: Value: 2075
I (39470) example: Value: 2083
I (39670) example: Value: 2062
I (39870) example: Value: 2085
I (40070) example: Value: 2090
I (40270) example: Value: 2094
I (40470) example: Value: 2096
I (40670) example: Value: 2096
I (40870) example: Value: 2101
I (41070) example: Value: 2107
I (41270) example: Value: 2126
I (41470) example: Value: 2112
I (41670) example: Value: 2112
I (41870) example: Value: 2112
I (42070) example: Value: 2107
I (42070) main_task: Returned from app_main()
```

### More Information.

As seen in the example code, if the ADC is initialized again with a call to `ulp_adc_init()`, then the ULP starts spitting out fresh and correct values again.

I noticed this bug after the ULP could no longer wake the SoC in one of my project. I'm assuming it got stuck on the `adc` instruction while reading the ADC that was miss configured (or erroneously de initialized by the wifi library?) and could never continue from that instruction.

I have read that Wifi can influence ADC operation. But that should only be the case for ADC2, or isn't it? I'm also not sure if I'm not supposed to initialize the ADC after every deep sleep anyway. So if I'm using the API wrong I'm happy to be corrected.

NikLeberg, 2023-01-05:
I did some debugging and compared the HW registers before and after the call to `esp_wifi_stop()`. The main difference observed was, that the ADC peripheral gets powered down, i.e. in the register `SENS_SAR_MEAS_WAIT2_REG` the field `SENS_FORCE_XPD_SAR` is set to a value `SENS_FORCE_XPD_SAR_PD`. To operate with the ULP this needs to be set to `SENS_FORCE_XPD_SAR_FSM` or at least that is what `ulp_adc_init()` does.

Setting the register manually to `SENS_FORCE_XPD_SAR_FSM` solves the issue for me.

I then debugged with GDB to see what piece of code changes this register. I found the set of functions `adc_power_acquire()` and `adc_power_release()` that seem to manage the power to the ADC peripheral. They manage a _reference count_ of the ADC usages and power it down if no code uses it any more. PD is done differently for the ESP32 than for the other ESP modules:
```c

static void adc_power_off_internal(void)
{
#if CONFIG_IDF_TARGET_ESP32
    adc_hal_set_power_manage(ADC_POWER_SW_OFF);
#else
    adc_hal_set_power_manage(ADC_POWER_BY_FSM);
#endif
}
```
The `ADC_POWER_BY_FSM` would be equivalent to the required `SENS_FORCE_XPD_SAR_FSM` to solve the issue in my case.

This precompiler switch was added in commit [`cc6bfcd`](https://github.com/espressif/esp-idf/commit/cc6bfcd9ac17036824e005f1569df4b66c6415d3) but I sadly could not find a reason for the special handling of ESP32.
If I remove the ESP32 special case and always let it set `ADC_POWER_BY_FSM` then the ADC usage with ULP works.

I'm assuming, that initially that reference count is incremented with the use of `ulp_adc_init()`. But then after the deep sleep the count is obviously lost and the HAL no longer _knows_ that the ULP is using it. As such after the last module (the wifi stack in this case) stops to use the ADC, the ADC is powered down.

Am I supposed to use the ADC API in a different way?
What is the specific reason for the special handling of original ESP32 module?

Xiehanxin, 2023-01-13:
hi @NikLeberg  sorry for late reply, thanks for your feedback, we will take a test