espressif/esp-idf #10472: chip_revision.h - "expected ',' before ')' token"

vindicatorr, 2023-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Commit 8464186e67e34b417621df6b6f1f289a6c60b859

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build the example successfully.

### What is the actual behavior?

https://github.com/espressif/esp-idf/blob/master/components/soc/include/soc/chip_revision.h#L36
`components/soc/include/soc/chip_revision.h:36:66: error: expected ',' before ')' token
 _Static_assert(CONFIG_ESP_REV_MIN_FULL <= CONFIG_ESP_REV_MAX_FULL);`
Needs the message part.

### Steps to reproduce.

1. cd examples/get-started/hello_world/
2. cmake -B builddir/
3. cd builddir/
4. make

### Build or installation Logs.

```plain
...
[ 14%] Building C object esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32c3/rtc_sleep.c.obj
In file included from /<pathTo>/esp-idf/components/esp_hw_support/port/esp32c3/rtc_sleep.c:19:
/<pathTo>/esp-idf/components/soc/include/soc/chip_revision.h:36:66: error: expected ',' before ')' token
 _Static_assert(CONFIG_ESP_REV_MIN_FULL <= CONFIG_ESP_REV_MAX_FULL);
                                                                  ^
cc1: warning: unrecognized command line option '-Wno-enum-conversion'
make[5]: *** [esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/build.make:160: esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/port/esp32c3/rtc_sleep.c.obj] Error 1
make[4]: *** [CMakeFiles/Makefile2:1225: esp-idf/esp_hw_support/CMakeFiles/__idf_esp_hw_support.dir/all] Error 2
make[3]: *** [Makefile:91: all] Error 2
make[2]: *** [CMakeFiles/bootloader.dir/build.make:86: bootloader-prefix/src/bootloader-stamp/bootloader-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1646: CMakeFiles/bootloader.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```
```

### More Information.

riscv32-esp-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0

igrr, 2023-01-02:
It looks like you aren't using the correct version of the toolchain for the IDF commit you are using. The version should be `esp-2022r1-11.2.0`, not `esp-2021r2-patch5`.

Please try to run install.sh again, then update the environment variables using export.sh. Finally, do `idf.py fullclean` and then build the project again. Verify in CMake output that `esp-2022r1-11.2.0` version of the toolchain is used.

vindicatorr, 2023-01-02:
Ah crud! Actually, I had originally been working with with the crosstool-ng build I did in November, but was having issues with Matter (commented elsewhere).
So for due-diligence, I downloaded the toolchain at the top of Releases without looking at anything other than it was sorted by date.
When I tested Matter, and it still failed, I thought I'd test hello-world. This is where this issue came in.
Since then I noticed the top "Release" was the patched one for r2021.

I then downloaded the r2022 from July, and you're correct, that it worked. But still not for my Matter issue which I brought up over there.
My bad, sorry.

closed 2023-01-02 by vindicatorr

igrr, 2023-01-02:
> EDIT0: I should add, shouldn't it still have a "message" component in that assert?

I think this isn't strictly necessary in the newer versions of the C standard: `static_assert` and `_Static_assert` now both work without the message argument. Given that the assert condition here is pretty much self-explanatory, I think it's okay to leave this as it is.