espressif/esp-idf #10473: xtensa-esp32-elf-gcc not found even though idf_tools.py says it is installed

tongclement, 2023-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I accidently deleted the .espressif file that contains the idf tools when trying to trouble shoot an idf update. When updating, what I did was delete the esp_idf folder and reinstall via VS Code. 
However, whatever I attempted, idf_tools.py or install.sh thinks the xtensa-esp32-elf-gcc compiler is installed but it clearly isn't working. All I want is a full uninstallation and re-installation of the idf as things are so messed up now. Many thanks for any advice. 

### What is the actual behavior?

Not sure how to proceed. 
What I attempted and the log files:
```
(base) clementtongpersonal@HISHK-MBP1210-3 esp-idf % cd tools 
(base) clementtongpersonal@HISHK-MBP1210-3 tools % idf_tools.py check
Checking for installed tools...
Checking tool xtensa-esp32-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool xtensa-esp32s2-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool xtensa-esp32s3-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool xtensa-clang
    version found in PATH: unknown
    version installed in tools directory: 12.0.1-d9341b81fc
Checking tool riscv32-esp-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool esp32ulp-elf
    no version found in PATH
    version installed in tools directory: 2.35_20220830
Checking tool cmake
    version found in PATH: 3.20.2
    version installed in tools directory: 3.23.1
Checking tool openocd-esp32
    no version found in PATH
    version installed in tools directory: v0.11.0-esp32-20220706
Checking tool ninja
    version found in PATH: 1.10.2
    version installed in tools directory: 1.10.2
(base) clementtongpersonal@HISHK-MBP1210-3 tools % idf_tools.py export
Not using an unsupported version of tool xtensa-clang found in PATH: unknown. To use it, run '/Users/clementtongpersonal/miniforge3/bin/python /Users/clementtongpersonal/esp/esp-idf/tools/idf_tools.py export --prefer-system'
Not using an unsupported version of tool cmake found in PATH: 3.20.2. To use it, run '/Users/clementtongpersonal/miniforge3/bin/python /Users/clementtongpersonal/esp/esp-idf/tools/idf_tools.py export --prefer-system'
export OPENOCD_SCRIPTS="/Users/clementtongpersonal/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts";export IDF_PYTHON_ENV_PATH="/Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env";export PATH="/Users/clementtongpersonal/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/Users/clementtongpersonal/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin:/Users/clementtongpersonal/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:/Users/clementtongpersonal/.espressif/tools/xtensa-clang/12.0.1-d9341b81fc/xtensa-esp32-elf-clang/bin:/Users/clementtongpersonal/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin:/Users/clementtongpersonal/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Users/clementtongpersonal/.espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Users/clementtongpersonal/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin:/Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/bin:$PATH"
(base) clementtongpersonal@HISHK-MBP1210-3 tools % idf_tools.py check
Checking for installed tools...
Checking tool xtensa-esp32-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool xtensa-esp32s2-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool xtensa-esp32s3-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool xtensa-clang
    version found in PATH: unknown
    version installed in tools directory: 12.0.1-d9341b81fc
Checking tool riscv32-esp-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool esp32ulp-elf
    no version found in PATH
    version installed in tools directory: 2.35_20220830
Checking tool cmake
    version found in PATH: 3.20.2
    version installed in tools directory: 3.23.1
Checking tool openocd-esp32
    no version found in PATH
    version installed in tools directory: v0.11.0-esp32-20220706
Checking tool ninja
    version found in PATH: 1.10.2
    version installed in tools directory: 1.10.2
(base) clementtongpersonal@HISHK-MBP1210-3 tools % idf_tools.py install all
Current system platform: macos-arm64
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, xtensa-clang, riscv32-esp-elf, esp32ulp-elf, cmake, openocd-esp32, ninja
Skipping xtensa-esp32-elf@esp-2021r2-patch5-8.4.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2021r2-patch5-8.4.0 (already installed)
Skipping xtensa-esp32s3-elf@esp-2021r2-patch5-8.4.0 (already installed)
Skipping xtensa-clang@12.0.1-d9341b81fc (already installed)
Skipping riscv32-esp-elf@esp-2021r2-patch5-8.4.0 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping cmake@3.23.1 (already installed)
Skipping openocd-esp32@v0.11.0-esp32-20220706 (already installed)
Skipping ninja@1.10.2 (already installed)
(base) clementtongpersonal@HISHK-MBP1210-3 tools % idf_tools.py check
Checking for installed tools...
Checking tool xtensa-esp32-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool xtensa-esp32s2-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool xtensa-esp32s3-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool xtensa-clang
    version found in PATH: unknown
    version installed in tools directory: 12.0.1-d9341b81fc
Checking tool riscv32-esp-elf
    no version found in PATH
    version installed in tools directory: esp-2021r2-patch5-8.4.0
Checking tool esp32ulp-elf
    no version found in PATH
    version installed in tools directory: 2.35_20220830
Checking tool cmake
    version found in PATH: 3.20.2
    version installed in tools directory: 3.23.1
Checking tool openocd-esp32
    no version found in PATH
    version installed in tools directory: v0.11.0-esp32-20220706
Checking tool ninja
    version found in PATH: 1.10.2
    version installed in tools directory: 1.10.2
(base) clementtongpersonal@HISHK-MBP1210-3 tools % 
```

### Steps to reproduce.

N.A.

dobairoland, 2023-01-02:
Hi @tongclement. I'm linking here your open issue for vscode https://github.com/espressif/vscode-esp-idf-extension/issues/862

Can you please try the `install.sh` and `. export.sh` scripts in order to set up an environment? Note that this is the way how to set up ESP-IDF terminal environment. If you prefer Vscode then you should follow-up on the Vscode installation issue.

The supported way to set up ESP-IDF terminal environments is explained here: https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/get-started/index.html#step-3-set-up-the-tools

You don't need to use `idf_tools.py` directly. It is a low level tool which `install.sh` and `export.sh` uses.  But looking at what you are intend to do, I guess you are missing the `eval` step (https://github.com/espressif/esp-idf/blob/release/v4.4/export.sh#L95), i.e. it is not enough to run `idf_tools.py export` but the result should be exported: 
```
eval `idf_tools.py export`
```

But please use the `install.sh` and `. export.sh` commands instead and let us know if the issue is persistent. Note: don't forget the "dot & space" before `export.sh`.

I recommend to check the IDF_PATH environment variable to ensure it points to the right version of ESP-IDF (in case you have multiple copies on your computer).

tongclement, 2023-01-02:
Thanks for all the suggestions. It seems like I managed to bypass the above issue although it still causes some errors. For example, I am experiencing the follow error when monitoring the chip:
```
***ERROR*** A stack overflow in task esp2_rx_task has been detected.

Backtrace: 0x400819aa:0x3ffba2c0 0x400861f1:0x3ffba2e0 0x40089056:0x3ffba300 0x40087cc6:0x3ffba380 0x400862ec:0x3ffba3a0 0x4008629e:0xfe80d3e8 |<-CORRUPTED
xtensa-esp32-elf-addr2line -pfiaC -e /Users/clementtongpersonal/Desktop/Programming/2022/safelybuoy/safelybuoyv2/build/safelybuoyv2.elf 0x400819aa: [Errno 2] No such file or directory: 'xtensa-esp32-elf-addr2line'
xtensa-esp32-elf-addr2line -pfiaC -e /Users/clementtongpersonal/Desktop/Programming/2022/safelybuoy/safelybuoyv2/build/safelybuoyv2.elf 0x400861f1: [Errno 2] No such file or directory: 'xtensa-esp32-elf-addr2line'
xtensa-esp32-elf-addr2line -pfiaC -e /Users/clementtongpersonal/Desktop/Programming/2022/safelybuoy/safelybuoyv2/build/safelybuoyv2.elf 0x40089056: [Errno 2] No such file or directory: 'xtensa-esp32-elf-addr2line'
xtensa-esp32-elf-addr2line -pfiaC -e /Users/clementtongpersonal/Desktop/Programming/2022/safelybuoy/safelybuoyv2/build/safelybuoyv2.elf 0x40087cc6: [Errno 2] No such file or directory: 'xtensa-esp32-elf-addr2line'
xtensa-esp32-elf-addr2line -pfiaC -e /Users/clementtongpersonal/Desktop/Programming/2022/safelybuoy/safelybuoyv2/build/safelybuoyv2.elf 0x400862ec: [Errno 2] No such file or directory: 'xtensa-esp32-elf-addr2line'
xtensa-esp32-elf-addr2line -pfiaC -e /Users/clementtongpersonal/Desktop/Programming/2022/safelybuoy/safelybuoyv2/build/safelybuoyv2.elf 0x4008629e: [Errno 2] No such file or directory: 'xtensa-esp32-elf-addr2line'

ELF file SHA256: e79dd4d7af3d5b41

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6664
load:0x40078000,len:14848
xtensa-esp32-elf-addr2line -pfiaC -e /Users/clementtongpersonal/Desktop/Programming/2022/safelybuoy/safelybuoyv2/build/safelybuoyv2.elf 0x40078000: [Errno 2] No such file or directory: 'xtensa-esp32-elf-addr2line'
load:0x40080400,len:3792
xtensa-esp32-elf-addr2line -pfiaC -e /Users/clementtongpersonal/Desktop/Programming/2022/safelybuoy/safelybuoyv2/build/safelybuoyv2.elf 0x40080400: [Errno 2] No such file or directory: 'xtensa-esp32-elf-addr2line'
entry 0x40080694
xtensa-esp32-elf-addr2line -pfiaC -e /Users/clementtongpersonal/Desktop/Programming/2022/safelybuoy/safelybuoyv2/build/safelybuoyv2.elf 0x40080694: [Errno 2] No such file or directory: 'xtensa-esp32-elf-addr2line'
```

I made sure to run both install.sh and export.sh. Any advice on how to fix this is appreciated. Many thanks.

tongclement, 2023-01-02:
Just for reference, what I did to cause this error is to initiate both UART_1 and UART_2, by rerouting UART_1 to GPIO 2 and 4. Many thanks

tongclement, 2023-01-02:
Ok to confirm, the xtensa-esp32-elf-gcc not found error is still there after I do a full clean. To build it, I have to go to the integrated terminal, and run the install sh and export sh every time.

Following your advice about 'I recommend to check the IDF_PATH environment variable to ensure it points to the right version of ESP-IDF (in case you have multiple copies on your computer', I don't seem to find a IDF_PATH environment variable inside the ./zprofile 
Where would this IDF_PATH variable be located instead? Inside VS Code's preferences? In another directory? 
Instead, the zprofile (I am using the standard zsh on Mac) contains the following PATHs:
```
##Manual esp idf 2021-12-25
alias get_idf='. $HOME/esp/esp-idf/export.sh'
#Finished manual esp idf

##Manual esp idf 2023-01-02
export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin
``

Many thanks for your kind help as I have been stuck over the past few days

tongclement, 2023-01-02:
More related errors when building 
```
 *  Executing task: ninja  

[1/8] Performing build step for 'bootloader'
[1/1] cd /Users/clementtongpersonal/Desktop/Programming/2022/[projectname]/[projectname]v2/build/bootloader/esp-idf/esptool_py && /Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/clementtongpersonal/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /Users/clementtongpersonal/Desktop/Programming/2022/[projectname]/[projectname]v2/build/bootloader/bootloader.bin
Bootloader binary size 0x6330 bytes. 0xcd0 bytes (11%) free.
[2/6] Generating ld/sections.ld
FAILED: esp-idf/esp_system/ld/sections.ld 
cd /Users/clementtongpersonal/Desktop/Programming/2022/[projectname]/[projectname]v2/build/esp-idf/esp_system && /Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/clementtongpersonal/esp/esp-idf/tools/ldgen/ldgen.py --config /Users/clementtongpersonal/Desktop/Programming/2022/[projectname]/[projectname]v2/sdkconfig --fragments /Users/clementtongpersonal/esp/esp-idf/components/esp_ringbuf/linker.lf     /Users/clementtongpersonal/esp/esp-idf/components/driver/linker.lf  /Users/clementtongpersonal/esp/esp-idf/components/esp_pm/linker.lf  /Users/clementtongpersonal/esp/esp-idf/components/spi_flash/linker.lf       /Users/clementtongpersonal/esp/esp-idf/components/esp_gdbstub/linker.lf     /Users/clementtongpersonal/esp/esp-idf/components/espcoredump/linker.lf     /Users/clementtongpersonal/esp/esp-idf/components/esp_phy/linker.lf /Users/clementtongpersonal/esp/esp-idf/components/esp_system/linker.lf      /Users/clementtongpersonal/esp/esp-idf/components/esp_system/app.lf /Users/clementtongpersonal/esp/esp-idf/components/hal/linker.lf     /Users/clementtongpersonal/esp/esp-idf/components/esp_event/linker.lf       /Users/clementtongpersonal/esp/esp-idf/components/esp_wifi/linker.lf/Users/clementtongpersonal/esp/esp-idf/components/lwip/linker.lf        /Users/clementtongpersonal/esp/esp-idf/components/log/linker.lf     /Users/clementtongpersonal/esp/esp-idf/components/heap/linker.lf    /Users/clementtongpersonal/esp/esp-idf/components/soc/linker.lf     /Users/clementtongpersonal/esp/esp-idf/components/esp_hw_support/linker.lf  /Users/clementtongpersonal/esp/esp-idf/components/xtensa/linker.lf  /Users/clementtongpersonal/esp/esp-idf/components/esp_common/common.lf      /Users/clementtongpersonal/esp/esp-idf/components/esp_common/soc.lf /Users/clementtongpersonal/esp/esp-idf/components/freertos/linker.lf        /Users/clementtongpersonal/esp/esp-idf/components/newlib/newlib.lf  /Users/clementtongpersonal/esp/esp-idf/components/newlib/system_libs.lf     /Users/clementtongpersonal/esp/esp-idf/components/app_trace/linker.lf       /Users/clementtongpersonal/esp/esp-idf/components/bt/linker.lf      /Users/clementtongpersonal/esp/esp-idf/components/esp_lcd/linker.lf --input /Users/clementtongpersonal/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in --output /Users/clementtongpersonal/Desktop/Programming/2022/[projectname]/[projectname]v2/build/esp-idf/esp_system/ld/sections.ld --kconfig /Users/clementtongpersonal/esp/esp-idf/Kconfig --env-file /Users/clementtongpersonal/Desktop/Programming/2022/[projectname]/[projectname]v2/build/config.env --libraries-file /Users/clementtongpersonal/Desktop/Programming/2022/[projectname]/[projectname]v2/build/ldgen_libraries --objdump /usr/bin/objdump
Traceback (most recent call last):
  File "/Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/pyparsing.py", line 1548, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/pyparsing.py", line 2622, in parseImpl
    if (instring[loc] == self.firstMatchChar and
IndexError: string index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/clementtongpersonal/esp/esp-idf/tools/ldgen/entity.py", line 136, in add_sections_info
    results = parser.parseString(first_line, parseAll=True)
  File "/Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/pyparsing.py", line 1814, in parseString
    raise exc
  File "/Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/pyparsing.py", line 1804, in parseString
    loc, tokens = self._parse( instring, 0 )
  File "/Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/pyparsing.py", line 1548, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/pyparsing.py", line 3705, in parseImpl
    loc, resultlist = self.exprs[0]._parse( instring, loc, doActions, callPreParse=False )
  File "/Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/pyparsing.py", line 1548, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/clementtongpersonal/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/pyparsing.py", line 4067, in parseImpl
    return self.expr._parse( instring, loc, doActions, callPreParse=False )
[... cut ...]
  File "/Users/clementtongpersonal/esp/esp-idf/tools/ldgen/entity.py", line 138, in add_sections_info
    raise ParseException('Parsing sections info for library ' + sections_info_dump.name + ' failed. ' + p.msg)
pyparsing.ParseException: Parsing sections info for library /Users/clementtongpersonal/Desktop/Programming/2022/[projectname]/[projectname]v2/build/esp-idf/esp_ringbuf/libesp_ringbuf.a failed. Expected "In archive" (at char 0), (line:1, col:1)
ninja: build stopped: subcommand failed.

 *  The terminal process "/bin/zsh '-c', 'ninja '" terminated with exit code: 1. 

```

tongclement, 2023-01-02:
So the conclusion is something is wrong with my path - any guidance on fixing the path will be appreciated - many thanks

tongclement, 2023-01-02:
Here are some vs code settings that might be related:
USER: Idf: Custom Extra Paths
Paths to be appended to PATH
(Also modified in workspace)
`/Users/clementtongpersonal/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/Users/clementtongpersonal/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin:/Users/clementtongpersonal/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:/Users/clementtongpersonal/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin:/Users/clementtongpersonal/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Users/clementtongpersonal/.espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Users/clementtongpersonal/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin:/Users/clementtongpersonal/.espressif/tools/ninja/1.10.2`

WORKSPACE:
`/Users/clementtongpersonal/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/Users/clementtongpersonal/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/Users/clementtongpersonal/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/Users/clementtongpersonal/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/Users/clementtongpersonal/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/clementtongpersonal/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/clementtongpersonal/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin`

Ah - maybe this is the problem. The workspace and user have paths that don't match.

tongclement, 2023-01-02:
[1/3/2023, 1:20:49 AM] Unable to resolve configuration with compilerPath "/Users/clementtongpersonal/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc".  Using "/opt/homebrew/bin/gcc-11" instead.
[1/3/2023, 1:20:49 AM] Unable to resolve configuration with compilerPath "/Users/clementtongpersonal/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc".  Using "/opt/homebrew/bin/gcc-11" instead.

So I changed the WORKSPACE path inside VS Code but the error is still here and the building still doesn't work.  Some guidance on where the compiler path is will be helpful. many thanks for your help @dobairoland

tongclement, 2023-01-02:
Ok found it
It was in .vscode -> c_cpp_properties.json -> configurations -> 0 -> compiler path
Took me 2 days to figure it out ... 
Lets pray that it works.

tongclement, 2023-01-02:
Lets go - it worked.
However, the reason behind this path problem is unknown.
I uninstalled and installed the espressif vs code extension multiple times, deleted the ~/esp/esp-idf folder multiple times and reinstalled using the express option inside the vs code extension, and also deleted the .espressif folder and reran install.sh and export.sh 
Or am I supposed to do something to each esp project after updating the idf? I don't know

Recommendation:
- If this VS Code variable didn't change automatically as it is supposed to after an re-installation/update, please see if it can be fixed - or if it is my fault please let me know/put it in the docs about updating idf versions
- The update process can be much easier than this so everyone can use the latest version without too much hassle and you guys don't have to deal with debugging and answering questions about older versions as much. 
- For example: a one click "uninstaller" or an update command in the ESP-IDF command line

@dobairoland Thanks so much for your help, especially on pointing me to the direction that this might be due to environment variables and paths. Have a happy new year and hopefully the update process will be documented better/will be easier in future releases. Happy 2023 :)

tongclement, 2023-01-02:
Not fixed

dobairoland, 2023-01-03:
Thanks @tongclement for your notes on this issue. I don't think that anything could be done on the ESP-IDF side. Perhaps in the linked vscode issue the developers will be able to improve this.

ESP-IDF does not change the PATH permanently. Changes to environment variables are discarded after the terminal is closed. PATH is updated by adding items to the beginning, therefore, globally set values are not affecting it. However, we cannot handle VScode-related workspace changes at this level. I'm closing this for this reason.

closed 2023-01-03 by dobairoland