espressif/esp-idf #10474: esp_lcd_panel_draw_bitmap() does not display color in the correct location.

hyu7000, 2023-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0 7f5ecbe

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32s3 wroom 1 devkit

### Power Supply used.

USB

### What is the expected behavior?

lcd pixel size is 240 * 280

all pixel is display by color which i set in firmware without missing pixel

### What is the actual behavior?

range_1 : x_start : 0, y_start : 0, x_end : 240, y_end : 20
range_2 : x_start : 0, y_start : 0, x_end : 240, y_end : 280
range_3 : x_start : 0, y_start : 0, x_end : 240, y_end : 300

color is black

when i set range 1, lcd was display like below image.

[image]

when i set range 2, lcd was display like below image.

[image]

when i set range 2, lcd was display like below image.

[image]

### Steps to reproduce.

1. Step
add this code

```
uint16_t *background;

background = (uint16_t *)malloc(EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES); 

for(int i = 0; i < EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES; i++)
{
    background[i] = 0x0000;
}

esp_lcd_panel_draw_bitmap(panel_handle, 0, 0, EXAMPLE_LCD_H_RES, EXAMPLE_LCD_V_RES, background);
```

2. Step
EXAMPLE_LCD_H_RES 240
EXAMPLE_LCD_V_RES 280

### More Information.

it seem to be an offset. i think that esp-idf has not function to adjust this offset.

when i use Adafruit_ST7789 lib, my st7789 lcd was work fine without offset.

kriegste, 2023-01-02:
If it is an offset, you could try esp_lcd_panel_set_gap.

suda-morris, 2023-01-03:
Hi @hyu7000 You can refer to [this](https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/test_apps/spi_lcd/main/test_spi_lcd_panel.c#L81) as well.

closed 2023-01-03 by Alvin1Zhang

hyu7000, 2023-01-03:
I think I didnt checked the document properly.

thank