espressif/esp-idf #10475: Command `flash` fails with port option

harryzcy, 2023-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2658-g0025915dc4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It show flash successfully

### What is the actual behavior?

Prints out error `Option "port" provided for "flash" is already defined to a different value. This option can appear at most once in the command line.`

### Steps to reproduce.

1. Use `hello-world` from examples
2. Set target and run `idf.py build`
3. Run `idf.py flash`

### More Information.

I'm using ESP32-C3-DevKitM-1 and flashing using it's native usb port starts with `/dev/cu.usbserial-*`

dobairoland, 2023-01-02:
Hi @harryzcy. Could you please try to run `idf.py fullclean` in your project and then `idf.py set-target esp32c3`, `idf.py build`, and share the full log of `idf.py flash`.

I'm setting `ESPPORT` as well but cannot reproduce the issue.

harryzcy, 2023-01-02:
@dobairoland maybe I didn't explain clearly.

I'm getting the error when I use `idf.py -p /dev/cu.usbserial-1130 flash`, and when my `ESPPORT` is set to a different value `/dev/cu.SLAB_USBtoUART`

[image]

dobairoland, 2023-01-02:
Thanks @harryzcy. I've got it now. 

For our reference, here is how to reproduce this:
```
ESPPORT=/dev/ttyUSB1 idf.py monitor -p /dev/ttyUSB0
```
The issue is not present when the same value is specified with the argument and the environment variable.

dobairoland, 2023-01-03:
After looking into this more, I think the current behavior is correct. Changing this could lead to confusion when one forgets about that have set the environment variable in some of environment scripts. I agree that running `idf.py flash -p <port>` in that case could be clear but what about `idf.py flash`? One would expect port autodetection but instead only ESPPORT would be tested. Finding the root cause in this case could take long.  

However, this failure will at least remind one to not set ESPPORT in environment script if it is not the same one is desired to use every time.

I'll improve the error message because it is not clear in this situation.

closed 2023-01-13 by espressif-bot