espressif/esp-idf #10476: How does Provisioner know if a node in the mesh network is offline

Vanperdung, 2023-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In section 1.19 in the [FAQ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-ble-mesh/ble-mesh-faq.html), it talk about how to detect an offline node in the mesh network.
How can Node send Heartbeat to Provisioner? I haven't found API related to it yet.

Vanperdung, 2023-01-04:
Does anyone help me?

JiaLWang, 2023-01-04:
Hi @Vanperdung ，

You can send `config heartbeat publication set` message to enable the node send heartbeats.

Vanperdung, 2023-01-04:
Can you explain why this log "Ignoring old SeqAuth" appear? When I provision many Nodes, this log appears.

[image]

JiaLWang, 2023-01-04:
Means the node received an old SeqAuth packet and it will be discarded.

Vanperdung, 2023-01-07:
When Provisioner sends Config heartbeat publication set, Node print "BLE_MESH: Ignoring heartbeat to non-subscribed destination". Did I do it wrong? Here is the code of provisioner: 
[image]

Vanperdung, 2023-01-07:
I fixed it. But where is the event ESP_BLE_MESH_MODEL_OP_HEARTBEAT_PUB_SET handled? I see it doesn't handle in config server callback.

closed 2023-01-10 by Vanperdung

forx157, 2023-01-13:
Hi @Vanperdung ,
function `heartbeat_pub_set` in the `cfg_srv.c` will be trigged when a node receive the `ESP_BLE_MESH_MODEL_OP_HEARTBEAT_PUB_SET `, and update the heartbeat attribute  which was filled in the pub set msg you send. 
So the key issues have been handled internally, we don't need to handle it.
in addition, ESP_BLE_MESH_MODEL_OP_HEARTBEAT_PUB_SET is a opcode not a event.

Vanperdung, 2023-01-13:
Hi @forx157 
Thanks for your answer.
Now I want nodes to be able to know when the provisioner is offline. I plan to have the provisioner publish to a group periodically and nodes subscribe to this group. Is there any other way?

forx157, 2023-01-14:
@Vanperdung 
It depends on that how the provisioner goes to offline.
If it  is intentionally, you can send a msg to notify other node.
If it is unexpected, sending heatbeat msg periodically this is the only way I can think of.

Vanperdung, 2023-01-14:
@forx157 
How can Provisioner send heartbeat to Nodes?

forx157, 2023-01-15:
Hi @Vanperdung .
As mentioned above.
Node should use config client model send a heartbeat publication set message to provisioner to enable and set some attributes of  heartbeat, such as published destination, period, count and so on.
Then provisioner node's config server model  will handle that message, and start to send heartbeat message with the destination, period, and some other attributes which you set in the heartbeat set message.
Finally a node which could receive that destination the heartbeat send to will trigger a event `ESP_BLE_HEARTBEAT_MESSAGE_RECV_EVT` in the `bt_mesh_prov_cb` function when that node receive a heartbeat message.

lebrislo, 2023-12-19:
> When Provisioner sends Config heartbeat publication set, Node print "BLE_MESH: Ignoring heartbeat to non-subscribed destination". Did I do it wrong? Here is the code of provisioner: [image]

Hi,

I have the same issue as you had "BLE_MESH: Ignoring heartbeat to non-subscribed destination". I see that you fixed that point. How did you manage to do so ?