espressif/esp-idf #10479: in state connected by ble5.0, backlight of st7789 lcd is blink.

hyu7000, 2023-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0 7f5ecbe

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32s3 wroom 1 devkit

### Power Supply used.

USB

### What is the expected behavior?

in state connected by ble5.0, st7789 lcd remains turned off.

### What is the actual behavior?

in state connected by ble5.0, st7789 lcd repeat turn on and turn off.

if i change connection interval, repeat cycle of onoff was changed.

the below video show when connection interval is 500ms.
https://user-images.githubusercontent.com/82933946/210317688-ec8c64f7-82db-4036-a15f-87244e0df73a.mp4

the below video show when connection interval is 100ms.
https://user-images.githubusercontent.com/82933946/210318034-95402d35-cacc-4f5e-9063-18e3167a3ef5.mp4

if it not connected ble, lcd completly turn off.

### Steps to reproduce.

1. Step
 in ble50_security_server project, set bluetooth modem sleep 
and add this code at `gatts_profile_event_handler()`

```
case ESP_GATTS_CONNECT_EVT:
            ESP_LOGI(GATTS_TABLE_TAG, "ESP_GATTS_CONNECT_EVT");
            /* start security connect with peer device when receive the connect event sent by the master */
            esp_ble_set_encryption(param->connect.remote_bda, ESP_BLE_SEC_ENCRYPT_MITM);
            
            memcpy(conn_params.bda, param->connect.remote_bda, sizeof(esp_bd_addr_t));
            conn_params.latency = 0;
            conn_params.max_int = 0x50;   
            conn_params.min_int = 0x50;    
            conn_params.timeout = 400;    

            esp_ble_gap_update_conn_params(&conn_params);

            break;
```

2. Step
 connect st7789 lcd by spi connection.
 and add this code at `app_main()`

```
#define LCD_PIN_NUM_BK_LIGHT    7
...
ESP_ERROR_CHECK(esp_lcd_panel_disp_on_off(panel_handle, false)); 
 gpio_set_level(LCD_PIN_NUM_BK_LIGHT, false);
```

3. Step
 try to connect to a phone

### More Information.

gpio7 of esp32s3 was connected backlight pin(BLK) like below image

[image]

i can't find solution to solve this issue in the document and esp32 forum, git.

is there keyword to find solution?

suda-morris, 2023-01-03:
Maybe you need to attach an oscilloscope to the GPIO7 to see what happens on that GPIO. Is it possible that the power supply is not sufficient?

hyu7000, 2023-01-03:
@suda-morris when i use oscilloscope, this image was shown

[image]

then if i disconnect ble, graph indicated 0V.

i don't know why it's 3.3v if it's connected ble.

I think power is sufficient. It shows the same symptoms when i use power supply or usb.

when i use esp32-arduino, though it connected, lcd did not blink.

suda-morris, 2023-01-04:
what's the log output? is your program crashed somehow? are you sure GPIO7 is not used by other purpose in your program?

hyu7000, 2023-01-04:
@suda-morris there is no shut down and crash ,disconnection, other config of GPIO7.

though it blink st7789 lcd, BLE work well . but the issus is st7789 lcd

and if i except for code related to BLE and upload, display work well without blink.

this log is in vscode monitor.
```
I (43) boot: ESP-IDF v5.0-dirty 2nd stage bootloader
I (43) boot: compile time 14:30:45
I (43) boot: chip revision: v0.1
I (45) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (52) boot.esp32s3: Boot SPI Speed : 80MHz
I (57) boot.esp32s3: SPI Mode       : DIO
I (62) boot.esp32s3: SPI Flash Size : 8MB
I (67) boot: Enabling RNG early entropy source...
I (72) boot: Partition Table:
I (76) boot: ## Label            Usage          Type ST Offset   Length
I (83) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (90) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (98) boot:  2 app0             OTA app          00 10 00010000 00300000
I (105) boot:  3 spiffs           Unknown data     01 82 00310000 000f0000
I (113) boot: End of partition table
I (117) boot_comm: chip revision: 1, min. application chip revision: 0
I (124) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=24160h (147808) map
I (149) esp_image: segment 1: paddr=00034188 vaddr=3fc99800 size=03ff0h ( 16368) load
I (152) esp_image: segment 2: paddr=00038180 vaddr=40374000 size=07e98h ( 32408) load
I (160) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=80f88h (528264) map
I (223) esp_image: segment 4: paddr=000c0fb0 vaddr=4037be98 size=0d8bch ( 55484) load
I (232) esp_image: segment 5: paddr=000ce874 vaddr=50000000 size=00010h (    16) load
I (232) esp_image: segment 6: paddr=000ce88c vaddr=600fe000 size=00034h (    52) load
I (246) boot: Loaded app from partition at offset 0x10000
I (246) boot: Disabling RNG early entropy source...
I (260) cpu_start: Pro cpu up.
I (260) cpu_start: Starting app cpu, entry point is 0x403755f4
0x403755f4: call_start_cpu1 at C:/Users/hk910/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (325) cpu_start: Pro cpu start user code
I (325) cpu_start: cpu freq: 160000000 Hz
I (325) cpu_start: Application information:
I (328) cpu_start: Project name:     CremakerWatch
I (333) cpu_start: App version:      1
I (338) cpu_start: Compile time:     Jan  3 2023 14:31:09
I (344) cpu_start: ELF file SHA256:  d407910d28f4afa8...
I (350) cpu_start: ESP-IDF:          v5.0-dirty
I (355) heap_init: Initializing. RAM available for dynamic allocation:
I (362) heap_init: At 3FCA1B20 len 00047BF0 (286 KiB): D/IRAM
I (369) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (375) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (382) heap_init: At 600FE034 len 00001FCC (7 KiB): RTCRAM
I (389) spi_flash: detected chip: generic
I (393) spi_flash: flash io: dio
I (408) sleep: Configure to isolate all GPIO pins in sleep state
I (408) coexist: coexist rom version e7ae62f
I (409) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (437) pm: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 40, Light sleep: ENABLED
I (437) sleep: Enable automatic switching of GPIO sleep configuration
I (447) sleep: Code start at 42000020, total 515.88 KiB, data start at 3c090020, total 144.34 KiB
W (457) BT_INIT: esp_bt_controller_mem_release not implemented, return OK
I (467) BT_INIT: BT controller compile version [76c24c9]
I (477) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (517) system_api: Base MAC address is not set
I (517) system_api: read default base MAC address from EFUSE
I (517) BT_INIT: Bluetooth MAC: 68:b6:b3:29:af:ca

I (527) BLE: ble_init init bluetooth
I (587) BLE: ESP_GATTS_REG_EVT
I (597) BLE: The number handle = 8
[... 46 lines cut ...]
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Central Address Resolution [R] (0x2AA6)
Unknown Service (000000ff-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N R W] (0000ff01-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R] (0000ff02-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [W] (0000ff03-0000-1000-8000-00805f9b34fb)
D	22:05:31.743	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	22:05:31.745	gatt.setCharacteristicNotification(0000ff01-0000-1000-8000-00805f9b34fb, true)
I	22:05:31.823	Connection parameters updated (interval: 100.0ms, latency: 0, timeout: 4000ms)
D	22:05:32.172	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDED (12)
I	22:05:32.172	Device bonded
```

when i add this code, st7789 LCD no longer blink.

`gpio_sleep_sel_dis(LCD_PIN_NUM_BK_LIGHT);`

but i don't know what role _SLP_SEL_ play.

I'm not sure if it's okay to use this function.

hyu7000, 2023-01-05:
when i add `gpio_sleep_sel_dis(LCD_PIN_NUM_BK_LIGHT);` and `gpio_sleep_sel_en(LCD_PIN_NUM_BK_LIGHT);`, it work well.

but, current cumsumption is increasing. ( about 0.1mA -> about  0.3mA )

suda-morris, 2023-01-05:
Okay, now I see it's expected behavior because you enabled the `CONFIG_PM_SLP_DISABLE_GPIO` option, which will isolate your LCD backlight GPIO in the `esp_sleep_config_gpio_isolate` function. If you call `gpio_sleep_sel_dis` for that GPIO, `esp_sleep_config_gpio_isolate` will have no effect on that GPIO.

hyu7000, 2023-01-07:
if `gpio_sleep_sel_dis();` is enable, power comsumption is higher than i expect.

so i tested to reduce current consumption without `gpio_sleep_sel_dis();`

i found that when connection interval is in 10~35ms, it is work well without `gpio_sleep_sel_dis();`

at this time, there is no blink.

but when connection interval is over than 40ms, it blink again.

and i want to increase interval to reduce power consumption.

esp-cjh, 2024-02-26:
@hyu7000 Does this issue still exist in the latest IDF version?

closed 2024-12-17 by Alvin1Zhang