espressif/esp-idf #10480: ESP32S3/S2 RISC-V ULP touch read example

MacWyznawca, 2023-01-03:
### Is your feature request related to a problem?

The ESP32S3 and S2 are unable to support wake-up with more than one touchpad.

### Describe the solution you'd like.

However, the ULP RISC-V is able to read Touch registers and wake up the chip when any of them are touched. 

An example with such a solution would be very helpful.

MacWyznawca, 2023-05-10:
Hi! @sudeep-mohanty! Any progress with this example? Please!!! ;-)

MacWyznawca, 2023-05-16:
Thank you @sudeep-mohanty. 
This is very important for our company LuON.me, because handling touch wake-up with only one input heavily complicates the code, if we need to handle multiple touch fields. 
This increases response time and decreases repeatability.
That's why we're so looking forward to an example of Touch support in ULP to speed up and stabilize battery-powered touch buttons

sudeep-mohanty, 2023-05-24:
Hello @MacWyznawca,
Apologies once again for a delayed reply. We have started work on adding touch sensor driver support for the ULP RISC-V. In the meantime, please find attached a basic example of the touch sensor working with the ULP RISC-V core. This hasn't been tested thoroughly and is likely to have changes before it is upstreamed to GitHub but it would present the idea of how the touch peripheral would work with the ULP RISC-V core. I would encourage you to try it out and let us know your feedback on it. 
[ulp_touch.zip](https://github.com/espressif/esp-idf/files/11554760/ulp_touch.zip)

MacWyznawca, 2023-05-24:
Thank you very much @sudeep-mohanty, the example works. I thought it would be more difficult (reading registers directly).
However, there is another problem. The current consumption is at a suspiciously high level. It is ~300 uA. 
Our projects with ULP analog measurement and Touch Sleep enabled draw ~80-100uA in total (ESP32-S3 with other components and sensors). 
Where to look for a solution?

MacWyznawca, 2023-05-24:
OK, I understand @sudeep-mohanty 
For now, I'm coping by using TOUCH_FSM_MODE_SW and starting the reading before the measurement. I've managed to get down to 70-90uA (standard Touch sensor settings), because waiting for the readout also costs some energy.
If you have a better idea, I'll be glad ;-)

leeebo, 2023-06-07:
Hi @MacWyznawca Back to the original question `The ESP32S3 and S2 are unable to support wake-up with more than one touchpad.`  there is another solution to work around it, you can call `touch_pad_config`  to enable the channels you want to use for deep/light sleep wakeup and call `esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);` to enable the power for touch, then all the channels you configured can wakeup the chip

Do not call `xx_sleep_channel_xx` at all in this solution, and please keep in mind the more channels enabled, the more frequent the overall external charge and discharge, resulting in higher power consumption.

closed 2023-06-09 by espressif-bot