espressif/esp-idf #10482: Processor waked up because of  ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG

yangdx2022, 2023-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2658-g0025915dc4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-S3-32S-kit

### Power Supply used.

USB

### What is the expected behavior?

When GPIO17 reseted, ULP start to work and running cnt increases, when running cnt  meet conditions. processor will be waked up 

### What is the actual behavior?

processor is waked up because of ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG

### Steps to reproduce.

1. Step
set GPIO17 to Low Votage
2. Step
repeat step 1
3. Step
wait ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG ocurred

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11a8
load:0x403c9700,len:0xa50
load:0x403cc700,len:0x2c44
entry 0x403c9894
W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
Initializing RTC I2C ...
after init i2c
10 17 
initting ulp program
sleeping ... ...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
pro cpu reset by JTAG
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11a8
load:0x403c9700,len:0xa50
load:0x403cc700,len:0x2c44
entry 0x403c9894
W (88) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
reg : a4 12 
running cnt : 3
ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG
sleeping ... ...
```

### More Information.

`/*
 * SPDX-FileCopyrightText: 2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */
/* ULP RISC-V RTC I2C example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include <stdio.h>
#include <math.h>
#include "esp_sleep.h"
#include "ulp_riscv.h"
#include "ulp_riscv_i2c.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ulp_main.h"
#include "bmp180_defs.h"
#include "esp_sleep.h"
#include "soc/sens_reg.h"
#include "driver/gpio.h"
#include "ulp_riscv.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[]   asm("_binary_ulp_main_bin_end");

/************************************************
 * ULP utility APIs
 ************************************************/
static void init_ulp_program(void);

/************************************************
 * RTC I2C utility APIs
 ************************************************/
static void init_i2c(void);

static void wakeup_gpio_init(void)
{
    /* Configure the button GPIO as input, enable wakeup */
    gpio_config_t config;
    config.pin_bit_mask = BIT64(GPIO_NUM_17);
    config.mode = GPIO_MODE_INPUT;
    config.pull_up_en = true;
    config.pull_down_en = false;
    config.intr_type = GPIO_INTR_DISABLE;

    ESP_ERROR_CHECK(gpio_config(&config));
    ESP_ERROR_CHECK(gpio_wakeup_enable(GPIO_NUM_17, GPIO_INTR_LOW_LEVEL));
    ESP_ERROR_CHECK(gpio_hold_en(GPIO_NUM_17));

}

void app_main(void)
{

    esp_sleep_wakeup_cause_t cause;
    esp_err_t err;

    switch (cause = esp_sleep_get_wakeup_cause()) {
    case ESP_SLEEP_WAKEUP_ULP:
        printf("running cnt : %d\n", ulp_running_cnt);
        printf("reg : ");
        for(int i = 0; i < 2; i += 1) printf("%02x ", ((uint8_t *) &ulp_data)[i]);
        printf("\n");
        break;

    case ESP_SLEEP_WAKEUP_UNDEFINED:
        init_i2c();
        wakeup_gpio_init();
        printf("after init i2c\n");
        ulp_riscv_i2c_master_set_slave_addr(0x32);
        ulp_riscv_i2c_master_set_slave_reg_addr(0x00);
        ulp_riscv_i2c_master_read_from_device(((uint8_t *) &ulp_data), 2);
        for(int i = 0; i < 2; i += 1) printf("%02x ", ((uint8_t *) &ulp_data)[i]);
        printf("\n");
        init_ulp_program();
        break;

    case ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG:
        printf("reg : ");
        for(int i = 0; i < 2; i += 1) printf("%02x ", ((uint8_t *) &ulp_data)[i]);
        printf("\n");
        printf("running cnt : %d\n", ulp_running_cnt);
        printf("ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG\n");
        break;

    default:
        printf("running cnt : %d\n", ulp_running_cnt);
        printf("Unknown case : %d\n", cause);
        break;
    }

    printf("sleeping ... ...\n");
    ESP_ERROR_CHECK(err = esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON));
    ESP_ERROR_CHECK(err = esp_sleep_enable_ulp_wakeup());
    esp_deep_sleep_start();

}

static void init_i2c(void)
{
    /* Configure RTC I2C */
    esp_err_t err;
    printf("Initializing RTC I2C ...\n");
    ulp_riscv_i2c_cfg_t i2c_cfg = ULP_RISCV_I2C_DEFAULT_CONFIG();
    
    i2c_cfg.i2c_pin_cfg.sda_io_num = GPIO_NUM_1;
    i2c_cfg.i2c_pin_cfg.scl_io_num = GPIO_NUM_2;
    i2c_cfg.i2c_pin_cfg.sda_pullup_en = true;
    i2c_cfg.i2c_pin_cfg.scl_pullup_en = true;

    ESP_ERROR_CHECK(err = ulp_riscv_i2c_master_init(&i2c_cfg));
    vTaskDelay(pdMS_TO_TICKS(1000));
}

static void init_ulp_program(void)
{
    esp_err_t err;
    ulp_riscv_cfg_t cfg;

    printf("initting ulp program\n");

    cfg.wakeup_source = ULP_RISCV_WAKEUP_SOURCE_GPIO;
    
    ESP_ERROR_CHECK(err = ulp_riscv_load_binary(ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start)));
    ESP_ERROR_CHECK(err = ulp_riscv_config_and_run(&cfg));

}
`

yangdx2022, 2023-01-03:
this is ulp source
`
#include <stdint.h>
#include "ulp_riscv.h"
#include "ulp_riscv_utils.h"
#include "ulp_riscv_i2c_ulp_core.h"

uint32_t running_cnt = 0;
uint8_t data[2];

int main (void) {

    running_cnt += 1;
    memset(data, 0x0f, 2);

    ulp_riscv_i2c_master_set_slave_addr(0x32);
    ulp_riscv_i2c_master_set_slave_reg_addr(0x0f);
    ulp_riscv_i2c_master_read_from_device(data, 2);

    ulp_riscv_delay_cycles(10 * 1000 * ULP_RISCV_CYCLES_PER_US);
    ulp_riscv_gpio_wakeup_clear();

    if(0 == (running_cnt & 0x03)) ulp_riscv_wakeup_main_processor();
    // ulp_riscv_wakeup_main_processor();
    ulp_riscv_halt();
    return 0;

}
`
do you need any extra information? @sudeep-mohanty 
Thanks for your help!

sudeep-mohanty, 2023-01-05:
Hello @yangdx2022,
Thank you for reporting this issue to us.

I tried to reproduce the problem at my end but I wasn't able to. I had to make a few updates to get the code running but I have maintained the general structure of what you have shared. It is working on my end as expected. Here is a list of changes I made -
- I switched the GPIO wake pin to GPIO#0 just for convenience as on the esp32s3-devkitc board has a button connected to GPIO#0.
- Instead of using the I2C sensor with address 0x32, I used the BMP180 (address 0x77) and queried the chip id register at 0xD0. This returns the value of 0x55 which is as expected.

With the above changes, I could run the code and I do not see the main CPU being woken up by the `ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG` but rather by `ESP_SLEEP_WAKEUP_ULP` when the GPIO wake is triggered which is as expected.

Here is the code I used - [source](https://gist.github.com/sudeep-mohanty/942395710ede4800e45abf4679221c3c)

Here are the logs for the same -

```
--- idf_monitor on /dev/cu.usbserial-14110 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11a8
load:0x403c9700,len:0xa50
load:0x403cc700,len:0x2c44
entry 0x403c9894
W (88) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
Initializing RTC I2C ...
after init i2c
55
initting ulp program
sleeping ... ...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11a8
load:0x403c9700,len:0xa50
load:0x403cc700,len:0x2c44
entry 0x403c9894
W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
running cnt : 11
reg : 55
sleeping ... ...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11a8
load:0x403c9700,len:0xa50
load:0x403cc700,len:0x2c44
entry 0x403c9894
W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
running cnt : 15
reg : 55
sleeping ... ...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11a8
load:0x403c9700,len:0xa50
load:0x403cc700,len:0x2c44
entry 0x403c9894
W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
running cnt : 23
reg : 55
sleeping ... ...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11a8
load:0x403c9700,len:0xa50
load:0x403cc700,len:0x2c44
entry 0x403c9894
W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
running cnt : 31
reg : 55
sleeping ... ...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11a8
load:0x403c9700,len:0xa50
load:0x403cc700,len:0x2c44
entry 0x403c9894
W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
running cnt : 39
reg : 55
sleeping ... ...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x11a8
load:0x403c9700,len:0xa50
load:0x403cc700,len:0x2c44
entry 0x403c9894
W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
running cnt : 46
reg : 55
sleeping ... ...
```

Could you give it another shot as well, maybe switch to GPIO#0 and retry? If you still see the trap signal being generated then could you try to pin-point at what point the ULP crashes? I wonder if there is some issue with the I2C read that is being done on the ULP core.

yangdx2022, 2023-01-07:
> Hello @yangdx2022, Thank you for reporting this issue to us.
>
> I tried to reproduce the problem at my end but I wasn't able to. I had to make a few updates to get the code running but I have [...]

thank you for your help, i has noticed that your udpate the master branch of esp idf, so i update my esp idf. in my application,  the rtc i2c is connectd to SD2068, this is a rtc ic. gpio 17 is binded to sd2068's interupt io, with the last esp idf, it still doesn't work. then i change the GPIO to GPIO0, and i use the key to control the voltage to GPIO 0 the frequency to ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG is down. but it still exist. the frequency of signal from sd2068's interupt io is 1mHz. do you need any other info.

yangdx2022, 2023-01-07:
> 您好@yangdx2022，感谢您向我们报告此问题。
> 
> 我试图在最后重现问题，但我无法做到。我必须进行一些更新才能使代码运行，但我保留了您共享的内容的一般结构。正如预期的那样，它正在为我工作。以下是我所做的更改列表 -
> 
> * 为了方便起见，我将 GPIO 唤醒引脚切换到 GPIO#0，因为在 esp32s3-devkitc 板上有一个连接到 GPIO#0 的按钮。
> * 我没有使用地址0x32的 I2C 传感器，而是使用 BMP180（地址 0x77）并在0xD0处查询芯片 ID 寄存器。这将返回0x55的值，该值符合预期。
> 
> 通过上述更改，我可以运行代码，我没有看到主 CPU 被唤醒，而是在 GPIO 唤醒时被唤醒，这是预期的。`ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG``ESP_SLEEP_WAKEUP_ULP`
> 
> 这是我使用的代码 - [来源](https://gist.github.com/sudeep-mohanty/942395710ede4800e45abf4679221c3c)
> 
> 这是相同的日志 -
> 
> ```
> --- idf_monitor on /dev/cu.usbserial-14110 115200 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x11a8
> load:0x403c9700,len:0xa50
> load:0x403cc700,len:0x2c44
> entry 0x403c9894
> W (88) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> Initializing RTC I2C ...
> after init i2c
> 55
> initting ulp program
> sleeping ... ...
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x11a8
> load:0x403c9700,len:0xa50
> load:0x403cc700,len:0x2c44
> entry 0x403c9894
> W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> running cnt : 11
> reg : 55
> sleeping ... ...
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x11a8
> load:0x403c9700,len:0xa50
> load:0x403cc700,len:0x2c44
> entry 0x403c9894
> W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> running cnt : 15
> reg : 55
> sleeping ... ...
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x11a8
> load:0x403c9700,len:0xa50
> load:0x403cc700,len:0x2c44
> entry 0x403c9894
> W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> running cnt : 23
> reg : 55
> sleeping ... ...
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x11a8
> load:0x403c9700,len:0xa50
> load:0x403cc700,len:0x2c44
> entry 0x403c9894
> W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> running cnt : 31
> reg : 55
> sleeping ... ...
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x11a8
> load:0x403c9700,len:0xa50
> load:0x403cc700,len:0x2c44
> entry 0x403c9894
> W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> running cnt : 39
> reg : 55
> sleeping ... ...
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x5 (DSLEEP),boot:0x2b (SPI_FAST_FLASH_BOOT)
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3810,len:0x11a8
> load:0x403c9700,len:0xa50
> load:0x403cc700,len:0x2c44
> entry 0x403c9894
> W (86) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> running cnt : 46
> reg : 55
> sleeping ... ...
> ```
> 
> 您能否再试一次，也许切换到 GPIO#0 并重试？如果您仍然看到正在生成陷阱信号，那么您能否尝试精确定位ULP崩溃的时间点？我想知道在ULP内核上完成的I2C读取是否存在一些问题。

i think the program look like that when ulp sleep enough time, the program will ocurr after ulp wake up without waking main processor

sudeep-mohanty, 2023-01-11:
Hello @yangdx2022,
Apologies for the delayed reply, I have finally been able to reproduce the problem at my end as well. The trap signal does get triggered occasionally. I will be investigating this further.

AxelLin, 2023-06-25:
Any update? @sudeep-mohanty

shangke1988, 2023-09-23:
I removed `ulp_riscv_wakeup_main_processor();` in example “gpio_interrupt”. After press button `GPIO#0` some times, 
 main processor wakuped by `ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG`.
```
const char *wakeup_str[] = {
    "ESP_SLEEP_WAKEUP_UNDEFINED, In case of deep sleep, reset was not caused by exit from deep sleep",
    "ESP_SLEEP_WAKEUP_ALL, Not a wakeup cause, used to disable all wakeup sources with esp_sleep_disable_wakeup_source",
    "ESP_SLEEP_WAKEUP_EXT0, Wakeup caused by external signal using RTC_IO",
    "ESP_SLEEP_WAKEUP_EXT1, Wakeup caused by external signal using RTC_CNTL",
    "ESP_SLEEP_WAKEUP_TIMER, Wakeup caused by timer",
    "ESP_SLEEP_WAKEUP_TOUCHPAD, Wakeup caused by touchpad",
    "ESP_SLEEP_WAKEUP_ULP, Wakeup caused by ULP program",
    "ESP_SLEEP_WAKEUP_GPIO, Wakeup caused by GPIO (light sleep only on ESP32, S2 and S3)",
    "ESP_SLEEP_WAKEUP_UART, Wakeup caused by UART (light sleep only)",
    "ESP_SLEEP_WAKEUP_WIFI, Wakeup caused by WIFI (light sleep only)",
    "ESP_SLEEP_WAKEUP_COCPU, Wakeup caused by COCPU int",
    "ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG, Wakeup caused by COCPU crash",
    "ESP_SLEEP_WAKEUP_BT, Wakeup caused by BT (light sleep only)",
};
#define delay_ms(t) vTaskDelay(((t) + portTICK_PERIOD_MS - 1) / portTICK_PERIOD_MS)
void app_main(void)
{
    esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();
    /* not a wakeup from ULP, load the firmware */
    if (cause == ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG)
    {
        // init_ulp_program();
        ulp_riscv_reset();
    }
    else if (cause != ESP_SLEEP_WAKEUP_ULP)
    {
        printf("Not a ULP-RISC-V wakeup, initializing it! \n");
        wakeup_gpio_init();
        init_ulp_program();
    }

    if (cause == ESP_SLEEP_WAKEUP_ULP)
    {
        printf("ULP-RISC-V woke up the main CPU! \n");
    }

    /* Go back to sleep, only the ULP Risc-V will run */
    printf("Entering in deep sleep\n\n");

    /* Small delay to ensure the messages are printed */
    delay_ms(2000);
    if (cause <= 12)
    {
        printf("%s\n", wakeup_str[cause]);
    }
    else
    {
        printf("Unknown wakeup cause %d\n", cause);
    }
    printf("wakeup cause %d\n", cause);
    vTaskDelay(100);

    /* RTC peripheral power domain needs to be kept on to detect
       the GPIO state change */
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);

    ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());
    esp_deep_sleep_start();
}

static void init_ulp_program(void)
{
    esp_err_t err = ulp_riscv_load_binary(ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start));
    ESP_ERROR_CHECK(err);

    /* Start the program */
    ulp_riscv_cfg_t cfg = {
        .wakeup_source = ULP_RISCV_WAKEUP_SOURCE_GPIO,
    };

    err = ulp_riscv_config_and_run(&cfg);
    ESP_ERROR_CHECK(err);
}
```

```
int main (void)
  {
     //  ulp_riscv_wakeup_main_processor();
      /* Wakeup interrupt is a level interrupt, wait 1 sec to
         allow user to release button to avoid waking up the ULP multiple times */
      ulp_riscv_delay_cycles(500 * ULP_RISCV_CYCLES_PER_MS);
      ulp_riscv_gpio_wakeup_clear();
      /* ulp_riscv_halt() is called automatically when main exits */
      return 0;
  }
```

```
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM11 instead...
--- idf_monitor on \\.\COM11 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c36
0x40378c36: esp_cpu_wait_for_intr at C:/ESP/esp-idf/components/esp_hw_support/cpu.c:121

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x11f0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa64
load:0x403cc700,len:0x2cc0
entry 0x403c9894
W (88) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
Not a ULP-RISC-V wakeup, initializing it!
Entering in deep sleep

ESP_SLEEP_WAKEUP_UNDEFINED, In case of deep sleep, reset was not caused by exit from deep sleep
wakeup cause 0
None
Waiting for the device to reconnect........
ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG, Wakeup caused by COCPU crash
wakeup cause 11
None
Waiting for the device to reconnect......
ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG, Wakeup caused by COCPU crash
wakeup cause 11
None
Waiting for the device to reconnect...
ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG, Wakeup caused by COCPU crash
wakeup cause 11
None
Waiting for the device to reconnect.......
ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG, Wakeup caused by COCPU crash
wakeup cause 11
None
Waiting for the device to reconnect............
ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG, Wakeup caused by COCPU crash
wakeup cause 11
None
```

shangke1988, 2024-01-14:
It work well, but I don't know why!

I edit my code:
`  
int main (void)
{
    // do the work of the code

    ulp_riscv_gpio_wakeup_clear();
    /* ulp_riscv_halt() is called automatically when main exits */
    return 0;
}
`
To
`
int main (void)
{
    // do the work of the code

     REG_SET_FIELD(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_2_CLK_DIS, 0x3F);

    uint32_t r1,r2;
    r1 = READ_PERI_REG(RTC_CNTL_ULP_CP_TIMER_REG);
    r2 = READ_PERI_REG(RTC_CNTL_COCPU_CTRL_REG);
   
    r1|=RTC_CNTL_ULP_CP_GPIO_WAKEUP_CLR;
    r2|=RTC_CNTL_COCPU_DONE | RTC_CNTL_COCPU_SHUT_RESET_EN;
   
    WRITE_PERI_REG(RTC_CNTL_ULP_CP_TIMER_REG, r1);
    WRITE_PERI_REG(RTC_CNTL_COCPU_CTRL_REG, r2);
 
    while(1);
    return 0;
}
`
The shorter time between `WRITE_PERI_REG(RTC_CNTL_ULP_CP_TIMER_REG, r1);` and `WRITE_PERI_REG(RTC_CNTL_COCPU_CTRL_REG, r2);`  is the better.

Thinks boarchuz in https://esp32.com/viewtopic.php?f=13&t=18802
This method base on "Remove REG_SET_FIELD(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_2_CLK_DIS, 0x3F) from ulp_riscv_shutdown" . I move it befor `ulp_riscv_gpio_wakeup_clear();`, than code work better, but occasionally error.

I have found
`
int main (void)
{
    // do the work of the code

    ulp_riscv_gpio_wakeup_clear();
    ulp_riscv_delay_cycles(100 * ULP_RISCV_CYCLES_PER_MS);
    ulp_riscv_halt();
    return 0;
}
`
has crashed everytime.
Maybe ULP will crash after `ulp_riscv_gpio_wakeup_clear` in a short time. The way work well is halt ULP before it crash.