espressif/esp-idf #10484: led_strip had i nice interface in the past, now it is gone.

franz-ms-muc, 2023-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/led_strip

as of 4.4:

https://github.com/espressif/esp-idf/tree/release/v4.4/examples/peripherals/rmt/led_strip
https://github.com/espressif/esp-idf/tree/release/v4.4/examples/common_components/led_strip

the Inteface was nice, the Lib was encapsulated. 

ESP_ERROR_CHECK(strip->set_pixel(strip, j, red, green, blue));
ESP_ERROR_CHECK(strip->refresh(strip, 100));
strip->clear(strip, 50);

as of 5.0

https://github.com/espressif/esp-idf/tree/release/v5.0/examples/common_components/ --> led_strip gone, 

https://github.com/espressif/esp-idf/tree/release/v5.0/examples/peripherals/rmt/led_strip
is much worse compared to the Situation before.

franz-ms-muc, 2023-01-04:
ahhhhhh

finding: 

[https://components.espressif.com/components/espressif/led_strip](https://components.espressif.com/components/espressif/led_strip)

https://github.com/espressif/idf-extra-components/tree/master/led_strip

suda-morris, 2023-01-04:
Hi @franz-ms-muc  The previous led_strip component is not gone, its interface is getting better actually. But it's not reside in the esp-idf repo anymore, but in the component registry.

The led_strip [example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/led_strip) in the esp-idf example folder now has a different focus, and it is mainly to show the detail usage of RMT APIs, so you may think it's not that simple.

closed 2023-01-04 by suda-morris