espressif/esp-idf #10485: ledc existance with deep sleep

mbastida123, 2023-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-devkit

### Power Supply used.

USB

### What is the expected behavior?

ledc library should work normally even after entering deep sleep.

### What is the actual behavior?

After a wake up from deep sleep the `CLK8M_CLK` cannot be calibrated,

### Steps to reproduce.

```
gpio_pad_select_gpio(CONFIG_BUZZER_GPIO_NUMBER);
gpio_set_direction((gpio_num_t)CONFIG_BUZZER_GPIO_NUMBER, GPIO_MODE_OUTPUT);
gpio_pulldown_dis((gpio_num_t)CONFIG_BUZZER_GPIO_NUMBER);
gpio_pullup_dis((gpio_num_t)CONFIG_BUZZER_GPIO_NUMBER);
gpio_set_level((gpio_num_t)CONFIG_BUZZER_GPIO_NUMBER, 0);

ledc_timer_config_t timer_conf;
timer_conf.speed_mode = SPEED_MDOE;
timer_conf.timer_num  = LEDC_TIMER_0;
timer_conf.clk_cfg=LEDC_USE_RTC8M_CLK;
timer_conf.duty_resolution=LEDC_TIMER_10_BIT;
timer_conf.freq_hz    = 5000;
ledc_timer_config(&timer_conf);

ledc_channel_config_t ledc_conf;
ledc_conf.gpio_num   = CONFIG_BUZZER_GPIO_NUMBER;
ledc_conf.speed_mode = SPEED_MDOE;
ledc_conf.channel    = LEDC_CHANNEL_0;
ledc_conf.intr_type  = LEDC_INTR_DISABLE;
ledc_conf.timer_sel  = LEDC_TIMER_0;
ledc_conf.duty       = 0x0; // 50%=0x3FFF, 100%=0x7FFF for 15 Bit
	                          // 50%=0x01FF, 100%=0x03FF for 10 Bit
ledc_channel_config(&ledc_conf);
```
The above code works (I use setter methods to change frequency and duty) if I don't enter in deep sleep. 

### Debug Logs.

```plain
When I enter in deep sleep and wakeup I get the following errors: 

E (693) ledc: Calibrate CLK8M_CLK failed
E (693) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=0
D (703) ledc: LEDC_PWM CHANNEL 0|GPIO 14|Duty 0000|Time 0
```

### More Information.

Looking more in depth it seems like the problem is here:

`ledc.c`
```
//We know that CLK8M is about 8M, but don't know the actual value. So we need to do a calibration.
static bool ledc_slow_clk_calibrate(void)
{
    if (periph_rtc_dig_clk8m_enable()) {
        s_ledc_slow_clk_8M = periph_rtc_dig_clk8m_get_freq();
        ESP_LOGD(LEDC_TAG, "Calibrate CLK8M_CLK : %d Hz", s_ledc_slow_clk_8M);
        return true;
    }
    ESP_LOGE(LEDC_TAG, "Calibrate CLK8M_CLK failed");
    return false;
}
```

Particularly in `periph_rtc_dig_clk8m_enable()`
_If this function is called a number of times, the `periph_rtc_dig_clk8m_disable` function needs to be called same times to disable._

So I have tried adding `periph_rtc_dig_clk8m_disable()` before deep sleep but the issue still persists.

songruo, 2023-01-04:
Hi @mbastida123,
The problem should have been fixed and backported to release/v4.4 (commit id: 883e54aa71abee7ef298c55858b2df2aeee01012). The fix is not included in tag v4.4.3, but we are going to release v4.4.4 very soon, and it will be there!

mbastida123, 2023-01-04:
Thanks @songruo for the info.

Do you have an ETA on when that will be?

closed 2023-01-04 by mbastida123

songruo, 2023-01-05:
@mbastida123 Should be available within the month~