espressif/esp-idf #10486: examples\network\simple_sniffer's apptrace can not collect data

WangShuoran, 2023-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-release

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I want to learn how to use `esp_apptrace_write()` and `esp_apptrace_read()` so I find all of the example with above API. I find examples\network\simple_sniffer config SNIFFER_PCAP_DESTINATION_JTAG is depend on APPTRACE_DEST_TRAX(in main/Kconfig.projbuild). But when I open Apptrce such as CONFIG_APPTRACE_DEST_JTAG, I can not select SNIFFER_PCAP_DESTINATION_JTAG. I find in esp_app_trace.h:
```c
typedef enum {
    ESP_APPTRACE_DEST_JTAG = 1,                         ///< JTAG destination
    ESP_APPTRACE_DEST_TRAX = ESP_APPTRACE_DEST_JTAG, 	///< xxx_TRAX name is obsolete, use more common xxx_JTAG
    ESP_APPTRACE_DEST_UART0,	                        ///< UART0 destination
    ESP_APPTRACE_DEST_MAX = ESP_APPTRACE_DEST_UART0,
    ESP_APPTRACE_DEST_NUM
} esp_apptrace_dest_t;
```
So does APPTRACE_DEST_TRAX == APPTRACE_DEST_JTAG?

I am delete Kconfig.projbuild `depends on APPTRACE_DEST_TRAX` and open CONFIG_APPTRACE_DEST_JTAG and SNIFFER_PCAP_DESTINATION_JTAG. I hope I can collect any data with esp_apptrace_write() in cmd_pcap.c.

### What is the actual behavior?

When I build and fuse elf to my develop board(ESP32C3), **and this develop board is running ok with example/system/app_trace_to_host**, so I think only check esp_apptrace_write is also OK, but it is not.

I fuse the elf too the board, and debug this(prepare to running `esp apptrace start xxx` in telnet), and I open telnet to running 
`esp apptrace start xxx(I try any status any datasize and any timeout)`, at same time I put debug window “running” button, but the data can not collect. I don't know why.

### Steps to reproduce.

1. build and fuse elf to my develop board
2. debug this
3. open telnet to running `esp apptrace start xxx`
4. at same time I put debug window “running” button

Alvin1Zhang, 2025-03-12:
Thanks for reporting.

erhankur, 2025-03-12:
@WangShuoran yes. APPTRACE_DEST_TRAX  is APPTRACE_DEST_JTAG 

Sorry for the late response. Let us know if you still need help.

closed 2025-03-12 by erhankur