espressif/esp-idf #10488: Need to change UART_RX_SIZE of esp32 in idf.

beginner140BB, 2023-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

 please help me change the value of UART_RX_SIZE in the ESP32 project.

 I have looked at the ESP32 technical reference manual (page 364) and the forum post at https://www.esp32.com/viewtopic.php?f=2&t=14428, but I have not been able to find any inbuilt functions related to this. 
Is there another way to modify the UART_RX_SIZE value?

From the forum based on this link
 https://github.com/espressif/esp-idf/blob/c65b67115a52af67fc682425304a7f04cbcdb9d5/components/soc/soc/esp32/include/soc/uart_struct.h#L315-L316
 
 They have said try modifying the value of the mentioned variable.
 
 
 I was wondering if anyone would be willing to share a code sample or some guidance on how you implemented this feature.

AxelLin, 2023-01-04:
https://github.com/espressif/esp-idf/issues/9682

beginner140BB, 2023-01-06:
@AxelLin 

is there any way to solve 

UART_FIFO_OVF  (UART FIFO overflow event) from uart event.

what can we do to solve this.?

beginner140BB, 2023-01-06:
@AxelLin 
did  you know what is actually causing 
 UART_FIFO_OVF,          /*!< UART FIFO overflow event*/

is there any way to solve  this:

```
xSemaphoreTake( xMutex, portMAX_DELAY );// Reading serial commands from palmtec
uart0_rx(Serial_data1);
xSemaphoreGive( xMutex )
```

- where uart reading is happening

- the function uart0_rx is given below

```
int uart0_rx(char *data)
{
	int length = 0;
	ESP_ERROR_CHECK(uart_get_buffered_data_len(UART_NUM_0,(size_t*)&length));
	length=uart_read_bytes(UART_NUM_0,data,1024, 20 / portTICK_RATE_MS);
	uart_flush_input(UART_NUM_0);
	return length;
}
```

```
void uart_init(int BaudRate)
{
    const uart_config_t uart_config = {
    		.baud_rate = BaudRate,
    		.data_bits = UART_DATA_8_BITS,
    		.parity = UART_PARITY_DISABLE,
    		.stop_bits = UART_STOP_BITS_1,
    		.flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
			.source_clk = UART_SCLK_APB,
    };
    uart_driver_install(UART_NUM_0, 2048, 1024, 20, &uart0_queue, 0);
    uart_param_config(UART_NUM_0, &uart_config);
    uart_set_pin(UART_NUM_0, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
}
```

- where uart init takes place

do you have any suggestion as i am getting fifo overflow sometimes only? 

**the data is actually ~250 bytes with baudrate of 115200**

closed 2023-04-25 by beginner140BB