espressif/esp-idf #10490: WiFi crashed when a long-period soft timer task is running in a sleep mode

dizcza, 2023-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release 5.0 branch

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO T8

### Power Supply used.

USB

### What is the expected behavior?

`GPIO ON` and `GPIO OFF` is logged periodically.

### What is the actual behavior?

WiFi crashes.

With WiFi initialization commented out, the behavior is expected (no crashes).

### Steps to reproduce.

Just add a software periodic timer to wifi station example. Here is the full code:

```c
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_timer.h"
#include "driver/gpio.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via project configuration menu

[... 192 lines cut ...]

0x40148aa1: esp_log_writev at /home/dizcza/tools/esp-idf/components/log/log.c:200

0x4008e04d: esp_log_write at /home/dizcza/tools/esp-idf/components/log/log.c:210

0x400e940e: wifi_log at ??:?

0x4010304a: pm_set_next_tbtt at ??:?

0x40092d75: pm_tbtt_process at ??:?

0x4014738a: pm_connectionless_wake_window_timeout_process at ??:?

0x401473f5: pp_timer_do_process at ??:?

0x40090f12: ppTask at ??:?

0x4008bc5d: vPortTaskWrapper at /home/dizcza/tools/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

```
```

### More Information.

Reproduced only with large sleep delays in the timer callback.

dizcza, 2023-01-04:
By sleep mode I mean the IDLE mode, not a deep sleep.

dizcza, 2023-01-04:
And I know that the timer is restarted as soon as it finishes. I've shown this code in order to point out the bug only.

igrr, 2023-01-04:
Note that esp_timer is a shared resource, so delaying in your callback for a long time will prevent the callbacks from all other parts of the program from being executed.

In this case the crash is probably a real bug which occurs when a particular error log is printed by the Wi-Fi library; however under normal conditions, when esp_timer callbacks are not artificially delayed, this error log is never printed.

We can fix the particular crash which occurs due to incorrect usage of ESP_LOG, but note that this will not make the program work "correctly": the Wi-Fi stack and probably other parts of ESP-IDF won't be able to work if their esp_timer callbacks are delayed for so long. Please try to refactor your code so that esp_timer callbacks do just the bare minimum of work necessary, and delegate the rest to a lower priority application task.

dizcza, 2023-01-04:
Yes, I've already refactored the code to alternate between two successive timers - on and off - started one after another. I thought the issue is something you wanted to have a look at.

I doubt the users are aware of such behavior. If this is expected and won't fix, at least add your note to the esp_timer API description.

igrr, 2023-01-05:
There is a sentence about this in the esp-timer API reference:

> Timer callbacks are dispatched from a high-priority esp_timer task. Because all the callbacks are dispatched from the same task, it is recommended to only do the minimal possible amount of work from the callback itself, posting an event to a lower priority task using a queue instead.

If you have a suggestion how to rephrase this, i can update the docs.

dizcza, 2023-01-05:
Hmm, looks good.

Feel free to close the issue then.

closed 2023-01-05 by igrr

emaayan, 2023-08-04:
hi.. i have the exact same crash, but i'm not using any articial delays. 
in fact i'm using callback for wi-fi promiscious mode, on each callback, the data is written to socket opened from a tcp client from another machine. 
one high load when there's a lot of traffic, i see activity is frozen on the send portion  and than a crash with this stack trace occurs.