espressif/esp-idf #10491: How to proper implement ESP32 Nimble Bound with security PIN before connection

urbito, 2023-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Environment**

Development Kit: none
Module or chip used: ESP32-WROVER-E
IDF version: v4.4
Build System: ESPIDF
Compiler version: 
Operating System: windows10
Using an IDE?: VSCode - ESPIDF
Power Supply: external 5V

**Problem Description**

I am building a custom app based on bleprph and spp_server.

The goal is that this app do: 

-Write Characteristic : OK
-Read Characteristic: OK
-PIN request before connection/bund

Right now this last point is NOT OK, i am using android 12, with an APP developed in AppInventor to test.

If i try to bound the phone to the esp32 without the app (like paring BT speaker...) it does request for password, if i put a wrong or short password it does provide ( i have added the BLE_GAP_EVENT_ENC_CHANGE) me an error with a number like 1028 or 1228 in the consol (PC), and also another one if i do "cancel" the pair. If i provide the correct password, the code is 0.

So without the APP seems like a normal behaivour.

When i use the APP, when i attempt a conection, if i provide a wrong or short password, it does "connect" for a little time, lets say 3 seconds, and then the BLE_GAP_EVENT_ENC_CHANGE provide the code error, same if i put the correct password. But if i do not provide a password but press the CANCEL button, the app still connect to the esp32 (because it does even connect before i do provide a wrong/short password). And the console does not provide any error/message about the BLE_GAP_EVENT_ENC_CHANGE. I can even start sending some "text" from the app to the esp32 and they do arrive, then if i stop sending them, the esp32 breaks/disconnect, like, he does realise there was no security PIN provided.

Expected Behavior

No connection if no pin if provided.

Actual Behavior

Connection between app and esp32 if no security pin is provided for a indeterminated period of time.

Steps to reproduce

I do use the following parts:

[quote]

case BLE_GAP_EVENT_CONNECT:
	/* A new connection was established or a connection attempt failed. */
        
        MODLOG_DFLT(INFO, "connection %s; status=%d ",
                    event->connect.status == 0 ? "established" : "failed",
                    event->connect.status);
        if (event->connect.status == 0) {
            rc = ble_gap_conn_find(event->connect.conn_handle, &desc);
            assert(rc == 0);
            ble_spp_server_print_conn_desc(&desc);              // <-- INFORMACION POR CONSOLA CUANDO SE CONECTA 
	    is_connect=true;
	    connection_handle = event->connect.conn_handle;
        }
        MODLOG_DFLT(INFO, "\n");
        if (event->connect.status != 0) {
            /* Connection failed; resume advertising. */
            ble_spp_server_advertise();
        }

        rc = ble_gap_security_initiate(event->connect.conn_handle);

        MODLOG_DFLT(INFO, "gap secury = %d", rc);
        

        return 0;

case BLE_GAP_EVENT_PASSKEY_ACTION:
        ESP_LOGI(tag, "PASSKEY_ACTION_EVENT started \n");
        struct ble_sm_io pkey = {0};
        int key = 0;

        if (event->passkey.params.action == BLE_SM_IOACT_DISP) {
            pkey.action = event->passkey.params.action;
            pkey.passkey = 123456; // This is the passkey to be entered on peer
            ESP_LOGI(tag, "Enter passkey %d on the peer side", pkey.passkey);
            rc = ble_sm_inject_io(event->passkey.conn_handle, &pkey);
            ESP_LOGI(tag, "ble_sm_inject_io result: %d\n", rc);
        } else if (event->passkey.params.action == BLE_SM_IOACT_NUMCMP) {
            ESP_LOGI(tag, "Passkey on device's display: %d", event->passkey.params.numcmp);
            ESP_LOGI(tag, "Accept or reject the passkey through console in this format -> key Y or key N");
            pkey.action = event->passkey.params.action;
            if (scli_receive_key(&key)) {
                pkey.numcmp_accept = key;
            } else {
                pkey.numcmp_accept = 0;
                ESP_LOGE(tag, "Timeout! Rejecting the key");
            }
            rc = ble_sm_inject_io(event->passkey.conn_handle, &pkey);
            ESP_LOGI(tag, "ble_sm_inject_io result: %d\n", rc);
        } else if (event->passkey.params.action == BLE_SM_IOACT_OOB) {
            static uint8_t tem_oob[16] = {0};
            pkey.action = event->passkey.params.action;
            for (int i = 0; i < 16; i++) {
                pkey.oob[i] = tem_oob[i];
            }
            rc = ble_sm_inject_io(event->passkey.conn_handle, &pkey);
            ESP_LOGI(tag, "ble_sm_inject_io result: %d\n", rc);
        } else if (event->passkey.params.action == BLE_SM_IOACT_INPUT) {
            ESP_LOGI(tag, "Enter the passkey through console in this format-> key 123456");
            pkey.action = event->passkey.params.action;
            if (scli_receive_key(&key)) {
                pkey.passkey = key;
            } else {
                pkey.passkey = 0;
                ESP_LOGE(tag, "Timeout! Passing 0 as the key");
            }
            rc = ble_sm_inject_io(event->passkey.conn_handle, &pkey);
            ESP_LOGI(tag, "ble_sm_inject_io result: %d\n", rc);
        }else{
            int ret = ble_gap_terminate(event->connect.conn_handle, BLE_ERR_CONN_LIMIT);
            if (ret != 0) 
            { 
                ESP_LOGE(tag, "Error 0x%X disconnecting", ret);
            }
        }
        return 0;

case BLE_GAP_EVENT_ENC_CHANGE:
        /* Encryption has been enabled or disabled for this connection. */
        MODLOG_DFLT(INFO, "encryption change event; status=%d ",
                    event->enc_change.status);
        rc = ble_gap_conn_find(event->enc_change.conn_handle, &desc);
        assert(rc == 0);
        ble_spp_server_print_conn_desc(&desc);
        MODLOG_DFLT(INFO, "\n");
        return 0;

[/quote]

And in the configuration i do use:

[quote]
 nimble_port_init();                                                             // -> 3

    /* Initialize the NimBLE host configuration. */
    ble_hs_cfg.reset_cb = ble_spp_server_on_reset;
    ble_hs_cfg.sync_cb = ble_spp_server_on_sync;
    ble_hs_cfg.gatts_register_cb = gatt_svr_register_cb;
    ble_hs_cfg.store_status_cb = ble_store_util_status_rr;
    ble_hs_cfg.sm_io_cap =  BLE_SM_IO_CAP_DISP_ONLY  ;

    ble_hs_cfg.sm_bonding = 1;
    ble_hs_cfg.sm_mitm = 1;
    ble_hs_cfg.sm_sc = 0;
    ble_hs_cfg.sm_our_key_dist = 1;
    ble_hs_cfg.sm_their_key_dist = 1;

[/quote]

RoshanESP, 2023-01-09:
Hi @urbito .

 For your usecase , you basically need passkey action, and hence you are updating the IO capability. At my end, i tried the bleprph example of IDF and modified it to use BLE_SM_IO_CAP_DISP_ONLY  ( instead of the default BLE_SM_IO_CAP_NO_IO ) . I used **nRF Connect** app with esp32 as bleprph. My observations are :

1. When I do everything right, device connects successfully, enc_change event gives status code = 0
2. On Mobile when I cancel the popup of pairing, I get enc_change event with status code 1289 on esp32. and also, disconnect event with status code 531. 
3. On mobile when I click ok on pair, but click cancel on popup for entering passkey, enc_change event with status 13 and disconnect event with status 531 on esp32. 
4. When I put wrong passkey, enc_change with status 1028, and disconnect event with status code 531. 
 

So overall, functionality wise it is working ok. So for your case I have few queries: 

 1. Please share a complete debug enabled log .
 2. "if i provide a wrong or short password, it does "connect" for a little time"  . How do you confirm that it is connected ?. Is any event connect event posted ? 
3. "But, if I do not provide a password but press the CANCEL button, the app still connect to the esp32"  . What does your app do when pressed cancel ? Do you have logs to show what is sent to stack ? How do you confirm that it is connected ? 
4. "can even start sending some "text" from the app to the esp32 and they do arrive, then if i stop sending them, the esp32 breaks/disconnect, like, he does realise there was no security PIN provided."  ... Am sorry, but i did not understand what you meant here. May be if you can provide some log, then i can understand more
5. I assume the app you are developing is on mobile phone side . If yes, then is it possible for you to use nRF Connect app on mobile and confirm the functionality? I suspect may be some functionality is incorrect in the app. This test will confirm my suspicion.

rahult-github, 2023-04-25:
Hi @urbito  , any updates ?

rahult-github, 2023-06-28:
Closing issue. Please feel free to reopen, in case of futher updates.

closed 2023-06-28 by rahult-github