espressif/esp-idf #10492: nvs_flash_init crashes when disturbed by ISR

dizcza, 2023-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release 5.0 branch

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO T8

### Power Supply used.

USB

### What is the expected behavior?

The app runs without crashes.

### What is the actual behavior?

NVS init crashed at one of the interrupts.

What really puzzles me is that everything works if I substitute

```diff
-    if (rtc_dev.initialized) {
+    if (1) {
         attach_rtc_sqw_handler(pin_gpio_in, read_sensor_task_handle);
     }  else {
         int divider = 5;
```

or remove the else loop. This is the first time I'm experiencing the subtlety of conditional branching.

Also, of course, the code works fine if I remove `xTaskNotifyFromISR(...)` in the interrupt callback.

### Steps to reproduce.

```c
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_log.h"
#include "nvs_flash.h"
[... 205 lines cut ...]
0x400d829d: ensure_partitions_loaded at /home/dizcza/tools/esp-idf/components/esp_partition/partition.c:283
 (inlined by) esp_partition_find at /home/dizcza/tools/esp-idf/components/esp_partition/partition.c:265

0x400d82ff: esp_partition_find_first at /home/dizcza/tools/esp-idf/components/esp_partition/partition.c:321

0x400d7fe5: nvs::partition_lookup::lookup_nvs_partition(char const*, nvs::NVSPartition**) at /home/dizcza/tools/esp-idf/components/nvs_flash/src/nvs_partition_lookup.cpp:14

0x400d708d: nvs::NVSPartitionManager::init_partition(char const*) at /home/dizcza/tools/esp-idf/components/nvs_flash/src/nvs_partition_manager.cpp:46 (discriminator 2)

0x400d6a58: nvs_flash_init_partition at /home/dizcza/tools/esp-idf/components/nvs_flash/src/nvs_api.cpp:128
 (inlined by) nvs_flash_init_partition at /home/dizcza/tools/esp-idf/components/nvs_flash/src/nvs_api.cpp:120

0x400d6a6e: nvs_flash_init at /home/dizcza/tools/esp-idf/components/nvs_flash/src/nvs_api.cpp:164

0x400d423c: nvs_flash_init_check at /home/dizcza/Projects/Embedded/eclipse-workspace/nvs_isr_bug/main/nvs_isr_bug.c:108 (discriminator 1)
 (inlined by) app_main at /home/dizcza/Projects/Embedded/eclipse-workspace/nvs_isr_bug/main/nvs_isr_bug.c:175 (discriminator 1)

0x400eb2be: main_task at /home/dizcza/tools/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

```
```

### More Information.

Today I've done bug hunting in the ESP-IDF framework but this one is exceptional. I was going crazy tracking down where the issue occurs. Who would have thought that a simple if-branching could cause such trouble!

suda-morris, 2023-01-05:
> 0x400d3fd8: sdptimer_callback at /home/dizcza/Projects/Embedded/eclipse-workspace/nvs_isr_bug/main/nvs_isr_bug.c:73

The function `sdptimer_callback` is not placed in the IRAM, but in the flash. When the flash cache is disabled, CPU can't get the text of that function, so crashed.

You may want to add `IRAM_ATTR` for that function. Or remove the `ESP_INTR_FLAG_IRAM` flag when you install the GPIO interrupt. Or enable the `CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y` and `CONFIG_SPIRAM_RODATA=y` if your ESP32S3 has PSRAM.

dizcza, 2023-01-05:
Dear me, you're right! I thought it's inlined so I don't need to add the IRAM attribute. With your suggestion, the problem is gone.

Thanks.

dizcza, 2023-01-05:
But still... why does it work in the first place if I change

```diff
-    if (rtc_dev.initialized) {
+    if (1) {
         attach_rtc_sqw_handler(pin_gpio_in, read_sensor_task_handle);
     }  else {
         int divider = 5;
```

and place no IRAM attr to the timer callback?