espressif/esp-idf #10493: SD card fseek is much slower on esp-idf 5.0 comparing to 4.4

tkuc, 2023-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello.
I got strange problem after updating esp-idf from 4.4 to 5.0 (release branch)
SD card reading time increased from 124ms to 1153ms. There is no difference in code or configuration, also content of SD card was not changed.

**Main problem**: Execution time of fseek raised from 2ms to 81ms, this is clearly visible on difference between timestamp t1-t

Important part of code
`        t = esp_timer_get_time();
        lastline = currentLine;
        fseek(f,ef->lineLength*(currentLine-1), SEEK_SET); 

        t1 = esp_timer_get_time();
        fgets(line, sizeof(line), f);
        t2 = esp_timer_get_time();

        valueInLine = strtoll(line,&pEnd,10);
        ESP_LOGI(TAG, "Line :%u val:%lld", currentLine, valueInLine);
        t3 = esp_timer_get_time();

        ESP_LOGI(TAG, "t-t1:%llu,  t1-t2:%llu, t2-t3:%llu,",(t1-t)/1000, (t2-t1)/1000, (t3-t2)/1000 );`
[image]

I was trying 
- change task priority
- running on different core
- change data size 
- changing max_transfer_sz
- change to fast seek CLMT buffer and enable fast seek algorithm

i'm using fseek to quiclkly jump in big file between lines. reading one by one is to slow.

Thanks in advance for any ideas.

adokitkat, 2023-02-06:
On the latest `release/v5.0` branch I am getting with this benchmark code

```c
int32_t r, currentLine;
int64_t t, t1;
currentLine = 1;
        
ESP_LOGI(TAG, "Opening benchmark.txt");
f = fopen("/sdcard/benchmark.txt", "rb");
        
ESP_LOGI(TAG, "Benchmark:");
for (int i = 0; i < 10; ++i) {
    t = esp_timer_get_time();
    fseek(f, 7*(currentLine-1), SEEK_SET);
    t1 = esp_timer_get_time();

    ESP_LOGI(TAG, "t-t1:%llu, line: %ld", (t1-t)/1000, currentLine);
    r = esp_random() % 10000;
    currentLine = r;
}
fclose(f);
```

these results

```
I (415) example: Benchmark task
I (425) example: Opening benchmark.txt
I (425) example: Benchmark:
I (425) example: t-t1:0, line: 1
I (435) example: t-t1:1, line: 4227
I (435) example: t-t1:0, line: 1442
I (445) example: t-t1:1, line: 5735
I (445) example: t-t1:1, line: 3911
I (455) example: t-t1:0, line: 1261
I (455) example: t-t1:1, line: 9194
I (455) example: t-t1:1, line: 4127
I (465) example: t-t1:1, line: 8109
I (465) example: t-t1:0, line: 436
```

I am using SDMMC and FATFS. My benchmark file has 10000 lines, each line is long 6 characters + 1 newline = 7. Lines are randomly selected each time. Time between `t` and `t1` varies between 0 and 1 ms as you can see.

tkuc, 2023-02-06:
Hi @adokitkat 
Thank you for your effort.
I'm using SDSPI with SDMMC_FREQ_DEFAULT speed. Filesystem is FAT32. In attachment i've pasted sdkconfig from my project.

Just now I redo my test and remove all other functions/task (in the same project). Also i took your example code - whole thing is pasted below.
Result are the same and there is no difference if we got 6 or 13 digits.

Sd: FAT32 Block 2048 Toshiba Class-4 8GB (i've tested also on different ones 32GB Class-V10 and result are the same)

Result from ESP-IDF 5.0 ( #7f5ecbe533b2666df6b10658a023b5f637320696)
`I (19381) test: Benchmark6:
I (19421) test: t-t1:40, line: 8487
I (19501) test: t-t1:81, line: 151
I (19581) test: t-t1:81, line: 9187
I (19671) test: t-t1:81, line: 9913
I (19751) test: t-t1:81, line: 4179
I (19831) test: t-t1:81, line: 6995
I (19911) test: t-t1:81, line: 1750
I (19991) test: t-t1:81, line: 1750
I (20071) test: t-t1:81, line: 4913
I (20161) test: t-t1:81, line: 5562
I (20401) test: Benchmark13:
I (20441) test: t-t1:40, line: 1134
I (20531) test: t-t1:81, line: 2273
I (20611) test: t-t1:81, line: 3547
I (20691) test: t-t1:81, line: 8521
I (20771) test: t-t1:81, line: 4830
I (20851) test: t-t1:81, line: 9140
I (20931) test: t-t1:81, line: 5557
I (21021) test: t-t1:81, line: 7678
I (21101) test: t-t1:81, line: 4726
I (21181) test: t-t1:81, line: 8503`

ESP-IDF 4.4.0 ( #8153bfe4125e6a608abccf1561fd10285016c90a)
`I (20892) test: Benchmark6:
I (20892) test: t-t1:1, line: 7117
I (20892) test: t-t1:1, line: 261
I (20892) test: t-t1:0, line: 2209
I (20902) test: t-t1:1, line: 1196
I (20902) test: t-t1:1, line: 607
I (20902) test: t-t1:0, line: 1025
I (20912) test: t-t1:1, line: 8846
I (20912) test: t-t1:1, line: 9554
I (20922) test: t-t1:1, line: 4656
I (20922) test: t-t1:1, line: 5338
I (20932) test: Benchmark13:
I (20932) test: t-t1:1, line: 2403
I (20932) test: t-t1:1, line: 5953
I (20942) test: t-t1:1, line: 6338
I (20942) test: t-t1:1, line: 5911
I (20942) test: t-t1:1, line: 715
I (20952) test: t-t1:1, line: 871
I (20952) test: t-t1:1, line: 8009
I (20962) test: t-t1:1, line: 860
I (20962) test: t-t1:0, line: 4466
I (20962) test: t-t1:1, line: 9635
I (21972) test: Benchmark:`

Below you can check whole code
`#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "misc.h"
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_vfs_fat.h"
#include "sdmmc_cmd.h"
#include "esp_timer.h"
#include "bootloader_random.h"

static const char *TAG = "test";
#define MOUNT_POINT "/sdcard"
#define PIN_NUM_MISO 2
#define PIN_NUM_MOSI 15
#define PIN_NUM_CLK  14
#define PIN_NUM_CS   4
#define SPI_DMA_CHAN    1

void fileScan_Task(void* arg);

void app_main(void)
{
    esp_err_t ret;
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = false,
        .max_files = 5,
        .allocation_unit_size = 16 * 1024
    };
    sdmmc_card_t *card;
    const char mount_point[] = MOUNT_POINT;
    ESP_LOGI(TAG, "Initializing SD card");
    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    spi_bus_config_t bus_cfg = {
        .mosi_io_num = PIN_NUM_MOSI,
        .miso_io_num = PIN_NUM_MISO,
        .sclk_io_num = PIN_NUM_CLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 4000,
    };
    host.max_freq_khz = SDMMC_FREQ_DEFAULT; //20000
    ret = spi_bus_initialize(host.slot, &bus_cfg, SPI_DMA_CHAN);
    sdspi_device_config_t slot_config = SDSPI_DEVICE_CONFIG_DEFAULT();
    slot_config.gpio_cs = PIN_NUM_CS;
    slot_config.host_id = host.slot;

    ESP_LOGI(TAG, "Mounting filesystem");
    ret = esp_vfs_fat_sdspi_mount(mount_point, &host, &slot_config, &mount_config, &card);
    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);

    bootloader_random_enable();
    xTaskCreate(fileScan_Task, "fileScan", 10000, NULL, 10, NULL);

    while (1) {
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10669461/sdkconfig.txt)

        vTaskDelay(100);    //feed WDG
    }
}

void fileScan_Task(void* arg)
{
    int32_t r, currentLine=1;
    int64_t t, t1;
    FILE *f;

    ESP_LOGI(TAG, "Opening benchmark.txt");
    while(1)
    {

        f = fopen("/sdcard/6digit.txt", "rb");
                
        ESP_LOGI(TAG, "Benchmark6:");
        for (int i = 0; i < 10; ++i) {
            t = esp_timer_get_time();
            fseek(f, 7*(currentLine-1), SEEK_SET);
            t1 = esp_timer_get_time();

            ESP_LOGI(TAG, "t-t1:%llu, line: %ld", (t1-t)/1000, currentLine);
            r = (rand()%UINT32_MAX) % 10000;
            currentLine = r;
        }
        fclose(f);

        f = fopen("/sdcard/13digit.txt", "rb");
        ESP_LOGI(TAG, "Benchmark13:");
        for (int i = 0; i < 10; ++i) {
            t = esp_timer_get_time();
            fseek(f, 14*(currentLine-1), SEEK_SET);
            t1 = esp_timer_get_time();

            ESP_LOGI(TAG, "t-t1:%llu, line: %ld", (t1-t)/1000, currentLine);
            r = (rand()%UINT32_MAX) % 10000;
            currentLine = r;
        }
        fclose(f);
        vTaskDelay(1000/portTICK_PERIOD_MS);
    }
}

`

adokitkat, 2023-02-07:
This is strange. I cannot reproduce the issue. I am using the exact ESP-IDF version v5.0 ( #7f5ecbe533b2666df6b10658a023b5f637320696), your example and still getting those good numbers.

I attached my moddified example project to this message. Before flashing it you have to do `idf.py erase-flash` and physically remove and reinsert the SD card. Please let me know what results do you get.
[example.zip](https://github.com/espressif/esp-idf/files/10671155/example.zip)

Only things I changed in `idf.py menuconfig` is not using custom partition table and setting 4MB flash instead of 8. I am running the example on ESP32-WROVER KIT. This is the output:

```
entry 0x4008064c
I (27) boot: ESP-IDF v5.0 2nd stage bootloader
I (27) boot: compile time 03:05:31
I (27) boot: chip revision: v1.0
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) boot_comm: chip revision: 1, min. application chip revision: 0
I (100) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0ca04h ( 51716) map
I (127) esp_image: segment 1: paddr=0001ca2c vaddr=3ffb0000 size=02800h ( 10240) load
I (132) esp_image: segment 2: paddr=0001f234 vaddr=40080000 size=00de4h (  3556) load
I (135) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=25424h (152612) map
I (197) esp_image: segment 4: paddr=0004544c vaddr=40080de4 size=0c8b8h ( 51384) load
I (218) esp_image: segment 5: paddr=00051d0c vaddr=50000000 size=00010h (    16) load
I (225) boot: Loaded app from partition at offset 0x10000
I (225) boot: Disabling RNG early entropy source...
I (238) cpu_start: Pro cpu up.
I (238) cpu_start: Starting app cpu, entry point is 0x4008119c
0x4008119c: call_start_cpu1 at /home/ado/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (253) cpu_start: Pro cpu start user code
I (253) cpu_start: cpu freq: 240000000 Hz
I (253) cpu_start: Application information:
I (257) cpu_start: Project name:     sd_card_user
I (263) cpu_start: App version:      1
I (267) cpu_start: Compile time:     Feb  7 2023 03:05:25
I (273) cpu_start: ELF file SHA256:  f287ae473e071c72...
I (279) cpu_start: ESP-IDF:          v5.0
I (284) heap_init: Initializing. RAM available for dynamic allocation:
I (291) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (297) heap_init: At 3FFB3188 len 0002CE78 (179 KiB): DRAM
I (304) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (310) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (316) heap_init: At 4008D69C len 00012964 (74 KiB): IRAM
I (324) spi_flash: detected chip: gd
I (327) spi_flash: flash io: dio
I (332) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (342) test: Initializing SD card
I (342) test: Mounting filesystem
I (352) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (362) sdspi_transaction: cmd=52, R1 response: command not supported
I (412) sdspi_transaction: cmd=5, R1 response: command not supported
Name: APPSD
Type: SDHC/SDXC
Speed: 20 MHz
Size: 3840MB
CSD: ver=2, sector_size=512, capacity=7864320 read_bl_len=9
SSR: bus_width=1
I (442) test: Writing to bench.txt, wait please
I (742) test: Running benchmark
I (752) test: Benchmark on bench.txt:
I (752) test: t-t1:0, line: 1, first 3 chars: 123
I (752) test: t-t1:0, line: 9002, first 3 chars: ABC
I (752) test: t-t1:1, line: 5968, first 3 chars: ABC
I (762) test: t-t1:1, line: 6278, first 3 chars: ABC
I (762) test: t-t1:1, line: 7500, first 3 chars: ABC
I (772) test: t-t1:1, line: 3965, first 3 chars: ABC
I (782) test: t-t1:1, line: 2499, first 3 chars: ABC
I (782) test: t-t1:1, line: 8788, first 3 chars: ABC
I (792) test: t-t1:1, line: 8994, first 3 chars: ABC
I (792) test: t-t1:1, line: 6033, first 3 chars: ABC
I (1802) test: Benchmark on bench.txt:
I (1802) test: t-t1:0, line: 2341, first 3 chars: ABC
I (1802) test: t-t1:1, line: 2659, first 3 chars: ABC
I (1802) test: t-t1:1, line: 5834, first 3 chars: ABC
I (1812) test: t-t1:1, line: 7472, first 3 chars: ABC
I (1812) test: t-t1:1, line: 4072, first 3 chars: ABC
I (1822) test: t-t1:1, line: 9547, first 3 chars: ABC
I (1832) test: t-t1:1, line: 1439, first 3 chars: ABC
I (1832) test: t-t1:1, line: 9507, first 3 chars: ABC
I (1842) test: t-t1:1, line: 3346, first 3 chars: ABC
I (1842) test: t-t1:1, line: 8062, first 3 chars: ABC
I (2852) test: Benchmark on bench.txt:
I (2852) test: t-t1:0, line: 2023, first 3 chars: ABC
I (2852) test: t-t1:0, line: 72, first 3 chars: ABC
I (2852) test: t-t1:1, line: 5635, first 3 chars: ABC
I (2862) test: t-t1:1, line: 5377, first 3 chars: ABC
I (2862) test: t-t1:1, line: 2727, first 3 chars: ABC
I (2872) test: t-t1:1, line: 1532, first 3 chars: ABC
I (2882) test: t-t1:1, line: 5191, first 3 chars: ABC
I (2882) test: t-t1:1, line: 953, first 3 chars: ABC
I (2892) test: t-t1:1, line: 4380, first 3 chars: ABC
I (2892) test: t-t1:1, line: 8886, first 3 chars: ABC
...
```

I made the first line 123456 instead of ABCDEF to be distinguishable easily. I print the first 3 characters from current line so I know it jumps around correctly to the start of a line.

My microSD card info according to `sdmmc_card_print_info`:
```
Name: APPSD
Type: SDHC/SDXC
Speed: 20 MHz
Size: 3840MB
CSD: ver=2, sector_size=512, capacity=7864320 read_bl_len=9
SSR: bus_width=1
```

tkuc, 2023-02-07:
@adokitkat 
I took your example and run without any changes, also without running espconfig or erase. 
Every example i'm running in vscode because i got some environmental problems when i'm trying to run from cmd.
`I (43065) test: Benchmark on bench.txt:
I (43065) test: t-t1:1, line: 9192, first 3 chars: ABC
I (43065) test: t-t1:2, line: 7952, first 3 chars: ABC
I (43075) test: t-t1:2, line: 7880, first 3 chars: ABC
I (43075) test: t-t1:2, line: 7932, first 3 chars: ABC
I (43085) test: t-t1:2, line: 6702, first 3 chars: ABC
I (43085) test: t-t1:2, line: 7676, first 3 chars: ABC
I (43095) test: t-t1:2, line: 4724, first 3 chars: ABC
I (43105) test: t-t1:2, line: 143, first 3 chars: ABC
I (43105) test: t-t1:2, line: 5108, first 3 chars: ABC
I (43115) test: t-t1:2, line: 9265, first 3 chars: ABC`

i'm trying to figure out which option is causing this problem, because on clean example from you i got correct times, so it seems to be related to project config

tkuc, 2023-02-07:
@adokitkat 
I was able to find out what is causing issue to be visible - have no idea why this is happening, but still it is strange.
In your example you are using CS as GPIO13. i'm using GPIO4 and this is what was in my example.
When i changed GPIO4 to GPIO13 then sudsenly SEEK time fall to 1ms (hardware wiring remain the same for whole test - still connected to CS-GPIO4).

I checked with osciloscope and logic analyzer.
When i switched config to GPIO13 then on SDcard CS pin i got constant 0.83V and on GPIO13 there is correct CS signal.
When i use GPIO4 as CS then SEEK time is going up to ~80ms. On SD CS i can see transitions between 0 and 3.3V and it seems to be quite correct CS signal.
Question is why this is working if i'm using incorrect CS? I'm guessing because i got 0.83V which is considered as active CS

Here is my board schematic.
[image]

I've recorded saleale log (need to slowdown SCK to 4MHz, seek time increased to 10-20ms in "good case" and ~100-120ms in "bad" case)
Here is graphic comparation with simmilar timebase
[image]
And here are raw measurments in Logic 2 from saleale (you need to remove .txt extension because github dosn't allow this .sal)
[CS_IO13.sal.txt](https://github.com/espressif/esp-idf/files/10672857/CS_IO13.sal.txt)
[CS_IO4.sal.txt](https://github.com/espressif/esp-idf/files/10672860/CS_IO4.sal.txt)

adokitkat, 2023-02-07:
I would guess it still works because CS is not really needed when there is only one SPI device connected. 

However I am not quite sure why the slowdown on correct CS connection is happening, this will need a further examination.

adokitkat, 2023-02-07:
If you're using this SPI peripheral only with one device (sd slot + card), then you can try to set CS in `slot_config.gpio_cs` to `GPIO_NUM_NC` or `-1`.

tkuc, 2023-02-07:
As workaround this solution will be fine. I'm not planning to connect any other SPI device. But still this is not clear to me, why CS is affecting reading. I'm guessing, that CS timing someimes is not correct, and it dissable SD card during read/write operation. I don't have capabilities to check this deeper. I think that this was introduced in esp-idf 5.0, because on 4.4.0 it is fine.
If you think that further investigation is not needed we can close this topic. 
Thank you for your support @adokitkat!.

adokitkat, 2023-02-07:
No problem, thank you for express answers :) 

I still cannot reproduce the issue nor find an exact part of the code which caused the regression, if you want to maybe you could please share the output of both the good and bad case with log verbosity set to debug (`Component config -> Log output -> Default log verbosity: Debug`) and task watchdog timer disabled (`Component config -> ESP System Settings -> Enable Task Watchdog Timer: off`)? Thank you.

tkuc, 2023-02-07:
I've made tests
Here is comparation for GPIO4 and GPIO13 on IDF4.4
[image]
[IDF_4.4_IO13.sal.txt](https://github.com/espressif/esp-idf/files/10676411/IDF_4.4_IO13.sal.txt)
[IDF_4.4_IO4.sal.txt](https://github.com/espressif/esp-idf/files/10676421/IDF_4.4_IO4.sal.txt)
I don't see bigger differences - so CS don't have any impact on 4.4

Here are logs with debug from both version. Please notice that SCK is running on 4MHz so everything is slower, but you can see that on 5.0 GPIO4 it is longer than on 5.0 GPIO13
[LOG_IDF_4.4_IO4.txt](https://github.com/espressif/esp-idf/files/10676427/LOG_IDF_4.4_IO4.txt)
[LOG_IDF_5.0_IO4.txt](https://github.com/espressif/esp-idf/files/10676428/LOG_IDF_5.0_IO4.txt)
[LOG_IDF_5.0_IO13.txt](https://github.com/espressif/esp-idf/files/10676430/LOG_IDF_5.0_IO13.txt)
[LOG_IDF_4.4_IO13.txt](https://github.com/espressif/esp-idf/files/10676431/LOG_IDF_4.4_IO13.txt)

GPIO13 ESP-IDF-5.0:
```
I (27) boot: ESP-IDF v5.0 2nd stage bootloader
I (27) boot: compile time 15:18:22
I (27) boot: chip revision: v3.0
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 3, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0db4ch ( 56140) map 
I (130) esp_image: segment 1: paddr=0001db74 vaddr=3ffb0000 size=024a4h (  9380) load
I (134) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=25b80h (154496) map 
I (191) esp_image: segment 3: paddr=00045ba8 vaddr=3ffb24a4 size=0037ch (   892) load
I (192) esp_image: segment 4: paddr=00045f2c vaddr=40080000 size=0d678h ( 54904) load
I (220) esp_image: segment 5: paddr=000535ac vaddr=50000000 size=00010h (    16) load
I (227) boot: Loaded app from partition at offset 0x10000
I (227) boot: Disabling RNG early entropy source...
I (240) cpu_start: Pro cpu up.
I (240) cpu_start: Starting app cpu, entry point is 0x4008119c
0x4008119c: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
D (248) clk: RTC_SLOW_CLK calibration value: 3354790
I (259) cpu_start: Pro cpu start user code
I (259) cpu_start: cpu freq: 240000000 Hz
I (259) cpu_start: Application information:
I (264) cpu_start: Project name:     ethernet_basic
I (270) cpu_start: App version:      v1.20-6-gc65b1f0-dirty
I (276) cpu_start: Compile time:     Feb  7 2023 15:18:16
I (282) cpu_start: ELF file SHA256:  bb57b8f7c2c68146...
I (288) cpu_start: ESP-IDF:          v5.0
D (293) memory_layout: Checking 7 reserved memory ranges:
D (298) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (304) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb31a0
D (311) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (317) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (324) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (330) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at C:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1742

D (337) memory_layout: Reserved memory range 0x40080000 - 0x4008d678
0x40080000: _WindowOverflow4 at C:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1742

D (343) memory_layout: Building list of available memory regions:
D (349) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (356) memory_layout: Available memory region 0x3ffb31a0 - 0x3ffb8000
D (362) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
D (369) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
D (376) memory_layout: Available memory region 0x3ffc2000 - 0x3ffc4000
[... 727 lines cut ...]
D (4260) spi_master: device5 locked the bus
D (4310) sdspi_host: poll_busy: timeout
D (4320) spi_master: device5 release bus
I (4320) test: t-t1:117, line: 8330
D (4320) vfs_fat: vfs_fat_lseek: offset=0, filesize:=140000
D (4320) spi_master: device5 locked the bus
D (4360) sdspi_host: poll_busy: timeout
D (4380) spi_master: device5 release bus
D (4380) vfs_fat: vfs_fat_lseek: offset=0, filesize:=140000
D (4380) vfs_fat: vfs_fat_lseek: offset=103110, filesize:=140000
D (4380) spi_master: device5 locked the bus
D (4420) sdspi_host: poll_busy: timeout
D (4440) spi_master: device5 release bus
I (4440) test: t-t1:117, line: 7366
D (4440) vfs_fat: vfs_fat_lseek: offset=0, filesize:=140000
D (4440) spi_master: device5 locked the bus
D (4480) sdspi_host: poll_busy: timeout
D (4490) spi_master: device5 release bus
D (4490) vfs_fat: vfs_fat_lseek: offset=0, filesize:=140000
D (4490) vfs_fat: vfs_fat_lseek: offset=30352, filesize:=140000
D (4500) spi_master: device5 locked the bus
D (4540) sdspi_host: poll_busy: timeout
D (4550) spi_master: device5 release bus
I (4550) test: t-t1:117, line: 2169
```

adokitkat, 2023-02-17:
Thank you very much for the detailed info :) 

Would it be ok if this issue was closed as your problem was solved (albeit with a workaround) and I would open another issue to track the possible regression down?

tkuc, 2023-02-17:
Yes, for me it is fine to close this one as issue is solved by woraround.

Workaround: 
In case of SPI SD communication use CS defined as -1 (not used) to avoid problem with long fseek() execution. This workaround will work only if there is only one slave device on SPI bus connected.

closed 2023-02-17 by tkuc

adokitkat, 2023-02-17:
> Workaround: In case of SPI SD communication use CS defined as -1 (not used) to avoid problem with long fseek() execution. This workaround will work only if there is only one slave device on SPI bus connected.

I would add your IDF version needs to be `master` or `release/v5.0` to be able to set CS pin as `-1` (unused) and the physical CS connection on SD card reader has to be pulled down / connected to GND.

adokitkat, 2023-03-28:
@tkuc Hi again! I've noticed on your schematics you don't have CS and MISO/SD_DO connected to a pull-up resistor. This could be the issue. I've figured out the change causing this issue was introduced in this PR https://github.com/espressif/esp-idf/pull/7710. The added `poll_busy` is timing out because it waits until MISO goes high - which it automatically doesn't in your case I would assume.

This is how the Micro SD card reader is connected in our ESP32-WROVER-KIT dev board:
[image]

In our [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#sd-pull-up-requirements) it is also stated it should be that way.

If you cannot modify your board you can try to set the pull mode on those pins which don't have an external pull-up resistor?
I am not sure if it is going to work, but ESP32 has some internal pull-up and pull-down resistors. 

[`esp_err_t gpio_set_pull_mode(gpio_num_t gpio_num, gpio_pull_mode_t pull)`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html?#_CPPv418gpio_set_pull_mode10gpio_num_t16gpio_pull_mode_t)

Can you please try it? Thank you very much.

tkuc, 2023-03-29:
Nice finding.
Once i will have possibilities i will test this this approach on v5.0. Based on your measurments, seek duration are close to my measurment so this have to be it.

adokitkat, 2023-04-05:
Could you try to add this line before the `esp_vfs_fat_sdspi_mount` call (code without the CS == -1 workaround)?

`gpio_pullup_en(GPIO_NUM_19);` (change the number to number of GPIO pin you use as MISO)

I tried to simulate your SD reader schematics by externally connecting MISO to 1MOhm pull-down resistor, which resulted in slowdown you're seeing and then I added the `gpio_pullup_en` function to enable internal pull-up resistor which restored the IO speeds again. If this works for you too then it's better solution than the previous workaround.