espressif/esp-idf #10496: Linking static library created by custom command causes `undefined reference to `__wrap_longjmp'`.

yuuu, 2023-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.0

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Can link static libraries generated by custom commands, including longjmp.

### What is the actual behavior?

When I run `idf.py build`, I get the following error.

```
/Users/user/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/user/setjmp_project/components/setjmp_component/libsetjmp_test.a(setjmp_test.o): in function `func':
setjmp_test.c:(.text+0x29): undefined reference to `__wrap_longjmp'
```

A project for reproduction was created and published in the following repository.
https://github.com/yuuu/setjmp_project

### Steps to reproduce.

1. `git clone git@github.com:yuuu/setjmp_project.git`
2. `cd setjmp_project`
3. `idf.py build`
...

### More Information.

According to my verification, if I modify the code to call `longjmp()` in `app_main()`, the link succeeds.
Is there some condition for `__wrap_longjmp()` to be linked?

igrr, 2023-01-06:
Hi @yuuu,

There are two issues which need to be fixed for your project to work:

1. The imported library uses `longjmp` and `setjmp` functions which are provided by another library. Therefore you need to tell CMake that the imported library needs to be linked to the library providing these functions. In the case of these particular functions, the CMake library is `idf::esp_rom`.
2. There is also an error `setjmp_test.c:(.text+0x4d): dangerous relocation: call8: call target out of range: setjmp`. You need to compile the source files with `-mlongcalls` flag in your custom Makefile.

<details><summary>Here's the diff to make the project work</summary>

```diff
diff --git a/components/setjmp_component/CMakeLists.txt b/components/setjmp_component/CMakeLists.txt
index e6a9ef5..c3e632f 100644
--- a/components/setjmp_component/CMakeLists.txt
+++ b/components/setjmp_component/CMakeLists.txt
@@ -14,6 +14,7 @@ set_target_properties(libsetjmp_test PROPERTIES IMPORTED_LOCATION ${LIB_FILE})
 set_target_properties(libsetjmp_test PROPERTIES INTERFACE_INCLUDE_DIRECTORIES "${COMPONENT_PATH}")
 
 target_link_libraries(${COMPONENT_LIB} INTERFACE libsetjmp_test)
+target_link_libraries(libsetjmp_test INTERFACE idf::esp_rom)
 
 add_custom_target(setjmp_test DEPENDS ${LIB_FILE})
 add_dependencies(${COMPONENT_LIB} setjmp_test)
diff --git a/components/setjmp_component/Makefile b/components/setjmp_component/Makefile
index a302ad3..6c99caf 100644
--- a/components/setjmp_component/Makefile
+++ b/components/setjmp_component/Makefile
@@ -2,7 +2,7 @@ libsetjmp_test.a: setjmp_test.o
 	xtensa-esp32-elf-ar rcs libsetjmp_test.a setjmp_test.o
 
 setjmp_test.o: setjmp_test.c setjmp_test.h
-	xtensa-esp32-elf-gcc -c setjmp_test.c -o setjmp_test.o
+	xtensa-esp32-elf-gcc -mlongcalls -c setjmp_test.c -o setjmp_test.o
 
 clean:
 	rm -f setjmp_test.o libsetjmp_test.a
```
</details>

Additionally, you can simplify the process of importing a prebuilt library using a helper function [`add_prebuilt_library`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#using-prebuilt-libraries-with-components):

```cmake
add_prebuilt_library(libsetjmp_test ${LIB_FILE}
             PRIV_REQUIRES esp_rom)
target_link_libraries(${COMPONENT_LIB} INTERFACE libsetjmp_test)
```

<details><summary>(Another diff)</summary>

```diff
diff --git a/components/setjmp_component/CMakeLists.txt b/components/setjmp_component/CMakeLists.txt
index e6a9ef5..901c398 100644
--- a/components/setjmp_component/CMakeLists.txt
+++ b/components/setjmp_component/CMakeLists.txt
@@ -9,10 +9,8 @@ add_custom_command(
   VERBATIM
 )
 
-add_library(libsetjmp_test STATIC IMPORTED)
-set_target_properties(libsetjmp_test PROPERTIES IMPORTED_LOCATION ${LIB_FILE})
-set_target_properties(libsetjmp_test PROPERTIES INTERFACE_INCLUDE_DIRECTORIES "${COMPONENT_PATH}")
-
+add_prebuilt_library(libsetjmp_test ${LIB_FILE}
+             PRIV_REQUIRES esp_rom)
 target_link_libraries(${COMPONENT_LIB} INTERFACE libsetjmp_test)
 
 add_custom_target(setjmp_test DEPENDS ${LIB_FILE})
diff --git a/components/setjmp_component/Makefile b/components/setjmp_component/Makefile
index a302ad3..6c99caf 100644
--- a/components/setjmp_component/Makefile
+++ b/components/setjmp_component/Makefile
@@ -2,7 +2,7 @@ libsetjmp_test.a: setjmp_test.o
 	xtensa-esp32-elf-ar rcs libsetjmp_test.a setjmp_test.o
 
 setjmp_test.o: setjmp_test.c setjmp_test.h
-	xtensa-esp32-elf-gcc -c setjmp_test.c -o setjmp_test.o
+	xtensa-esp32-elf-gcc -mlongcalls -c setjmp_test.c -o setjmp_test.o
 
 clean:
 	rm -f setjmp_test.o libsetjmp_test.a
```
</details>

yuuu, 2023-01-06:
Thank you, @igrr .

`add_prebuilt_library(libsetjmp_test ${LIB_FILE} PRIV_REQUIRES esp_rom)` is the magic code for me. I am so happy to know this function.

Thanks for your help with the missing `-mlongcalls` description. I think I need that too.

The problem I had has been solved, so I am closing this issue.

closed 2023-01-06 by yuuu

nopnop2002, 2024-01-14:
@yuuu 

I also have the same problem.
https://github.com/yuuu/setjmp_project
I got this error.
```
undefined reference to `__wrap_longjmp'
```

I would like to know the final solution.