espressif/esp-idf #10497: Disabling Scan Mode in ESP32S3 Touch Interface

5ami, 2023-01-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi I am using the ESP32s3 touch peripheral to read Four different pads. in the TRM it is mentioned that "scan mode can be enabled by specifying the bit map of enabled touch pins in RTC_CNTL_TOUCH_SCAN_PAD_MAP in  RTC_CNTL_TOUCH_SCAN_CTRL_REG" 

My question is how can I get multiple touch pads to sample simultaneously, I couldn't find an API function in the esp-idf v5.0 to do this, I also didn't find any useful information regarding it in the ESP32s3 TRM.   The reason I want to do that is to reduce the overall sensor delay.

I tried for two pads: 
-setting the corresponding   RTC_CNTL_TOUCH_SCAN_PAD_MAP bits  to 0 ==>no pin start the measurement. 
-setting the corresponding   RTC_CNTL_TOUCH_SCAN_PAD_MAP bits  to 1  ==>Both pins start the measurements but in scan mode. 
--setting one pad  bit to 1  and  the other to 0 in   RTC_CNTL_TOUCH_SCAN_PAD_MAP will start the channels simultaneously but the the magnitude of the other channel is lower by 10 and it gives a false reading.  

[image]

5ami, 2023-03-04:
@L-KAYA any answer regarding this ?

L-KAYA, 2023-03-20:
Sorry for the late reply!

Unfortunately, the touch sensor in hardware only supports scanning the pads one by one. The register `RTC_CNTL_TOUCH_SCAN_PAD_MAP ` is to enable the pads which is supposed to be scanned, if the corresponding bit is not set, the touch sensor scanner will skip this pad, it's kinda `enable` logic.

5ami, 2023-03-21:
@L-KAYA Thank you for the replay

closed 2023-06-05 by L-KAYA