espressif/esp-idf #10501: fatal error: driver/uart.h: No such file or directory

Oleg-Perevyshin, 2023-01-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release 5.0

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Successful project build

### What is the actual behavior?

fatal error: driver/uart.h: No such file or directory

### Steps to reproduce.

1. Create an empty project (no template) with Espressif-IDE
2. Create new component with Espressif-IDE (_uart_)
3. Attach header file _uart_.h to main.c
4. Attach header file _uart_.h to _uart_.c
5. Include driver/uart.h to _uart_.c or _uart_.h
6. Start configuring the port UART0
7. Get build error "fatal error: driver/uart.h: No such file or directory"

### Build or installation Logs.

```plain
Building in: D:\SOI-Tech\System\WiFi-Generic\build
cmake --build . -- -v
[1/11] cmd.exe /C "cd /D D:\SOI-Tech\System\WiFi-Generic\build\bootloader && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe --build ."
[1/1] cmd.exe /C "cd /D D:\SOI-Tech\System\WiFi-Generic\build\bootloader\esp-idf\esptool_py && python C:/Espressif/frameworks/esp-idf-v5.0/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 D:/SOI-Tech/System/WiFi-Generic/build/bootloader/bootloader.bin"
Bootloader binary size 0x5310 bytes. 0x1cf0 bytes (26%) free.
[2/9] ccache C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -ID:/SOI-Tech/System/WiFi-Generic/build/config -ID:/SOI-Tech/System/WiFi-Generic/components/_uart_/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/esp_additions/include/freertos -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/esp_additions/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/include/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/port/esp32/private_include -IC:/Espressif/frameworks/esp-idf-v5.0/components/heap/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/log/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/soc/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/soc/esp32/. -IC:/Espressif/frameworks/esp-idf-v5.0/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/hal/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/include/esp32 -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/esp32 -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/soc -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/include/private -IC:/Espressif/frameworks/esp-idf-v5.0/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/xtensa/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/port/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/port/esp32/include/arch -mlongcalls -Wno-frame-address  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -O2 -fmacro-prefix-map=D:/SOI-Tech/System/WiFi-Generic=. -fmacro-prefix-map=C:/Espressif/frameworks/esp-idf-v5.0=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0-dirty\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/_uart_/CMakeFiles/__idf__uart_.dir/_uart_.c.obj -MF esp-idf\_uart_\CMakeFiles\__idf__uart_.dir\_uart_.c.obj.d -o esp-idf/_uart_/CMakeFiles/__idf__uart_.dir/_uart_.c.obj -c D:/SOI-Tech/System/WiFi-Generic/components/_uart_/_uart_.c
FAILED: esp-idf/_uart_/CMakeFiles/__idf__uart_.dir/_uart_.c.obj 
ccache C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -ID:/SOI-Tech/System/WiFi-Generic/build/config -ID:/SOI-Tech/System/WiFi-Generic/components/_uart_/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/esp_additions/include/freertos -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/freertos/esp_additions/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/include/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_hw_support/port/esp32/private_include -IC:/Espressif/frameworks/esp-idf-v5.0/components/heap/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/log/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/soc/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/soc/esp32/. -IC:/Espressif/frameworks/esp-idf-v5.0/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/hal/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/include/esp32 -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_rom/esp32 -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/soc -IC:/Espressif/frameworks/esp-idf-v5.0/components/esp_system/port/include/private -IC:/Espressif/frameworks/esp-idf-v5.0/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/xtensa/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/port/esp32/include -IC:/Espressif/frameworks/esp-idf-v5.0/components/lwip/port/esp32/include/arch -mlongcalls -Wno-frame-address  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -O2 -fmacro-prefix-map=D:/SOI-Tech/System/WiFi-Generic=. -fmacro-prefix-map=C:/Espressif/frameworks/esp-idf-v5.0=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAW [... cut ...]
void app_main(void) {
  uart_init();
}
```
Content _uatr_.c
```cpp
#include <stdio.h>
#include "driver/uart.h"

esp_err_t uart_init(void) {
  uart_config_t uart_config = {
      .baud_rate = 115200,
      .data_bits = UART_DATA_8_BITS,
      .parity = UART_PARITY_DISABLE,
      .stop_bits = UART_STOP_BITS_1,
      .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
      .source_clk = UART_SCLK_REF_TICK
  };
  if (uart_param_config(0, &uart_config) == ESP_FAIL) {
    return ESP_FAIL;
  }
// ...
  return ESP_OK;
}
```

suda-morris, 2023-01-08:
It's due to a dependency change in 5.0. You may want to check [this migration guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/build-system.html#specify-component-requirements-explicitly) first.

Oleg-Perevyshin, 2023-01-09:
I read it, but nothing works.
I took as a basis the project from the example. It works, but when you add a component, the project is not built.
Maybe there is some example of settings using the UART component?

igrr, 2023-01-09:
Could you please show the contents of components/uart/CMakeLists.txt file? Very likely the issue is that you are missing `REQUIRES driver` argument in idf_conponent_register() call.

(The same works inside the `main` component without the extra requirement die to this: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#main-component-requirements)

Oleg-Perevyshin, 2023-01-09:
My file CMakeLists.txt (components/_uart_)

```cmake
idf_component_register(
	SRCS
		"_uart_.c"
	
	INCLUDE_DIRS
		"include"

	REQUIRES
		uart

	PRIV_REQUIRES
		
)

```

igrr, 2023-01-09:
Could you please remove the build directory, do a build of the project and attach
* the full build log you get
* `build/compile_commands.json` file

(with the 2nd version of CMakeLists.txt, which has `REQUIRES driver`)

Oleg-Perevyshin, 2023-01-09:
Deleted the folder and the project was assembled correctly!
Thank you!

closed 2023-01-09 by Oleg-Perevyshin