espressif/esp-idf #10503: ESP-IDF Flash encryption in development mode

abhi-ratnman, 2023-01-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32 WROOM 32

### Power Supply used.

USB

### What is the expected behavior?

I am using the flash encryption in development mode, the menu config for flash encryption is as per below image

[image]

generation the key and nvs file using the below command
`nvs_partition_gen.py generate credentials.csv nvs.bin 0x4000`

and flashing the same using this command
`
esptool.py -b 460800 --before default_reset --after no_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build\bootloader\bootloader.bin 0x10000 build\partition_table\partition-table.bin 0x15000 build\ota_data_initial.bin 0x20000 build\app_main.bin 0x21a000 build\storage.bin 0x11000 nvs.bin 0x22a000 keys\keys.bin`

After this the code gets uploaded and on first start the flash gets encrypted. And after that it restarts and works corrently. But when we reflash using the same command we used to flash it throws the below error.
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371`

### What is the actual behavior?

We should be able to reflash the code

### Steps to reproduce.

As per above

mahavirj, 2023-01-08:
@abhi-ratnman 

Please try using `idf.py encrypted-flash` command to flash encrypted partitions, see [Reference](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#re-flashing-updated-partitions).

Alternatively, you may also use `esptool` based `write_flash` command with `--encrypt` argument.

abhi-ratnman, 2023-01-09:
Hi thanks for the help. I used the above command and it solves the purpose. Thanks

closed 2023-01-09 by abhi-ratnman